moveit/moveit #719: OMPL Info and Debug logs are shown since latest update, even though verbose/debug mode is not specified

hansondon, 2017-12-12:
### Description
OMPL Info and Debug logs are shown since latest update, even though verbose/debug mode is not specified.

### Your environment
* ROS Distro: Kinetic
* OS Version: Ubuntu 16.04
* Binary build:  0.9.10

### Steps to reproduce
From package created with Setup Assistant, launch move_group.launch with arguments `debug` and `info` set to false.

### Expected behaviour
Not to see OMPL Debug or Info logs in the terminal.

### Actual behaviour
[image]

### Backtrace or Console output
https://gist.github.com/hansondon/229df6962bb55389cc72fcb2b2fa7832

beckss, 2018-01-04:
Same issue here.

guihomework, 2018-05-24:
same here :+1:

PR #916 (2018-05-24): OMPL logging that uses rosconsole
### Description

Fixes #719. Adds a custom `ompl::msg::OutputHandler` that uses `rosconsole` to log OMPL messages at the appropriate logging level.

### Checklist
- [x] **Required**: Code is auto formatted using [clang-format](http://moveit.ros.org/documentation/contributing/code)
closed 2018-05-24 by rhaschke