moveit/moveit #722: waitForCurrentState(wait_time, ?planning_group=""?)

v4hn, 2017-12-12:
Some time ago @rhaschke resolved an inherent issue with our [`CurrentStateMonitor`](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_ros/planning/planning_scene_monitor/include/moveit/planning_scene_monitor/current_state_monitor.h#L56) and made the [`waitForCurrentState`](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_ros/planning/planning_scene_monitor/src/current_state_monitor.cpp#L265) method __actually wait__ for processed joint state messages that refer to the current time. In order to do so, he used a [`ros::Time current_state_time_` member](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_ros/planning/planning_scene_monitor/include/moveit/planning_scene_monitor/current_state_monitor.h#L208) that is updated whenever a new joint state is received.

However this poses another inherent problem that still exists in the current implementation:
It assumes that the current message contains information **on all joints**, which is usually not the case if the hardware setup has more than one component with different hardware drivers.

To workaround this problem, users can remap their `joint_states` topics and merge them before they are fed into MoveIt, but this is almost never done and adds overhead and a delay.

A solution to this problem probably has to store a timestamp *per joint* (or joint sets published together)  and use the oldest timestamp (possibly within the set of joints of some planning group of interest) to represent the actual `current_state_time_`.

cbames, 2018-01-22:
I do believe I'm running into the effects of this issue. 

I'm observing behavior very similar to what #716 was supposed to fix. Namely, the start state for a plan is very different than what it currently is. This results in jolts and pops at the beginning of the motion, of course it only happens intermittently.  

My current setup: 
14.04, indigo, source build from indigo-devel. I've made one small modification so that I can use TRAC-IK's distance solve type and python moveit commander.

I have one Jaco arm and a FLIR PTU both publishing separate messages to joint_states. 

What I think is happening; 

If moveit's state gets updated on a PTU joint state message, the old arm data is retained, and then used for planning. Thus causing incorrect behaviour. 

 I'll take a look at #716, but I'd appreciate if @rhaschke  or @v4hn could point me to the group of files to fix.

gavanderhoorn, 2018-01-22:
@cbames: slightly off-topic, but are you using the `source_list` parameter of `joint_state_publisher` ([wiki/joint_state_publisher - Parameters](http://wiki.ros.org/joint_state_publisher#Parameters))?

rhaschke, 2018-01-22:
@cbames: Could you please update the issue references in your comment. #716 doesn't seem to be related to this issue and it doesn't exist in the old [`moveit_core`](https://github.com/ros-planning/moveit_core) repo.

Recently, #717 was merged to fix an issue in Indigo related to `waitForCurrentState`. I guess, you already merged this one. 

If you want to tackle the present issue #722, you should have a look into `planning_scene_monitor::CurrentStateMonitor`. New joint states are received in method [`jointStateCallback`](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_ros/planning/planning_scene_monitor/src/current_state_monitor.cpp#L331), which [updates the global variable `current_state_time_`](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_ros/planning/planning_scene_monitor/src/current_state_monitor.cpp#L345) as well as joint-specific variables [`joint_time_`](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_ros/planning/planning_scene_monitor/src/current_state_monitor.cpp#L355). I think, it suffices to compute the global variable `current_state_time_` as the minimum of all joint-specific variables (instead of setting it to the timestamp of the processed message).
To stay performant in the most common case, where all variables are published by a single publisher, you should add a shortcut like `number of joints in message == number of variables/joint in model => use timestamp as before`. And, last but not least, thanks for working on this!

cbames, 2018-01-22:
@rhaschke  updated to point to the PR that I was referring to. It might also be unrelated, but when I read it, there were similarities in symptoms. 

@gavanderhoorn  I'm not using that parameter, but it looks like a nifty one! I didn't know it existed, thanks for pointing it out!

gavanderhoorn, 2018-01-22:
It's at best a work-around (although we could argue that making all consumers of `JointState` messages responsible for coalescing them into a single view of the system seems rather wasteful).

v4hn, 2018-01-26:
> I think, it suffices to compute the global variable `current_state_time_` as the minimum of all joint-specific variables (instead of setting it to the timestamp of the processed message).

I agree. When I wrote this issue I wasn't aware of the `joint_time_` member.
@cbames did you/do you look into this approach? It would be great to see a pull-request for it.

cbames, 2018-01-29:
@v4hn  I did take a look at it, I wrote up a solution. However, it didn't fix the problem I was experiencing. So I'll have to circle back around after the RSS deadline and see if I can better understand the problem this is causing so that I can be sure my fix actually fixes it.