moveit/moveit #723: attached bodies are not shown in trajectory visualization anymore

rhaschke, 2017-12-14:
Since MoveIt 0.9.10 attached bodies are not shown in trajectory visualization using rviz.
Is anybody aware of changes related to that? I appreciate any hint. Otherwise I will go and bisect...

v4hn, 2017-12-14:
This issue is a bit underspecified...

Do you mean if you send a `moveit_msgs::DisplayTrajectory` with `AttachedCollisionObjects` in `trajectory_start.attached_collision_objects`, then they are not shown in the visualized trajectory?
I'm not aware of any change there.

jeffreyling, 2017-12-15:
I ran into a similar issue, and I think I found the problem. In [PR 663](https://github.com/ros-planning/moveit/pull/663/files) the function `attachedBodiesToAttachedCollisionObjectMsgs` in `robot_state/conversions.h` should take a vector reference as argument, not a vector copy. 

This is messing up the part where the attached bodies are supposed to be added to the msg.

rhaschke, 2017-12-15:
@jeffreyling Thanks for tracking this down. I will have a look.

rhaschke, 2017-12-17:
Closing via #724

closed 2017-12-17 by rhaschke