moveit/moveit #728: Octomap not transformed properly in rviz

jeffreyling, 2017-12-20:
### Description

I'm having a similar issue as the one reported [here](http://webcache.googleusercontent.com/search?q=cache:https://answers.ros.org/question/253132/discrepancy-between-octomap-visualization-and-actual-octomap-in-moveit/). Basically, rviz is ignoring the pose I give my octomap.

I add an octomap to the planning scene, putting it in its own frame and set a transform for that frame (say at coordinates `1, 0, 0`). When I visualize the scene in rviz, however, the octomap appears in the default pose (`0, 0, 0`). On the other hand, planning with the scene gives collisions consistent with the octomap in the transformed pose.

Example:
``` C++
// msg is octomap_msgs::Octomap where header.frame_id is "/my_pcl"
Eigen::Affine3d pose(Eigen::Translation3d(1., 0., 0.) *
                     Eigen::Quaternion<double>(1., 0., 0., 0.));
planning_scene_->getTransformsNonConst().setTransform(pose, "/my_pcl");
std::cout << "Knows my_pcl: " << planning_scene_->knowsFrameTransform("/my_pcl") << std::endl; // true
auto tf = planning_scene_->getTransforms().getTransform("/my_pcl");
auto t = tf.translation();
std::cout << "the actual tf: " << t[0] << " " << t[1] << " " << t[2] << std::endl; // 1 0 0
planning_scene_->processOctomapMsg(msg);

// check
for (auto p : planning_scene_->getWorld()->getObject("<octomap>")->shape_poses_) {
    auto t = p.translation();
    std::cout << "octomap pose: " << t[0] << t[1] << t[2] << std::endl; // 1 0 0
}
```
All the checks give `1 0 0` as expected. However, rviz publishes the octomap at `0 0 0`.

I looked into this a bit, and it seems like it might be because the pose `p` is ignored for the octomap case here: https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_ros/visualization/rviz_plugin_render_tools/src/render_shapes.cpp

Please let me know if I'm missing something obvious, thanks!

### Your environment
* ROS Distro: Kinetic
* OS Version: Ubuntu 16.04
* Binary 0.9.9

eyalu, 2018-03-12:
I confirm that I have the same problem. It forces me to rely blindly on moveit for the obstacle avoidance without rviz for visual debugging.

Environment:
Kinetic, ubuntu 16.04 with rviz 1.12.15
Qt 5.5.1 and OGRE 1.9.0

rhaschke, 2021-04-15:
I wasn't able to reproduce the described behavior and the code snipped doesn't show how you actually communicate your scene to rviz. Usually, the `move_group` node maintains the central PlanningScene, receiving updates on `/planning_scene` and `/planning_scene_world` and publishing regularly to `move_group/monitored_planning_scene`, which is then picked up by rviz.
When feeding the `PlanningScene` via the two input topics, it is important to note that `/planning_scene` expects self-contained messages, including all required transforms. The corresponding update function doesn't retrieve transforms from `tf`!
On the other hand, feeding `/planning_scene_world` current transforms are fetched from `tf` and inserted into the move_group's PlanningScene. The PlanningScene send on the output topic to rviz always uses the base robot frame for all `frame_ids`. Thus, if you inspect the topic, the transforms might have changed (in numbers), because they are represented w.r.t. a different frame now.
However, the shown code snippet correctly sets required transforms using `ps_->getTransformsNonConst().setTransform()`.

closed 2021-04-15 by rhaschke