moveit/moveit #730: Can I get the query start state and query goal state from the rviz plugin?

onionsflying, 2017-12-29:
It would be nice If I can get the query start state and query goal state from the rviz plugin. But I can't find interfaces for that. Does the rviz plugin have such a function? Hope someone could give me an answer.

rhaschke, 2017-12-29:
There is no explicit interface to access the state of the rviz plugin. This plugin is intended for visualization and manual interaction with the `move_group` node. 
However, the rviz plugins use ROS communication to trigger planning in the `move_group` node. A plan request is sent through the topic `/move_group/goal` when planning is triggered. If you listen to this topic, you can extract the start and goal state.
The Cartesian poses are communicated via interactive markers, which you can listen to on the topic `/rviz_moveit_motion_planning_display/robot_interaction_interactive_marker_topic/update`.

onionsflying, 2017-12-29:
Got it. Thanks for the reply.

closed 2017-12-29 by rhaschke