moveit/moveit #735: Visualized rviz path is different from the executed one

ptriantd-ocado, 2018-01-02:
### Description

I have a scene with some obstacles and a robot for which all joint constraints are enabled (position, velocity & acceleration). There are also some extra position constraints for the robot's end-effector. 

Moveit finds a plan from one position to another and I am visualizing it in rviz. What I notice is that the plan has some "jumps" in it as if the robot is passing close to a singular configuration. However, when I query the velocities at the path points, they seem to be lower than the manifacturer's limits (which are the same limits I have set at joint_limits.yaml).

But when I execute the plan in simulation, the path followed by the robot is not the one visualized in rviz. The robot goes all over the place, violating the cartesian position constraints and passing through obstacles.

Is the plan seen in rviz not the final executed plan? 

### Your environment
* ROS Distro: Indigo
* OS Version: Ubuntu 14.04
* Source or Binary build? Binary
* If binary, which release version? 0.7.12

rhaschke, 2018-01-03:
`rviz` only visualizes the via points of a path. During execution controllers will smoothly interpolate between those via points. If two consecutive via points have a strong positional deviation (in joint space only, e.g. switching from elbow up to down) this will result in a weird trajectory.
Then the question why MoveIt returns such a path. This is mainly due to limited sampling resolution for planning and collision checking. You could try to decrease the sampling step size.
To better understand the problem, it would be very helpful if you could provide your visualized and executed trajectory as a ROS msg.

ptriantd-ocado, 2018-01-03:
Thanks for the quick reply! So, you suggest to lower the `longest_valid_segment_fraction` parameter, right? I did that and, indeed, the plans stopped being "jumpy" with `longest_valid_segment_fraction < 0.005`. But now the planner does not usually manage to plan successfully even after 1 minute of planning time. Is there any way to get smoother trajectories without bumping up the required planning time that much? Or any other parameters I could change?

felixvd, 2021-08-10:
Closing this for inactivity

closed 2021-08-10 by felixvd