moveit/moveit #736: Interactive marker missing after upgrade

teerawat-to, 2018-01-04:
Interactive marker missing after upgrade. Please help us
best regard

rhaschke, 2018-01-04:
Unfortunately, this is a regression in `rviz`. I already filed https://github.com/ros-visualization/rviz/pull/1179 to fix the issue. Meanwhile you could apply this patch and build rviz yourself.

closed 2018-01-04 by rhaschke

benbernhard1, 2019-02-21:
@rhaschke Sorry, I am new to this, but it is still not working for me.

 I used the code from the commit   [https://github.com/ros-visualization/rviz/pull/1179](url)  by using `git checkout 866d37618093d4e1296d664b6151ca1b02f3467b` 

Then I built again using cmake, make, make install.

Yet, when trying to complete the tutorial at [http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/quickstart_in_rviz/quickstart_in_rviz_tutorial.html](url) the interactive markers do not display.

I also get the warning: `[ WARN] [1550721475.091377953]: Unable to update multi-DOF joint 'virtual_joint': TF has no common time between '/world' and 'panda_link0': 
`

rhaschke, 2019-02-21:
The fix in rviz is released for a long time. No need to build from source.
The error msgs looks like your panda robot is not correctly fixed to the world frame. Did you inspect the TF tree? Are there disconnected sub trees?

benbernhard1, 2019-02-21:
@rhaschke 
Here is my transform tree. 
[frames.pdf](https://github.com/ros-planning/moveit/files/2889281/frames.pdf)

Note that I followed the tutorials exactly as shown for MoveIt. It is the same issue here [https://answers.ros.org/question/315758/no-interactive-markers-displayed-whilst-following-moveit-tutorial-ros-kinetic/](url)

rhaschke, 2019-02-21:
The TF tree is missing the `world` frame. Probably, there is a bug in the tutorials then (which shouldn't use `world`). @mlautman Can you have a look, please?