moveit/moveit #739: Path error when trying to follow Move Group Interface Tutorial.

daniela0113, 2018-01-09:
### Description

I have built MoveIt! from source and am trying to follow the Move Group Interface Tutorial. However, I am unable to successfully launch move_group_interface_tutorial.launch. I get some sort of path error or node error. Not sure, pretty new to ROS.

ERROR: cannot launch node of type [moveit_tutorials/move_group_interface_tutorial]: moveit_tutorials
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/opt/ros/kinetic/share

### Your environment
* ROS Distro: Kinetic
* OS Version: Ubuntu 16.04
* Source kinetic-devel

### Steps to reproduce

Step 1: Build MoveIt! from source: http://moveit.ros.org/install/source/
```
rosdep update
sudo apt-get update
sudo apt-get dist-upgrade

sudo apt-get install python-wstool python-catkin-tools clang-format-3.8

mkdir -p ~/ws_moveit/src
cd ~/ws_moveit/src

wstool init .
wstool merge https://raw.githubusercontent.com/ros-planning/moveit/kinetic-devel/moveit.rosinstall
wstool update
rosdep install -y --from-paths . --ignore-src --rosdistro kinetic
cd ..
catkin config --extend /opt/ros/kinetic --cmake-args -DCMAKE_BUILD_TYPE=Release
catkin build
source ~/ws_moveit/devel/setup.bash
```
2. Follow Move Group Interface Tutorial: http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/pr2_tutorials/planning/src/doc/move_group_interface_tutorial.html

```
cd ~/ws_moveit/src
git clone https://github.com/ros-planning/moveit_tutorials.git
git clone https://github.com/PR2/pr2_common.git -b kinetic-devel
git clone https://github.com/davetcoleman/pr2_moveit_config.git

```
 Before installing the dependencies and rebuilding the workspace, I noticed there was a bug in ~/ws_moveit/src/moveit_tutorials/doc/pr2_tutorials/planning/src/move_group_interface_tutorial.cpp

I made changes based on a suggestion at the end of https://robotics.stackexchange.com/questions/14801/catkin-make-unable-to-build-and-throws-makefile138-recipe-for-target-all-fa

```
rosdep install --from-paths . --ignore-src --rosdistro kinetic
cd ~/ws_moveit
catkin config --extend /opt/ros/kinetic --cmake-args -DCMAKE_BUILD_TYPE=Release
catkin build
source ~/ws_moveit/devel/setup.bash
```

Before this change, I would get the following errors.
[image]

After editing the move_group_interface_tutorial.cpp file, I don't get any errors, however, some of my packages are skipped/blacklisted. I'm sure this creates issues...
[image]

3. Start Rviz
```
cd ~/ws_moveit/src
roslaunch pr2_moveit_config demo.launch
```

4. Run demo, in separate terminal 
```
roslaunch ~/ws_moveit/src/moveit_tutorials/doc/pr2_tutorials/planning/launch/move_group_interface_tutorial.launch
```

### Expected behaviour
move_group_interface_tutorial.launch runs without errors.

### Actual behaviour
After launching move_group_interface_tutorial.launch, this is the error that I get. Note that pr2_moveit_config demo.launch runs perfectly fine.

Demo output:
[image]

Rviz output:
[image]

rhaschke, 2018-01-09:
Thank you for your detailed error report. The "skipped or blacklisted" packages don't pose an issue. These are simply meta-packages that are never build.
The most probable reason for your failure is that you missed to source your workspace _after_ the successful build, i.e. `source ~/ws_moveit/devel/setup.bash`. If you did it only before the moveit_tutorials build completed successfully (which wasn't the case initially), ROS cannot find the `moveit_tutorials` package, which is exactly the error message you get.

daniela0113, 2018-01-10:
@rhaschke Thank you for your time and feedback. You suggest that perhaps I forgot to source my workspace after the successful build. However, as you can see from the second image I posted, that I did source my workspace after catkin build. I started from scratch for a clean start. I deleted/removed my workspace and rebuilt it following the same steps (sourcing after any catkin build). Unfortunately I get the same error :(

rhaschke, 2018-01-10:
@daniela0113 In the next-to-last image you posted, the failing roslaunch reports about its ROS path. Only two pathes are listed there:
```
[0] /opt/ros/kinetic/share/ros
[1] /opt/ros/kinetic/share/ros
```
Hence, I'm pretty sure, that your ROS environment is not correctly loaded. Please check `ROS_PACKAGE_PATH` environment variable after sourcing `setup.bash`. It should comprise your local workspace.

daniela0113, 2018-01-11:
I added the following line to my ~/.bashrc

`export ROS_PACKAGE_PATH=/home/adf589/ws_moveit/src:$ROS_PACKAGE_PATH`

I don't get a path error anymore, but still get an error.
Note that after I modified my ~/.bashrc I sourced it and rebuilt my workspace.

[image]

daniela0113, 2018-01-11:
@rhaschke

rhaschke, 2018-01-11:
Instead of manually adding a path to `ROS_PACKAGE_PATH` in your `.bashrc`, you should source the final workspace with `source setup.bash`. `ROS_PACKAGE_PATH` is only one of several environment variables that are set by the script.

daniela0113, 2018-01-22:
@rhaschke Sorry for the late reply. Just wanted to mention that your last comment was very helpful and I was able to fix my problem. Thank you for your time!

rhaschke, 2018-01-22:
If you were able to solve your problem, you should close the issue.

closed 2018-01-22 by rhaschke

amrith1007, 2018-03-16:
@daniela0113 @rhaschke Guys! A little help here from one of you would be deeply appreciated.
I have the exact same problem. Though I have sourced the setup.bash file. My error continues to persist. The only difference is that I have built MoveIt! from Binary. 

ROS Distro: Kinetic
OS Version: Ubuntu 16.04

on running` roslaunch moveit_tutorials move_group_interface_tutorial.launch`

The output is:

[image]

As seen in the image above, I have sourced my workspace (catkin_ws). I am however, able to run the move_group_python_interface_tutorial.py  node successfully. 

I have also tried to separately run the move_group_interface_tutorial.cpp node separately, on doing that I get the following output

[image]

FYI, I have not made any changes to the node files, I have cloned them as suggested in the moveit kinetic tutorial page.

Thank you very much in advance :) I look forward to your assistance with this problem.

rhaschke, 2018-03-18:
Have you actually build your workspace? If so, the binary should be there:
`devel/lib/moveit_tutorials/move_group_interface_tutorial`.
Obviously you cannot execute the `.cpp` file.

siksal, 2018-11-28:
I'm having the same issues as @amrith1007 but on ubuntu 18.04 with ros melodic. I've sourced my workspace but still getting the error below:

> ERROR: cannot launch node of type [moveit_tutorials/move_group_interface_tutorial]: can't locate node [move_group_interface_tutorial] in package [moveit_tutorials]

Also having errors while installing rosdep:

> ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
franka_visualization: Cannot locate rosdep definition for [libfranka]
franka_control: Cannot locate rosdep definition for [libfranka]
franka_hw: Cannot locate rosdep definition for [libfranka]
franka_example_controllers: Cannot locate rosdep definition for [libfranka]
franka_gripper: Cannot locate rosdep definition for [libfranka]

Still trying to figure out the problem. I've sourced the workspace but noticed the files I have in `ws_moveit/devel/lib` are `pkgconfig` and `python2.7`.

I don't know what to do, @rhaschke, any help?

rhaschke, 2018-11-28:
ROS Melodic doesn't yet have `libfranka` released. Did the workspace build successfully in first place?
Do you have a `move_group_interface_tutorial` binary in your WS?

rhaschke, 2018-11-29:
> the workspace was successfully built with errors.

This is a contradiction. Could you please provide the actual compiler errors from the build?
Otherwise, I cannot help you.

siksal, 2018-11-29:
This is the output from the terminal:
[image]

Continuation:
[image]

This:
[image]

And finally:
[image]

rhaschke, 2018-11-29:
This shows multiple errors: First of all, libfranka is not yet released to MoveIt, which is why the franka_ros related packages fail to build. Do you really need them? AFAIK, for the MoveIt tutorials they are not required.
Second, building the latest melodic-devel branch of MoveIt, needs updates to several MoveIt-related packages. Please ensure, all repositories are up-to-date. If necessary, repeat the steps at http://moveit.ros.org/install/source/#download-and-build-moveit.

siksal, 2018-11-29:
Thanks @rhaschke, I've repeated the steps from the provided link, yet, the error persists.

[image]

This is really challenging.

rhaschke, 2018-11-29:
You always need to show the first error. What you posted is only the final summary.
Why do you want to build MoveIt from source when you don't seem to be familiar with the ROS building process? You can also use prebuilt binary packages using the package management of the OS.

QNjeason, 2020-04-28:
[image]

What I have encountered is such a problem. How can I solve it

eves-d123, 2021-07-08:
I had the same errors for Ubuntu 18.04 with melodic but managed to fix it by repeating catkin builds. 

Steps followed: 
•	Removed unnecessary franka files (kept franka_description)
•	catkin clean
•	Running catkin build --jobs 1 a few times and catkin clean between them showed I needed to install these: 
       -	sudo apt-get install ros-melodic-moveit-visual-tools
       - 	sudo apt install ros-melodic-pcl-ros
•	catkin build --jobs 1 
Used catkin build --jobs 1 just in case https://github.com/ros-planning/moveit_tutorials/issues/469

simonschmeisser, 2021-10-05:
@hakancan14 this is completely unrelated! Please don't post to unrelated issues in unrelated repositories.

You need to install `ros-kinetic-map-server` and `ros-kinetic-amcl` 

Also please provide logs as text in future, not as screenshots

hakancan14, 2021-11-02:
Hello there. I am getting the following error: 'Map update loop missed its desired rate of 5.0000Hz... the loop actually took 0.3466 seconds'. How can I fix this error.  I beg you help....
________________________________
Gönderen: Simon Schmeisser ***@***.***>
Gönderildi: 5 Ekim 2021 Salı 17:20
Kime: ros-planning/moveit ***@***.***>
Bilgi: hakancan14 ***@***.***>; Mention ***@***.***>
Konu: Re: [ros-planning/moveit] Path error when trying to follow Move Group Interface Tutorial. (#739)

@hakancan14<https://github.com/hakancan14> this is completely unrelated! Please don't post to unrelated issues in unrelated repositories.

You need to install ros-kinetic-map-server and ros-kinetic-amcl

Also please provide logs as text in future, not as screenshots