moveit/moveit #741: How to install the specified version of moveit?

naijoug, 2018-01-16:
How to install the specified version of moveit? Now version 0.9.11, I want to install 0.9.10

rhaschke, 2018-01-16:
https://askubuntu.com/questions/138284/how-to-downgrade-a-package-via-apt-get
Why do you want to downgrade?

gavanderhoorn, 2018-01-16:
@rhaschke: not sure that is going to work: ROS package repositories are managed using `reprepro`, and that doesn't retain older versions of packages.

Only a MoveIt Docker image (old hash) and building from source would be options here afaik.

naijoug, 2018-01-17:
@rhaschke  Thank you for answering me.I tried your method, now simply can not find the lower version of the package.

BryceStevenWilley, 2019-04-26:
To answer the original question, as Docker seems to only save the [last few hundred commits](https://hub.docker.com/r/moveit/moveit/builds?page=10), at this point the only option is to build from source, which you could do by following the [source install instructions](https://moveit.ros.org/install/source/) and before running the `rosdep install` command, run `git checkout 0.9.10` in all of the cloned repos (listed [here](https://raw.githubusercontent.com/ros-planning/moveit/master/moveit.rosinstall)), or whichever tag is closest to `0.9.10`.

@naijoug, Did you resolve this a different way?