moveit/moveit #742: Thoughts on adding support for convex shapes?

Levi-Armstrong, 2018-01-16:
I have been working on an integration of TrajOpt into ROS and as a result I have had to get familiar with Bullet and how it compares to FCL. I noticed that bullet has the shape type ConvexShape and it turns out so does FCL which has better performance when collision checking.  Also the libraries used for distance queries for these libraries provide more accurate results for shape Type ConvexShape versus a generic mesh type. Also, I believe that collision checking speed would be improved since currently all geometries are being considered as generic mesh even though you may provide a mesh that is a convex hull. 

What is MoveIts thoughts on adding support for ConvexShape, I believe this could be exposed in the urdf by adding a new attribute to the mesh tag convex=true? I believe the type would be identical to the mesh type it would just get interpreted differently by set collision libraries.

Levi-Armstrong, 2018-01-16:
>Seeing as this would be mostly a MoveIt thing - and used by MoveIt internally - perhaps adding this to the SRDF or introducing a separate configuration file would be better. It could just reference links in the urdf - similar to how you can add Gazebo-specific properties using gazebo tags in urdf.
>
> Such a config file could then get its own tab in the MSA as well.

I like this approach.

Levi-Armstrong, 2018-01-16:
Actually you would like to tie it to the shape not necessarily the to the link. Would the proposed approach still work?

gavanderhoorn, 2018-01-16:
Well .. it could, but it would probably need more info than just the `link` name.

Multiple shapes for a single link is a bit iffy anyway, but it's probably something to support.

davetcoleman, 2018-01-17:
Utilizing convex shapes in MoveIt! is certainly a great speedup, I think this is a great idea as long as we still support non-convex by default so that we don't make MoveIt! even more complicated for beginners

v4hn, 2018-08-07:
@Levi-Armstrong Are you still interested in this?

I believe this is not actually such a big change, adding a tag for it to the srdf and maintaining one extra bit of information in the Link/Shape...
Did you look into this any further?

[Your recent blog post](https://rosindustrial.org/news/2018/7/5/optimization-motion-planning-with-tesseract-and-trajopt-for-industrial-applications) reads like this is the capital reason your project left out MoveIt, given that the collision checking API recently got reworked to address your needs.

gavanderhoorn, 2018-08-07:
I believe https://github.com/ros/urdfdom_headers/pull/24 is related.

davetcoleman, 2019-05-16:
I think @j-petit may be able to address and close this issue this summer re: https://github.com/ros-planning/moveit/issues/1427

felixvd, 2019-05-17:
Late comment, but this raised some questions for me:

- Would it make sense to assume that all user-added shapes are convex, so users get the performance out of the box? The SRDF tag could be set to `convex=false` for users who need precise collision checking.
- What happens if non-convex objects meshes are defined as ConvexShape? Are there checks for convexity in FCL/Bullet?

That aside, I'm in favor. Makes a lot of sense.

a-price, 2019-05-18:
One option might be to simply check if a mesh is convex when it's loaded, then dispatch either an `fcl::Convex` or the whatever the general mesh class is. This would eliminate the need to specify anything extra in the URDF/SRDF.

Checking for convexity is pretty straightforward, though someone should correct me if there's a better way. Let's assume we have a triangulated mesh from Assimp or whatever. Each face has a normal vector **n** facing outward. The halfspace representation for this face is (**n**, _d_), where _d_ is the distance from the origin to the plane; if we have a vertex **p** on the face, _d_=`dot`(**n**,**p**). Now, we just need to check that the distance to all vertices along **n** < _d_, so `dot`(**n**, **v**) < _d_. So for all _i_ in `faces` and for all _j_ in `vertices`, if `dot`(**n**_i_, **v**_j_) < _di_ then the mesh is convex. For robustness, we should probably do `dot`(**n**_i_, **v**_j_) - _di_ < epsilon.

a-price, 2019-05-31:
I spent a little time playing around with this, and ran into a small issue: with a triangulated mesh, there are many coplanar faces needed to represent a non-triangular face. FCL's `Convex` type supports this by allowing a vector of vertices to represent a polygonal face, rather than just a triangular one. So, when creating a `Convex`, these faces will likely need merging to produce the correct face representation.

Levi-Armstrong, 2019-06-01:
> I spent a little time playing around with this, and ran into a small issue: with a triangulated mesh, there are many coplanar faces needed to represent a non-triangular face. FCL's Convex type [...]

This is one reason why we have created a PR against urdf_dom and urdf_headers. Also all of the mesh parsing tools will have to be updated. Current when using existing mesh paring which leverage assimp, it forces triangulation and mergs all shapes into a single triangle list. I have custom mesh parsers in tesseract which is a modified version of the existing parsers to solve this issue which could be leverage to update the existing parsers. 

Also I am not sure why FCL requires the faces for convex shapes, because this is not required by Bullet. I have been meaning to ask the question of FCL.