moveit/moveit #743: "star" planners fail

AndyZe, 2018-01-18:
### Description

The 2 "star" planners I have tried are failing with the following:

> [ INFO] [1516294558.655058244]: Planner configuration 'arms[RRTstarkConfigDefault]' will use planner 'geometric::RRTstar'. Additional configuration parameters will be set when the planner is constructed.
Debug:   The value of parameter 'delay_collision_checking' is now: '1'
Debug:   The value of parameter 'goal_bias' is now: '0.050000000000000003'
Debug:   The value of parameter 'range' is now: '0'
Debug:   arms[RRTstarkConfigDefault]: Planner range detected to be 15.079645
Info:    arms[RRTstarkConfigDefault]: problem definition is not set, deferring setup completion...
[move_group-3] process has died [pid 12656, exit code -11, cmd /opt/ros/kinetic/lib/moveit_ros_move_group/move_group --debug __name:=move_group __log:=/home/nrgadmin/.ros/log/f2446fb2-fc61-11e7-b5c9-5ba35ae65981/move_group-3.log].
log file: /home/nrgadmin/.ros/log/f2446fb2-fc61-11e7-b5c9-5ba35ae65981/move_group-3*.log

### Your environment
* ROS Distro: Kinetic
* OS Version: Ubuntu 16.04
* Freshly-updated binary. `rosversion moveit_core` returns `0.9.11`

### Steps to reproduce
Launch a UR5 in MoveIt and select one of the "star" planners, (RRTstarkConfigDefault or LazyPRMstarkConfigDefault). Plan a move with the interactive marker interface. See MoveIt! crash in the terminal. Other planners I've spot-checked seem fine.

BryceStevenWilley, 2018-01-19:
Can duplicate on my machine. Running with `debug:=true` gives this stacktrace:
``` 0x00007fffd180a0fa in ompl::geometric::RRTstar::setup() () from /opt/ros/kinetic/lib/x86_64-linux-gnu/libompl.so.12
#1  0x00007fffd1d11308 in ?? () from /opt/ros/kinetic/lib/libmoveit_ompl_interface.so.0.9.11
#2  0x00007fffd1d11c96 in boost::detail::function::function_obj_invoker3<boost::_bi::bind_t<std::shared_ptr<ompl::base::Planner>, std::shared_ptr<ompl::base::Planner> (*)(std::shared_ptr<ompl::base::SpaceInformation> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, ompl_interface::ModelBasedPlanningContextSpecification const&), boost::_bi::list3<boost::arg<1>, boost::arg<2>, boost::arg<3> > >, std::shared_ptr<ompl::base::Planner>, std::shared_ptr<ompl::base::SpaceInformation> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, ompl_interface::ModelBasedPlanningContextSpecification const&>::invoke(boost::detail::function::function_buffer&, std::shared_ptr<ompl::base::SpaceInformation> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, ompl_interface::ModelBasedPlanningContextSpecification const&) () from /opt/ros/kinetic/lib/libmoveit_ompl_interface.so.0.9.11
#3  0x00007fffd1d39990 in std::_Function_handler<std::shared_ptr<ompl::base::Planner> (std::shared_ptr<ompl::base::SpaceInformation> const&), boost::_bi::bind_t<boost::_bi::unspecified, boost::function<std::shared_ptr<ompl::base::Planner> (std::shared_ptr<ompl::base::SpaceInformation> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, ompl_interface::ModelBasedPlanningContextSpecification const&)>, boost::_bi::list3<boost::arg<1>, boost::_bi::value<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, boost::_bi::value<ompl_interface::ModelBasedPlanningContextSpecification> > > >::_M_invoke(std::_Any_data const&, std::shared_ptr<ompl::base::SpaceInformation> const&) () from /opt/ros/kinetic/lib/libmoveit_ompl_interface.so.0.9.11
#4  0x00007fffd1902378 in ompl::geometric::SimpleSetup::setup() () from /opt/ros/kinetic/lib/x86_64-linux-gnu/libompl.so.12
#5  0x00007fffd1d350a9 in ompl_interface::ModelBasedPlanningContext::configure() () from /opt/ros/kinetic/lib/libmoveit_ompl_interface.so.0.9.11
#6  0x00007fffd1d0ff46 in ompl_interface::PlanningContextManager::getPlanningContext(std::shared_ptr<planning_scene::PlanningScene const> const&, moveit_msgs::MotionPlanRequest_<std::allocator<void> > const&, moveit_msgs::MoveItErrorCodes_<std::allocator<void> >&) const () from /opt/ros/kinetic/lib/libmoveit_ompl_interface.so.0.9.11
#7  0x00007fffd1cff348 in ompl_interface::OMPLInterface::getPlanningContext(std::shared_ptr<planning_scene::PlanningScene const> const&, moveit_msgs::MotionPlanRequest_<std::allocator<void> > const&, moveit_msgs::MoveItErrorCodes_<std::allocator<void> >&) const () from /opt/ros/kinetic/lib/libmoveit_ompl_interface.so.0.9.11
#8  0x00007fffd1f95c07 in ompl_interface::OMPLPlannerManager::getPlanningContext(std::shared_ptr<planning_scene::PlanningScene const> const&, moveit_msgs::MotionPlanRequest_<std::allocator<void> > const&, moveit_msgs::MoveItErrorCodes_<std::allocator<void> >&) const () from /opt/ros/kinetic/lib/libmoveit_ompl_planner_plugin.so
#9  0x00007ffff07edc74 in ?? () from /opt/ros/kinetic/lib/libmoveit_planning_request_adapter.so.0.9.11
#10 0x00007fffd0beabba in default_planner_request_adapters::FixStartStatePathConstraints::adaptAndPlan(boost::function<bool (std::shared_ptr<planning_scene::PlanningScene const> const&, moveit_msgs::MotionPlanRequest_<std::allocator<void> > const&, planning_interface::MotionPlanResponse&)> const&, std::shared_ptr<planning_scene::PlanningScene const> const&, moveit_msgs::MotionPlanRequest_<std::allocator<void> > const&, planning_interface::MotionPlanResponse&, std::vector<unsigned long, std::allocator<unsigned long> >&) const () from /opt/ros/kinetic/lib/libmoveit_default_planning_request_adapter_plugins.so
#11 0x00007ffff07edb2f in planning_request_adapter::PlanningRequestAdapter::adaptAndPlan(std::shared_ptr<planning_interface::PlannerManager> const&, std::shared_ptr<planning_scene::PlanningScene const> const&, moveit_msgs::MotionPlanRequest_<std::allocator<void> > const&, planning_interface::MotionPlanResponse&, std::vector<unsigned long, std::allocator<unsigned long> >&) const ()
   from /opt/ros/kinetic/lib/libmoveit_planning_request_adapter.so.0.9.11
#12 0x00007ffff07edd85 in ?? () from /opt/ros/kinetic/lib/libmoveit_planning_request_adapter.so.0.9.11
#13 0x00007fffd0be226e in default_planner_request_adapters::FixStartStateCollision::adaptAndPlan(boost::function<bool (std::shared_ptr<planning_scene::PlanningScene const> const&, moveit_msgs::MotionPlanRequest_<std::allocator<void> > const&, planning_interface::MotionPlanResponse&)> const&, std::shared_ptr<planning_scene::PlanningScene const> const&, moveit_msgs::MotionPlanRequest_<std::allocator<void> > const&, planning_interface::MotionPlanResponse&, std::vector<unsigned long, std::allocator<unsigned long> >&) const () from /opt/ros/kinetic/lib/libmoveit_default_planning_request_adapter_plugins.so
#14 0x00007ffff07edeb6 in ?? () from /opt/ros/kinetic/lib/libmoveit_planning_request_adapter.so.0.9.11
[... cut ...]
#26 0x00007fffd06e4202 in actionlib::SimpleActionServer<moveit_msgs::MoveGroupAction_<std::allocator<void> > >::executeLoop() () from /opt/ros/kinetic/lib/libmoveit_move_group_default_capabilities.so
#27 0x00007ffff27af5d5 in ?? () from /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.58.0
#28 0x00007ffff622a6ba in start_thread (arg=0x7fffb6ffd700) at pthread_create.c:333
#29 0x00007ffff57c341d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109
```

Currently trying to build OMPL and MoveIt from source to get more detailed info on where it's segfaulting.

BryceStevenWilley, 2018-01-20:
Interestingly enough, compiling in Debug mode makes the error go away, as well as adding print statements in RRTStar.setup, where the stacktrace above segfaults. This suggests some sort of threading/timing issue, possibly with the sampling thread?

In addition, this is almost identical to a previous issue (https://github.com/ros-planning/moveit/issues/274). This fix for this issue is present the latest version of OMPL in kinetic and from source, so it seems to be another distinct setup issue.

davetcoleman, 2018-03-19:
@mamoll has anything (off the top of your head) changed in OMPL recently that would cause this?

mamoll, 2018-03-21:
@gammell, can you have a look at this?

gammell, 2018-03-21:
@mamoll, I may be able to take a look in a week or two, but unfortunately not at the moment.

Are the "two star planners" that are failing RRT* and PRM*? I have never really looked at PRM* in OMPL, but I wouldn't expect it shares a lot of algorithm code with RRT*?... Could it be interfacing issues?

BryceStevenWilley, 2018-03-26:
Looked at this again and realized RRT* only crashes in the current release of MoveIt!/OMPL, but not in the latest OMPL version (as of Jan. 20th when I tested this), so it might be already fixed in OMPL. PRM* works for me in current version of OMPL too. @AndyZe, if you get the chance, can you follow the instructions [here](http://moveit.ros.org/install/source/dependencies/#open-motion-planning-library-ompl) to get the latest OMPL source and see if that fixes your issue?

enrico-g, 2018-05-17:
I had the same issue (RRT_star didn't run at all, while PRM_star was fine). My OMPL version turned out to be old, therefore I solved the problem by simply removing and reinstalling it (NOT from source!).

zkingston, 2018-05-21:
As @enrico-g suggests, the latest version of OMPL for kinetic (version `1.2.3`) from the package manager should have the fix that was pointed out by @jonbinney in #274 (https://github.com/ompl/ompl/commit/5efd0fa2e6dc0674f3bf0c1a6787920c546919a2, you can see the commit in the release here: https://github.com/ros-gbp/ompl-release/commit/479ec3e89e588c7198edaca1825c1dc83965b3a8#diff-57f4512e132fa498410cefd0c08d5545). It should be enough to just upgrade your OMPL (`apt-get update ros-kinetic-ompl`).

liuxin00738, 2018-12-13:
@davetcoleman @mamoll @zkingston 
Hi, This problem appears again on my system. Basically I use PRMstar, no problem and find one solution. But when I use RRTstar or RRT, it always fails to generate any solution. I tried other planners and they all seems fine and able to solve it.

I am using kinetic with moveit. The ros-kinetic-ompl has a package.xml with ompl version 1.2.3.
I run sudo apt-get update and sudo apt-get install ros-kinetic-ompl.
Any thoughts on this?

AndyZe, 2018-12-13:
I just updated/upgraded my packages and it seems like this issue is definitely solved. Closing it - thanks @BryceStevenWilley & @zkingston

closed 2018-12-13 by AndyZe