moveit/moveit #745: upgrade to tf2

dirk-thomas, 2018-01-19:
### Description

`tf` has been deprecated since ROS Hydro. As a consequence ROS 2 only supports `tf2`. Once MoveIt has been updated to use `tf2` we can consider working toward making it compatible with ROS 2.

Since the migration to `tf2` is also useful for ROS 1 this should happen in the ROS 1 code base *before* considering to support ROS 2.

### Expected behaviour
Remove all dependencies on `tf`. Only have dependencies on `tf2` instead.

dirk-thomas, 2018-01-19:
I am curious if this task is in the interest of the current MoveIt maintainers and if someone is willing to work on this groundwork?

gavanderhoorn, 2018-01-19:
I was curious what this would amount to, so did a quick check with `grep`.

Number of files:

```
$ grep -l -i 'tf::' -r /root/ws_moveit/src/moveit | sort | uniq | wc -l
57
```

Files:

```
$ grep -l -i 'tf::' -r /root/ws_moveit/src/moveit | sort | uniq
moveit_core/collision_detection/src/collision_tools.cpp
moveit_core/constraint_samplers/test/pr2_arm_kinematics_plugin.cpp
moveit_core/distance_field/src/distance_field.cpp
moveit_core/distance_field/test/test_distance_field.cpp
moveit_core/kinematic_constraints/src/kinematic_constraint.cpp
moveit_core/kinematic_constraints/test/test_constraints.cpp
moveit_core/planning_scene/src/planning_scene.cpp
moveit_core/robot_state/src/conversions.cpp
moveit_core/robot_state/src/robot_state.cpp
moveit_core/robot_trajectory/src/robot_trajectory.cpp
moveit_core/transforms/src/transforms.cpp
moveit_experimental/collision_distance_field/src/collision_common_distance_field.cpp
moveit_experimental/collision_distance_field/src/collision_robot_distance_field.cpp
moveit_experimental/kinematics_constraint_aware/src/kinematics_constraint_aware.cpp
moveit_kinematics/ikfast_kinematics_plugin/templates/ikfast61_moveit_plugin_template.cpp
moveit_kinematics/kdl_kinematics_plugin/src/kdl_kinematics_plugin.cpp
moveit_kinematics/lma_kinematics_plugin/src/lma_kinematics_plugin.cpp
moveit_planners/chomp/chomp_interface/include/chomp_interface/chomp_planning_context.h
moveit_planners/chomp/chomp_motion_planner/src/chomp_optimizer.cpp
moveit_planners/ompl/ompl_interface/src/ompl_planner.cpp
moveit_ros/benchmarks/benchmarks/src/benchmark_execution.cpp
moveit_ros/benchmarks/src/BenchmarkExecutor.cpp
moveit_ros/manipulation/pick_place/src/approach_and_translate_stage.cpp
moveit_ros/manipulation/pick_place/src/place.cpp
moveit_ros/manipulation/pick_place/src/reachable_valid_pose_filter.cpp
moveit_ros/move_group/src/default_capabilities/cartesian_path_service_capability.cpp
moveit_ros/move_group/src/default_capabilities/kinematics_service_capability.cpp
moveit_ros/move_group/src/move_group_capability.cpp
moveit_ros/move_group/src/move_group.cpp
moveit_ros/perception/depth_image_octomap_updater/include/moveit/depth_image_octomap_updater/depth_image_octomap_updater.h
moveit_ros/perception/depth_image_octomap_updater/src/depth_image_octomap_updater.cpp
moveit_ros/perception/mesh_filter/include/moveit/mesh_filter/transform_provider.h
moveit_ros/perception/mesh_filter/src/transform_provider.cpp
moveit_ros/perception/occupancy_map_monitor/include/moveit/occupancy_map_monitor/occupancy_map_monitor.h
moveit_ros/perception/occupancy_map_monitor/src/occupancy_map_monitor.cpp
moveit_ros/perception/occupancy_map_monitor/src/occupancy_map_server.cpp
moveit_ros/perception/pointcloud_octomap_updater/include/moveit/pointcloud_octomap_updater/pointcloud_octomap_updater.h
moveit_ros/perception/pointcloud_octomap_updater/src/pointcloud_octomap_updater.cpp
moveit_ros/perception/semantic_world/src/semantic_world.cpp
moveit_ros/planning_interface/common_planning_interface_objects/include/moveit/common_planning_interface_objects/common_objects.h
moveit_ros/planning_interface/common_planning_interface_objects/src/common_objects.cpp
moveit_ros/planning_interface/move_group_interface/include/moveit/move_group_interface/move_group_interface.h
moveit_ros/planning_interface/move_group_interface/src/move_group_interface.cpp
moveit_ros/planning_interface/move_group_interface/src/wrap_python_move_group.cpp
moveit_ros/planning/planning_components_tools/src/visualize_robot_collision_volume.cpp
moveit_ros/planning/planning_scene_monitor/include/moveit/planning_scene_monitor/current_state_monitor.h
moveit_ros/planning/planning_scene_monitor/include/moveit/planning_scene_monitor/planning_scene_monitor.h
moveit_ros/planning/planning_scene_monitor/src/current_state_monitor.cpp
moveit_ros/planning/planning_scene_monitor/src/planning_scene_monitor.cpp
moveit_ros/planning/trajectory_execution_manager/src/trajectory_execution_manager.cpp
moveit_ros/robot_interaction/include/moveit/robot_interaction/interaction_handler.h
moveit_ros/robot_interaction/src/interaction_handler.cpp
moveit_ros/robot_interaction/src/interactive_marker_helpers.cpp
moveit_ros/robot_interaction/src/robot_interaction.cpp
moveit_ros/visualization/motion_planning_rviz_plugin/src/motion_planning_frame_objects.cpp
moveit_ros/warehouse/warehouse/src/import_from_text.cpp
moveit_ros/warehouse/warehouse/src/save_to_warehouse.cpp
```

dirk-thomas, 2018-02-01:
@davetcoleman @rhaschke Since you guys seem to be the ones being active in the repo in the recent past maybe I can get a comment / opinion on this topic.

rhaschke, 2018-02-01:
@dirk-thomas Sorry for not being responsive. I think, we should switch to tf2 asap.
I will put this onto my agenda, but probably not before March...

dirk-thomas, 2018-02-01:
Thank you for the positive response. That would be great.

davetcoleman, 2018-02-14:
Agreed, I see nothing blocking this needed migration other than our time

amarsbars, 2018-02-27:
Hi, I'd like to help if possible.  I don't have a ton of experience, but I would like to learn more about MoveIt.  I am submitting my quals paper and presentation in mid march, so I should have time after that.

IanTheEngineer, 2018-03-21:
@dirk-thomas & @tfoote, I'm taking a look at this migration and wondering: are [`eigen_conversions`](https://github.com/ros/geometry/tree/indigo-devel/eigen_conversions) deprecated along with `tf`? 

Seems `moveit` is depending on a fair few of these conversions. As far as I can tell, there isn't always a one to one correlation with every function used in `eigen_conversions` in either [`tf2_geometry_msgs`](https://github.com/ros/geometry2/tree/indigo-devel/tf2_geometry_msgs), [`tf2_kdl`](https://github.com/ros/geometry2/tree/indigo-devel/tf2_kdl), or [`tf2_eigen`](https://github.com/ros/geometry2/tree/indigo-devel/tf2_eigen). Should `tf2` have corresponding functions for everything in the `*_conversions` packages of `tf` namespace?

tfoote, 2018-03-21:
Generically tf2 can support the conversions via the template logic. There may not be full coverage of every function, in particular because the twist logic had fundamental representation issues it was not ported. Adding in additional conversions as necessary would be appreciated.

IanTheEngineer, 2018-04-26:
@dirk-thomas there are currently two PR's that need feedback / merging before we can finish the tf2 migration for MoveIt:
https://github.com/ros-visualization/rviz/pull/1224
and
https://github.com/ros/geometry2/pull/292

I understand that we're nearing the Melodic release and ROS2 C-turtle release, which means your team must be swamped, but, in order to keep the ball rolling on this request, and get this MoveIt API change into Melodic, we need some feedback on those PR's.

IanTheEngineer, 2018-05-02:
Thanks for the pings @dirk-thomas. I implemented an tf2_ros::Buffer management / exposure change based on @tfoote comment ( https://github.com/ros-visualization/rviz/issues/1215#issuecomment-385124873 ) here: https://github.com/ros/geometry/pull/163

tfoote, 2018-05-03:
@IanTheEngineer I've released geometry and geometry2 with the updates you flagged as needed.

IanTheEngineer, 2018-05-03:
thanks for the release @tfoote. Now all the dependencies are unblocked for https://github.com/ros-planning/moveit/pull/830 to move forward

IanTheEngineer, 2018-05-18:
Resolved by https://github.com/ros-planning/moveit/pull/830

closed 2018-05-18 by IanTheEngineer