moveit/moveit #747: Missing Dependency for moveit_core on Lunar

allenh1, 2018-01-23:
### Description

This seems to be a trivial dependency issue. CMake just complains that the `moveit_resources` isn't
available when it tries to build `moveit_core`.

### Your environment
* ROS Distro: Lunar
* OS Version: Gentoo
* Source or Binary build? Source
* If source, which git commit or tag? release 0.9.11

### Steps to reproduce
Tell us how to reproduce this issue. Attempt to provide a working demo, perhaps using Docker.

```
docker run -ti --privileged --rm allenh1/ros_gentoo_base:latest bash -c 'emaint sync -r ros-overlay && emerge ros-lunar/moveit_core --jobs
```

### Expected behaviour
Build should pull in `moveit_resources` as it build `moveit_core`.

### Actual behaviour
`moveit_core` does not pull in `moveit_resources`, and the build fails.

### Backtrace or Console output

Gist file is [here](https://gist.github.com/allenh1/21e4bce7e00d2d2f3250c4f317e79ead).

jonbinney, 2018-01-23:
moveit_resources is already listed as a test_depend in moveit_core. I don't think it should be needed if you aren't running tests (could be a bug there though). I don't see the actual output of the cmake process in that gist; is it in the `/var/tmp/portage/ros-lunar/moveit_core-0.9.11/temp/build.log` file that is referenced in that log?

Looks like the cmakelists for those packages use `if(CATKIN_ENABLE_TESTING)` to not find_package moveit_resources - is it possible you have `CATKIN_ENABLE_TESTING` enabled for your build even though you aren't installing test_depends?

allenh1, 2018-01-23:
> Looks like the cmakelists for those packages use if(CATKIN_ENABLE_TESTING) to not find_package moveit_resources - is it possible you have CATKIN_ENABLE_TESTING enabled for your build even though you aren't installing test_depends?

Yes, that looks to be the case. My bad. I'll close this. Sorry for the noise!

closed 2018-01-23 by allenh1