moveit/moveit #749: Invalid Trajectory: start point deviates from current robot state

rkeatin3, 2018-01-23:
### Description
When planning/executing multiple plans in quick succession, execution often fails with the error `Invalid Trajectory: start point deviates from current robot state more than 0.01`.

### My environment
* ROS Distro: Indigo
* OS Version: Ubuntu 14.04
* MoveIt Version Binary Version 0.7.13

### Steps to reproduce
I believe the key to reproducing this error is planning/executing plans in quick succession; sleeping in between executing one trajectory and planning another mitigates this issue. I am also aggregating joint states from multiple publishers onto `/joint_states` by using the `source_list` parameter to the `joint_state_publisher`.

For more background, see my post on the MoveIt Users' group [here](https://groups.google.com/forum/#!topic/moveit-users/F6COChzgR7k).

I believe my issue is different than these similar issues: 

- [Issue 694](https://github.com/ros-planning/moveit/issues/694), [Issue 498](https://github.com/ros-planning/moveit/issues/498): Related to Cartesian planning
- [Issue 283](https://github.com/ros-planning/moveit/issues/283): Related to continuous rotation joints.

Based on my testing, I think the underlying cause may be linked to the following issues/PRs:
- [PR 232](https://github.com/ros-planning/moveit/pull/232)
- [PR 350](https://github.com/ros-planning/moveit/pull/350)
- [PR 298](https://github.com/ros-planning/moveit/pull/298)
- [Issue 501](https://github.com/ros-planning/moveit/issues/501)
- [PR 717](https://github.com/ros-planning/moveit/pull/717)
- [Issue 722](https://github.com/ros-planning/moveit/issues/722)

PRs 298, 717, and 350 have all been merged already. Issue 501 also references PR 717 as a possible fix. It was merged 9 days ago into `indigo_devel` according to the PR, but when I look at the branch itself it says the last commit was is 19 days ago...

Does the current binary release have PR 717 integrated?

rhaschke, 2018-01-24:
If you are using binary release 0.7.13, you don't yet have the latest merge of #717, because this is not yet released. Try to [build from source](http://moveit.ros.org/install/source). I'm closing this for now. Please feel free to reopen, if the issue remains in the source build. Thank you very much for your detailed report.

rkeatin3, 2018-01-24:
I think I am unable to reopen this issue since I'm not the one that closed it, but anyway, the source build of MoveIt (from commit with hash 4c9a74...) did not fix this issue.

rhaschke, 2018-01-25:
Hm. Indeed, the unittest that succeeds in Kinetic fails in Indigo. I need to have a deeper look what is missing in Indigo.

rhaschke, 2018-02-01:
Closed via #752

closed 2018-02-01 by rhaschke