moveit/moveit #750: Trajectory Discontinuities While Using Orientation Constraints

rkeatin3, 2018-01-23:
### Description

When using orientation constraints, the trajectory waypoints returned by MoveIt often have large discontinuities (typically a elbow, shouder, and/or wrist flip). In practice, this results in unsafe paths being executed (as the large discontinuities are filled in via interpolation by the controller without doing collision checking). 

This issue is definitely related to [Issue 562](https://github.com/ros-planning/moveit/issues/562), but that issue has been dead for a while. 

See more information from [this post](https://groups.google.com/forum/#!topic/moveit-users/_l6iYrzrrRc) on the MoveIt Users' group.

### Your environment
* ROS Distro: Indigo
* OS Version: Ubuntu 14.04
* MoveIt Version: Binary Version 0.7.13
* Planner: RRTConnectK
* Kinematics Solver: Trac-IK (Distance)

### Steps to reproduce:
Plan paths with an orientation constraint. It seems (and intuitively makes sense) that longer paths are more susceptible, especially whose straight-line Cartesian path passes near a singularity.

### Steps to mitigate:
- It was suggested the using Trac-IK with "Distance" `solve_type` would mitigate this issue to some extent, but I am still seeing the problem.
- When researching this issue more I found [PR 541](https://github.com/ros-planning/moveit/pull/541) and will give that a try.

Any other suggestions?

rkeatin3, 2018-01-24:
Using the parameter `enforce_joint_model_state_space` from PR 541 seems to help a bit. The paths are denser in general, and I haven't seen any shoulder/wrist/elbow flips. 

The planning often takes longer to execute, but more importantly, there are still discontinuities that can cause issues. I've attached an example of such a plan:

[image]

rhaschke, 2018-01-24:
As pointed out by @v4hn in https://github.com/ros-planning/moveit/pull/541#issuecomment-314417586, the IK solutions of consecutive sampling points should be distance-limited in joint space. Currently MoveIt's IK API doesn't provide an interface for this. @davetcoleman: I suggest to discuss this issue in the next maintainer meeting.

v4hn, 2018-01-25:
As noticed by the OP this is a duplicate of #562 no need for a new issue. I'll close this.

@rkeatin3 `enforce_joint_model_state_space` ensures that planning takes place in joint space, so there can't be any *discontinuities* anymore. At least not the same type of discontinuity you get without the parameter. If you mean to point out the sparse waypoints in the left part of your image, you might want to try decreasing `longest_valid_segment_fraction` in `ompl_planning.yaml`. This will slow down planning even more, but also plans with more waypoints internally (although path simplification might remove them afterwards).

@rhaschke Yes I believe we should provide a way to pass *optional* additional constraints to the IK plugin, but we shouldn't rely on the plugin to account for them. So it's only partially related to this issue.

As I wrote somewhere before, the actual problem with this issue is that neighboring states in the [`PoseModelStateSpace`](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_planners/ompl/ompl_interface/src/parameterization/work_space/pose_model_state_space.cpp) should not be close to each other if their joint space distance passes a threshold. There is [a parameter](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_planners/ompl/ompl_interface/src/parameterization/work_space/pose_model_state_space.cpp#L46) for this for interpolation already, but the last time I looked into this, this had no visible effect on the bug. I am not sure at all how changing the `distance` function would affect the different planners.

The code is rather convoluted but everyone is welcome to debug it.

closed 2018-01-25 by v4hn