moveit/moveit #751: Install Moveit from source

akashjinandra, 2018-01-23:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: |Kinetic]
* OS Version: Ubuntu 16.04
* Source kinetic-devel
Hello,

I'm trying to install moveit from source into my opt/ros/kinetic folder and I follow the directions on the website: http://moveit.ros.org/install/source/

I'm trying to do this because I have also pulled ompl from source and made some changes. So I've built the moveit in the ws_moveit folder and it all works great, but it doesn't seem to install all the proper files into the opt/ros/kinetic, it only does: moveit_core                   moveit_ros_planning
moveit_kinematics             moveit_ros_robot_interaction
moveit_msgs                   moveit_visual_tools
moveit_ros_perception        

It's missing the moveit_ros_manipulation packages and some others. It does have the proper moveit_ros_manipulationConfig.cmake files in the devel folder, but not in opt/kinetic/ros. 

Is there a step that I'm missing or not doing correctly?

Thanks,

Akash Jinandra

rhaschke, 2018-01-24:
The install directions don't aim for an `install` build, but only for a `devel` build. Hence, for installing, you need to configure catkin to do so like this:
`catkin config --extend /opt/ros/kinetic --install --install-space /opt/ros/kinetic --cmake-args -DCMAKE_BUILD_TYPE=Release`

See [catkin tools doc](https://catkin-tools.readthedocs.io/en/latest/verbs/catkin_config.html#installing-packages) for more details.

closed 2018-01-24 by rhaschke

akashjinandra, 2018-01-25:
Thanks so much

jih189, 2022-08-13:
This command should be in the readme page.

jih189, 2022-08-16:
I got warning:
WARNING: Your workspace is configured to explicitly
extend a workspace which yields a CMAKE_PREFIX_PATH which
is different from the cached CMAKE_PREFIX_PATH used last
time this workspace was built.

If you want to use a different CMAKE_PREFIX_PATH you
should call `catkin clean` to remove all references to
the previous CMAKE_PREFIX_PATH.

Cached CMAKE_PREFIX_PATH:
	/home/jiaming/cmake-install
Other workspace to extend:
	/opt/ros/melodic
Other workspace's CMAKE_PREFIX_PATH:
	/opt/ros/melodic:/home/jiaming/cmake-install
-----------------------
such warning causes me can not compile