moveit/moveit #757: multiple move_group nodes in different name spaces

smaassen, 2018-02-05:
Hi

Is it possible to run multiple move_group nodes under different name spaces and then specifiy in code which one to use? I want to control a robot that is running ros including a move_group and a high level interface for that (like go to pos x or move joint to y). Since it runs on a rasperzy pi its computational power is quite poor so i want to do the scene planing including object avoidance on another machine on the network that then should give the execution command to the robot. Therefore I launch a second move_group with planing scene context etc, in a different namespace (/planningNS)

I figured out how to send the trajectiry to the robot, but i have problems at generating the trajectory since in my code that runs on the "offline" machine i use the move group interface like this :

moveit::planning_interface::MoveGroupInterface::Options loadOptions("arm");

moveit::planning_interface::MoveGroupInterface move_group(loadOptions);
moveit::planning_interface::PlanningSceneInterface planning_scene_interface;

I could not find a way to specify which action servers should be taken for these instances (it takes the one from the /move_group node, but i want to take those of the /planningNS/move_group). 

The documentation does not say anything about different namespaces. Is it possible to add this feature?
Best

willcbaker, 2018-04-16:
#835

edetleon, 2020-05-11:
Hello,
I know this is old but did you figure how to edit namespaces and call the move_group nodes in code ?
I need to move 2 independant arms and I didn't find explanations about it.
Thanks,
Edith

v4hn, 2020-05-11:
You should be able to run `move_group.launch` twice with different namespaces.
Be sure that's really what you want to do though, because there is no way to relate the plans for the different pieces of hardware with multiple move-groups.
The `MoveGroupInterface` can be created with an `MGI::Options` structure. If you pass a different namespace for the contained nodehandle, you should be able to talk to the different nodes.

> After having digged into it a bit more i nailed down the problem. Basically no matter what, the move group interface creates the actionlib subcribers with the constant topics defined in the capabilities names header. In my eyes, this should be editable somehow and not hard coated into the header file.

You can remap them in the launch file. Topic names should almost never be parameters in ROS because remapping is always possible.

If you encounter any concrete issue or shortcoming in MoveIt while working on this, please file dedicated new issues.

closed 2020-05-11 by v4hn

edetleon, 2020-05-11:
Thank you v4hn. I read it was also possible to move several arms if they all are in the same planning group, but not individually with poses from the SRDF file, and that's what bothers me. I will dig into it to figure out what's the best solution.

v4hn, 2020-05-11:
> I read it was also possible to move several arms if they all are in the same planning group,
> but not individually with poses from the SRDF file, and that's what bothers me.

There is definitely ways in roughly 10 lines of code to do so,
but there is no trivial API for setting groups from named states of other groups.