moveit/moveit #759: catkin_make error

wheemzy1, 2018-02-08:
### Description

I'm trying to build/install a custom package named moveo_ros, which is based on moveit framework.

When I try to build it I receive an error. Seems like a problem with a data type conversion of  a MoveItErrorCode object in the move_group_interface_coor.cpp file. 

I am relatively new to ROS and not well versed in reading the reference materials so I need a little nudge. 

### Your environment
  ROS Distro: Kinetic 
  Ubuntu 16.04

### Steps to reproduce

Created a catkin workspace, placed the moveo_ros packages in src directory. Used catkin_build install which reached 90% then failed to finish. 

### Backtrace or Console output

matthew@matthew-System-Product-Name:~/moveo_ws$ catkin_make install
Base path: /home/matthew/moveo_ws
Source space: /home/matthew/moveo_ws/src
Build space: /home/matthew/moveo_ws/build
Devel space: /home/matthew/moveo_ws/devel
Install space: /home/matthew/moveo_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/matthew/moveo_ws/build"
####
####
#### Running command: "make install -j4 -l4" in "/home/matthew/moveo_ws/build"
####
Scanning dependencies of target move_group_1
[  0%] Built target std_msgs_generate_messages_py
[ 18%] Built target moveit_convert
[ 18%] Built target sensor_msgs_generate_messages_py
[ 18%] Built target _moveo_moveit_generate_messages_check_deps_ArmJointState
[ 18%] Built target sensor_msgs_generate_messages_lisp
[ 27%] Building CXX object moveo_moveit/CMakeFiles/move_group_1.dir/src/move_group_interface_coor_1.cpp.o
[ 27%] Built target std_msgs_generate_messages_nodejs
[ 27%] Built target std_msgs_generate_messages_lisp
[ 27%] Built target sensor_msgs_generate_messages_nodejs
[ 27%] Built target std_msgs_generate_messages_eus
[ 27%] Built target std_msgs_generate_messages_cpp
[ 27%] Built target sensor_msgs_generate_messages_cpp
[ 27%] Built target sensor_msgs_generate_messages_eus
[ 36%] Built target moveo_moveit_generate_messages_lisp
[ 54%] Built target moveo_moveit_generate_messages_py
[ 63%] Built target moveo_moveit_generate_messages_nodejs
[ 72%] Built target moveo_moveit_generate_messages_cpp
[ 90%] Built target moveo_moveit_generate_messages_eus
[ 90%] Built target moveo_moveit_generate_messages
/home/matthew/moveo_ws/src/moveo_moveit/src/move_group_interface_coor_1.cpp: In function ‘int main(int, char**)’:
/home/matthew/moveo_ws/src/moveo_moveit/src/move_group_interface_coor_1.cpp:126:41: error: cannot convert ‘moveit::planning_interface::MoveItErrorCode’ to ‘bool’ in initialization
   bool success = move_group.plan(my_plan);
                                         ^
moveo_moveit/CMakeFiles/move_group_1.dir/build.make:62: recipe for target 'moveo_moveit/CMakeFiles/move_group_1.dir/src/move_group_interface_coor_1.cpp.o' failed
make[2]: *** [moveo_moveit/CMakeFiles/move_group_1.dir/src/move_group_interface_coor_1.cpp.o] Error 1
CMakeFiles/Makefile2:2806: recipe for target 'moveo_moveit/CMakeFiles/move_group_1.dir/all' failed
make[1]: *** [moveo_moveit/CMakeFiles/move_group_1.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make install -j4 -l4" failed

BryceStevenWilley, 2018-02-08:
I believe that is because of [this PR](https://github.com/ros-planning/moveit/pull/670), which (if I understand it correctly) disallows implicit conversions from the MoveItErrorCode to booleans. I've also had some troubles with it. It's an easy fix however:
The current lines in moveo
```
// move_group_interface_coor_1.cpp:126
bool success = move_group.plan(my_plan);

ROS_INFO_NAMED("moveo", "Visualizing plan 1 (pose goal) %s", success ? "" : "FAILED");
```
would become
```
// You can use auto too if it's c++11. 
moveit::planning_interface::MoveItErrorCode success = move_group.plan(my_plan);

ROS_INFO_NAMED("moveo", "Visualizing plan 1 (pose goal) %s", success == moveit_msgs::MoveItErrorCodes::SUCCESS ? "" : "FAILED");
```

wheemzy1, 2018-02-08:
That did it. Thank you dude!

BryceStevenWilley, 2018-02-08:
No problem. Since everything is solved, you should close this issue: it helps the maintainers know at a glance that your problem got solved.

wheemzy1, 2018-02-08:
You got it boss.

closed 2018-02-08 by wheemzy1