moveit/moveit #764: Porting enhancements from ur_kinematics to the ikfast template

mlautman, 2018-02-13:
### Description

This is more of a proposed enhancement than it is an issue. It is also a duplicate of [this](https://github.com/ros-industrial/universal_robot/issues/341)

I have been working on porting some features available in the ur_kinematics into the ikfast template so that all robots can benefit from the enhancements @kphawkins and the other contributors built into the ur_kinematics solver. It seems pretty clear that ur_moveit_plugin.cpp was heavily influenced by the ikfast generated templates. My question is beyond the obvious, what would be involved in porting the enhancements back into the ikfast template? I'm particularly interested in supporting the ability to use the robot arm as a subset of a kinematic chain but additional features like setRedundantJoints are interesting as well.

### Your environment
* ROS Distro: [Kinetic]
* OS Version: Ubuntu 16.04
* Source build
* If binary, which release version?
* If source, which git commit or tag?
 kinetic-devel

davetcoleman, 2018-02-14:
> what would be involved in porting the enhancements back into the ikfast template? 

It seems like you are the best fit to answer this question given that you've looked into it already

> I'm particularly interested in supporting the ability to use the robot arm as a subset of a kinematic chain

Could you define this better?

> additional features like setRedundantJoints are interesting as well.

Does this mean passive joints?

gavanderhoorn, 2018-02-14:
I'm with Dave: it seems you have ideas about what you are after.