moveit/moveit #770: CHOMP: Optimizer not initializing

angadhn, 2018-02-15:
### Description
Hi folks,

I am following the instructions to test out CHOMP based on the interface tutorial on moveit docs. However, I get the following error:
```bash
[ERROR] [1518709837.241774873]: Could not initialize optimizer
```
Any suggestions on getting past this would be much appreciated.

### Your environment
* Kinetic
* Ubuntu 16.04
* Source build: kinetic-devel

### Steps to reproduce
1. cloned the kinetic-devel branch in ```catkin_ws/src``` by:
```git clone https://github.com/ros-planning/moveit.git```.
2. built with ```catkin_make```. 
3. ```roslaunch moveit_resources demo_chomp.launch``` launches RViz with the motion planning plugin.
4. Under the planning tab, I select a 'random valid' goal state and then 'Update' followd by Plan.

### Expected behaviour
A successful motion execute by the robot: Robot moves to valid goal state based on the plan generated by CHOMP.

### Actual behaviour
Planner fails and I get the message that the optimizer could not be initialized.

### Backtrace or Console output
```
started roslaunch server http://alienware2:37799/

SUMMARY
========

PARAMETERS
 * /joint_state_publisher/use_gui: False
 * /move_group/allow_trajectory_execution: True
 * /move_group/capabilities: move_group/MoveGr...
 * /move_group/controller_list: [{'joints': ['joi...
 * /move_group/jiggle_fraction: 0.05
 * /move_group/max_range: 5.0
 * /move_group/max_safe_path_cost: 1
 * /move_group/moveit_controller_manager: moveit_fake_contr...
 * /move_group/moveit_manage_controllers: True
 * /move_group/octomap_resolution: 0.025
 * /move_group/planning_plugin: chomp_interface/C...
 * /move_group/planning_scene_monitor/publish_geometry_updates: True
 * /move_group/planning_scene_monitor/publish_planning_scene: True
 * /move_group/planning_scene_monitor/publish_state_updates: True
 * /move_group/planning_scene_monitor/publish_transforms_updates: True
 * /move_group/start_state_max_bounds_error: 0.1
 * /move_group/trajectory_execution/allowed_execution_duration_scaling: 1.2
 * /move_group/trajectory_execution/allowed_goal_duration_margin: 0.5
 * /robot_description: <?xml version="1....
 * /robot_description_kinematics/manipulator/kinematics_solver: kdl_kinematics_pl...
 * /robot_description_kinematics/manipulator/kinematics_solver_attempts: 3
 * /robot_description_kinematics/manipulator/kinematics_solver_search_resolution: 0.005
 * /robot_description_kinematics/manipulator/kinematics_solver_timeout: 0.005
 * /robot_description_planning/joint_limits/joint_1/has_acceleration_limits: True
 * /robot_description_planning/joint_limits/joint_1/has_velocity_limits: True
 * /robot_description_planning/joint_limits/joint_1/max_acceleration: 0.734
 * /robot_description_planning/joint_limits/joint_1/max_velocity: 3.67
 * /robot_description_planning/joint_limits/joint_2/has_acceleration_limits: True
 * /robot_description_planning/joint_limits/joint_2/has_velocity_limits: True
 * /robot_description_planning/joint_limits/joint_2/max_acceleration: 0.664
 * /robot_description_planning/joint_limits/joint_2/max_velocity: 3.32
 * /robot_description_planning/joint_limits/joint_3/has_acceleration_limits: True
 * /robot_description_planning/joint_limits/joint_3/has_velocity_limits: True
 * /robot_description_planning/joint_limits/joint_3/max_acceleration: 0.734
 * /robot_description_planning/joint_limits/joint_3/max_velocity: 3.67
[... 99 lines cut ...]
[ INFO] [1518711513.750098155]: Loading robot model 'fanuc'...
[ INFO] [1518711513.844775686]: Starting scene monitor
[ INFO] [1518711513.848521576]: Listening to '/move_group/monitored_planning_scene'
[ INFO] [1518711513.918323640]: Constructing new MoveGroup connection for group 'manipulator' in namespace ''
[ WARN] [1518711514.868835457]: 
Deprecation warning: Trajectory execution service is deprecated (was replaced by an action).
Replace 'MoveGroupExecuteService' with 'MoveGroupExecuteTrajectoryAction' in move_group.launch
[ INFO] [1518711514.881028235]: Ready to take commands for planning group manipulator.
[ INFO] [1518711514.881208561]: Looking around: no
[ INFO] [1518711514.881296407]: Replanning: no
[ INFO] [1518711673.516395573]: Planning request received for MoveGroup action. Forwarding to planning pipeline.
[ INFO] [1518711673.521698036]: Setting joint joint_1 to position -0.381429
[ INFO] [1518711673.521805199]: Setting joint joint_2 to position -0.957978
[ INFO] [1518711673.521862023]: Setting joint joint_3 to position -0.889114
[ INFO] [1518711673.521952940]: Setting joint joint_4 to position 0.783808
[ INFO] [1518711673.522011248]: Setting joint joint_5 to position -1.56329
[ INFO] [1518711673.522067221]: Setting joint joint_6 to position -3.64563
[ INFO] [1518711673.524336845]: The following collision detectors are active in the planning scene.
[ INFO] [1518711673.524401001]: FCL
[ INFO] [1518711673.524464391]: Active collision detector is: FCL
[ WARN] [1518711673.524515991]: Could not initialize hybrid collision world from planning scene
[ERROR] [1518711673.524567754]: Could not initialize optimizer
[ WARN] [1518711673.552247080]: Fail: ABORTED: No motion plan found. No execution attempted.
```
Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

bmagyar, 2018-02-15:
This may have been my last PR. I'll check this weekend!

Use gist.github.com to copy-paste the console output or segfault backtrace
using gdb.

angadhn, 2018-02-15:
No worries; this is just an issue with the demo setup; I got the motion planner working with my UR5 no problem. However, I need to work out why the executed trajectory of the real robot is so jerky when compared to the plan shown on rviz. While I don't think this is necessarily a CHOMP issue, the execution does seem smoother with OMPL planners. I'll have to take a look at it tomorrow morning but any suggestions are welcome :)

bmagyar, 2018-02-15:
UR5-s have limits on velocities and accelerations enforced by the control
box. That may be one thing.

Out of curiosity: could you post here some videos of the jerky vs normal
trajectories please?

You could also compare the trajectory sampling rate for the 2 different
planners, CHOMP may do a higher rate sampling which'd force more aggressive
behaviours on the controller.

angadhn, 2018-02-16:
Will do it once I'm back on lab tomorrow. The first time we ran CHOMP, the UR jerked pretty hard and it stopped itself via the mechanism of the polyscope software (I imagine). So we then re-scaled the velocities (and accelerations) to 0.01 from within rviz because the first time, but the motion was really slow yet jerky and eventually timed itself out. There was a whole new message on the pendant that I hadn't seen in my last 4 months with the robot; I'll try to post shots of all of this tomorrow once I am back on lab. Just unwinding now after a fairly long day :)

angadhn, 2018-02-21:
@bmagyar Apologies for the tardy response; somehow my super old phone flubbed up the recording and the motion was aggressive enough that we didn't have the courage to try to attempt the move again. However, I did follow thorough on the PR that @gavanderhoorn shared and presto things are smooth again. Thanks both!

closed 2018-02-21 by angadhn