moveit/moveit #775: improve association of subgroup IK solvers in meta groups

rhaschke, 2018-02-17:
Following up on the discussion in https://github.com/ros-planning/moveit/pull/769#discussion_r168921872: I think MoveIt has weird behavior when assigning IK solvers of subgroups to a meta group.

I had the following situation: Using a bimanual robot, e.g. a Pepper robot, I defined planning groups `left_arm` and `right_arm` for the arms and a meta group `both_arms`, comprising the former ones.

If the arm groups are disjoint, the meta group correctly inherits IK solvers for both arm groups.
However, if the groups are overlapping (in my case they shared the torso, which had a fixed parent joint), only one arm's IK solver is found/inherited. There was no warning about disjoint sets either.

As suggested by @v4hn, in case of overlapping JMGs neither of them should be inherited and a warning should be issued. 
If - as in my case - the overlap only involves fixed joints at the "root" of the tree(s), I suggest that we still could assign both IK solvers without any message. This should be perfectly OK.

To reproduce the issue: rviz shows interactive markers for all subgroups with associated IK solvers.

v4hn, 2018-02-17:
This is not too much of a problem in practice, but that's partially because users will setup their groups to work with MoveIt as-is...

I don't think fixed joints should be considered at all when selecting the set of relevant solvers for a group.

Additionally, if a meta-group is setup and no solver can be found for some active joints of that group, that should trigger a warning. This would be the case at the moment with your pepper-setup.
Personally I would even yield an error and ignore the group definition, but maybe we don't want to enforce this for backward compatibility.

rhaschke, 2018-02-17:
It's absolutely fine to have no solvers for some active joints in a group. Simply consider the z-lift joint of PR2 for which you barely find an IK solver out-of-the-box. Nevertheless the group is well-defined without an IK solver for it and it can be used for sampling based planning.

v4hn, 2018-02-17:
> It's absolutely fine to have no solvers for some active joints in a group

I tend to disagree, but yes, you could still do sampling-based planning with the group.
That's probably enough to not trigger a warning.

So I guess the warning we could issue looks something like this then:

    Multiple overlapping kinematics solvers found for group 'xyz'.
    Selected solver 'ABC' and ignoring the rest.

> Simply consider the z-lift joint of PR2 for which you barely find an IK solver out-of-the-box

Incidentally https://github.com/TAMS-Group/bio_ik handles this quite well out-of-the box. :)

rhaschke, 2018-02-17:
> Incidentally https://github.com/TAMS-Group/bio_ik handles this quite well out-of-the box.

Interesting link! Thanks for pointing this out. I will need to have a look.
However, not related to the issue at hand. BioIK seems to provide an all-in-one solution instead of relaying to existing solvers in a "modular" fashion.

v4hn, 2018-02-21:
> BioIK seems to provide an all-in-one solution instead of relaying to existing solvers in a "modular" fashion.

Well, it provides a general way to account for arbitrary groups with many bells and whistles.
Whether to factorize the model or not is left to the user and depends on the application.