moveit/moveit #776: MoveIt python interfaces

bmagyar, 2018-02-17:
This is more of a question. 

I'm using the Python interfaces for moveit for a paper and had to patch up a few things. 

Could someone please confirm if the intended/last known state of these are matching what I'm experiencing:

PlanningSceneInterface generates the right messages but only publishes them on a topic while MoveIt currently only support setting those by a service call. (I have a fix for this) 

MoveitCommander.plan only returns a subset of the equivalent C++ call. Planning time and the error code could be returned along with the trajectory in a tuple. (I have a fix for this) 

Should I add these and perhaps other things? Let me know if other things need patching in the Python  interface side. 

Setup:
ROS Kinetic or above

rhaschke, 2018-02-17:
> PlanningSceneInterface generates the right messages but only publishes them on a topic while MoveIt currently only support setting those by a service call.

I guess, @v4hn only considered the C++ interface when moving from asynchronous (topic) to synchronous (service) API, because he doesn't use the python interface (afaik).

> MoveitCommander.plan only returns a subset of the equivalent C++ call. Planning time and the error code could be returned along with the trajectory in a tuple. (I have a fix for this)

PRs are very welcome ;-)

bmagyar, 2018-02-19:
Thanks, I'll add those changes and compliment them with a few basic tests
in around 2weeks. (Got iros deadlines...)

v4hn, 2018-02-21:
PlanningSceneInterface generates the right messages but only publishes them on a topic while MoveIt currently only support setting those by a service call.

    I guess, @v4hn only considered the C++ interface when moving from asynchronous (topic) to synchronous (service) API, because he doesn't use the python interface (afaik).

MoveIt still very much supports planning scene updates via topics! I only added a small capability to *additionally* allow for service updates.
And yes, I never used the Python interfaces myself (partly because I believe it should be rewritten), so I did not add appropriate synchronous ways to update there.

> MoveitCommander.plan only returns a subset of the equivalent C++ call. Planning time and the error code could be returned along with the trajectory in a tuple. (I have a fix for this)

I guess this will break Python API then? :-/

PR welcome as @rhaschke says.