moveit/moveit #780: Setup assistant does not show collision geometries

v4hn, 2018-02-21:
The preview robot state in the moveit setup assistant shows the visual meshes attached to links.
This is all good and nice for many of the dialogs (setting up the planning groups, poses, etc),
but it is not a good idea when the user tries to setup the allowed collision matrix for their models.
In this dialog, at least optionally, we should visualize the collision geometry instead.

This is especially important if you have "invisible walls" in your description - fixed links that have a collision geometry, but don't have a visual geometry associated. Currently they cannot be highlighted.

rhaschke, 2018-02-21:
I could take care of this somewhere in April.

v4hn, 2020-09-03:
It's somewhat frustrating to find issues I wrote two years ago about the same thing after I just wrote a new issue...
Anyway, here's my alternative proposal for a fix as written in #2291 :

> The "Self-Collisions" tab in the MoveIt Setup Assistant highlights the pair of relevant links in the 3d visualization when a specific collision pair is inspected/clicked on.
> Highlighting works by changing the color of the visual geometry of the link. But in many situations, people add geometry to the robot description that consists of collision geometry only and highlighting will not show anything for the respective link.
> Instead, these cases should be detected and the collision geometry should be shown in the highlight-color.

PR #2352 (2020-10-06): MSA: Allow to show both, visual and collision geometry
Fixes #780. Note, that this builds on top of #2349!
closed 2020-10-06 by rhaschke