moveit/moveit #783: Buggy behavior caused by magic numbers in computeCartesianPath

mlautman, 2018-03-01:
If you use computeCartesianPath with [Eigen::Affine3d](https://github.com/ros-planning/moveit/blob/61e2e304d2bb537784042ef96f510baa3610d865/moveit_core/robot_state/src/robot_state.cpp#L1903-L1955) you get different results than if you pass in an [EigenSTL::vector_Affine3d](https://github.com/ros-planning/moveit/blob/61e2e304d2bb537784042ef96f510baa3610d865/moveit_core/robot_state/src/robot_state.cpp#L1985-L2025) with a single point. The underlying issue is the magic +5 [here](https://github.com/ros-planning/moveit/blob/61e2e304d2bb537784042ef96f510baa3610d865/moveit_core/robot_state/src/robot_state.cpp#L1925) If you call computeCartesianPath with a EigenSTL::vector_Affine3d with a single point in it, the magic +5 doesn't happen and you can get paths returned with two waypoints which makes the testJointSpaceJump function meaningless. This is semi-related to [#773](https://github.com/ros-planning/moveit/issues/773) but different enough that it made sense to me to open a separate issue.

davetcoleman, 2018-03-01:
This makes sense

davetcoleman, 2018-05-23:
Is this still an issue @mlautman ?

closed 2018-05-23 by mlautman