moveit/moveit #786: Robot model is not displayed when loading a urdf file using the moveit setup assistant

martine1406, 2018-03-02:
### Description
Hi guys

I tried loading the urdf (xacro format) file from my robot. It shows the loading was successful, but I do not see the rviz model of the robot in the viewing window (right side stays black).

I first tested by installing ros-indigo-moveit, then I remembered that I was able to load that same model in the moveit setup assistant in May 2017. I therefore removed ros-indigo-moveit, and cloned this moveit source repo. Then I tried to checkout to some commits along the way, and I could identify that commit e9e069d [https://github.com/ros-planning/moveit/commit/e9e069d3b98e576edd4f127ead2d558150eb9894](url) was where the issue was introduced. 

Is it possible to revert some changes that were made in this commit? Or else, what's the trick to load the urdf model and get the rviz visualisation working?

Thanks a lot
Martine

Overview of your issue here.

### Your environment
* ROS Distro: Indigo
* OS Version: e.g. Ubuntu 14.04
* Source or Binary build? both source and binary built version tried

mbharatheesha, 2018-03-02:
I had a similar problem, although with Rviz not being able to display the model despite loading the robot model successfully to the parameter server. Turned out, the issue was the locale setting, because I used a non-English locale setting. I am not sure if this is also the case with you.  
If you type ``echo $LC_NUMERIC`` in the terminal and if it returns anything other than en_US.UTF-8, then that could be a problem. You can change that by ``export LC_NUMERIC=en_US.UTF-8``.

martine1406, 2018-03-02:
hi 

thanks @mbharatheesha for the suggestion. did not work for me though . did export LC_NUMERIC=en_US.UTF-8, then tried loading my robot model with the latest binary built version of moveit I could get (got it from apt-get install ros-indigo-moveit) and I still have the same issue

ozzdemir, 2018-04-09:
Hi,
@martine1406 did you solve your problem? I also have the same issue with setup assistant.

martine1406, 2018-04-09:
Hi guys

I did not push the investigation further unfortunately. Let me know if you find any solution.

v4hn, 2018-04-10:
I am not in my home lab for a month so I can't test this at the moment, sorry.

varunvp, 2018-04-16:
I'm having the same issue. The fix above didn't work for me.:-/

rhaschke, 2018-04-16:
Can you please provide more details: 
- [ ] Which MoveIt version you are running? Which robot model do you want to load?
- [ ] Please, try to load a default model provided with MoveIt: `roslaunch moveit_resources demo.launch`
If this works (i.e. shows a robot in rviz), it's probably an issue with your robot description.
- [ ] Does rviz' `RobotModel` display work?

martine1406, 2018-04-17:
okay we are getting to something

 **Which MoveIt version you are running? Which robot model do you want to load?**
Well I use the binary install I get when doing 'sudo apt-get install ros-indigo-moveit', so I expect the latest available moveit version for ros Indigo? After doing  'sudo apt-get install ros-indigo-moveit', I got installation message such as Setting up ros-indigo-moveit (0.7.13-0trusty-20180326-123655-0800)
other than that:
 * /rosdistro: indigo
 * /rosversion: 1.11.21

I am trying to open the jaco2 model from the kinova-ros repo (https://github.com/Kinovarobotics/kinova-ros). The urdf for this robot is located here: https://github.com/Kinovarobotics/kinova-ros/blob/master/kinova_description/urdf/j2n6s300_standalone.xacro

**Please, try to load a default model provided with MoveIt: roslaunch moveit_resources demo.launch
If this works (i.e. shows a robot in rviz), it's probably an issue with your robot description.**
roslaunch moveit_resources demo.launch works. However, if I try to launch the moveit_setup_assistant, then create a new moveit configuration package, then load the fanuc robot urdf file (part of moveit_resources, on my computer located here: /opt/ros/indigo/share/moveit_resources/fanuc_description/urdf), it does the same result as for the jaco2 robot. 

 **Does rviz' RobotModel display work?**
when I roslaunch the moveit_setup_assistant, then load the urdf model of the robot I want to work with, then open Rviz, the rviz' RobotModel display work does not work properly. In the jaco2 case, it tells me that there is no transform between root and world. In the fanuc case, it tells me there is no root frame. In both cases, if I rosrun joint_state_publisher joint_state_publisher and rosrun robot_state_publisher robot_state_publisher, the display in rviz becomes fine. In the fanuc case, I also have to change the fixed frame from root to base. 

 **Are there any errors reported on the console?**
In the moveit_setup_assistant console, you have this: 
[ INFO] [1523968853.211934261]: Loaded j2n6s300 robot model.
[ INFO] [1523968853.212108872]: Setting Param Server with Robot Description
[ INFO] [1523968853.224460302]: Robot semantic model successfully loaded.
[ INFO] [1523968853.224504388]: Setting Param Server with Robot Semantic Description
[ INFO] [1523968853.242404264]: Loading robot model 'j2n6s300'...
[ INFO] [1523968853.242462804]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ INFO] [1523968854.251044917]: Stereo is NOT SUPPORTED
[ INFO] [1523968854.251239355]: OpenGl version: 2.1 (GLSL 1.2).
[ INFO] [1523968854.417290146]: Loading Setup Assistant Complete
[ INFO] [1523968854.436416385]: Loading robot model 'j2n6s300'...
[ INFO] [1523968854.436540277]: No root/virtual joint specified in SRDF. Assuming fixed joint

Thanks a lot for your help
Martine

Hymwgk, 2018-04-18:
I'm having this issue  too......

Hymwgk, 2018-04-18:
This problem may be caused by the virtual joints, just ignore it and add some virtual joints in the third step ,view the result in the  setup assistant

Hymwgk, 2018-04-18:
I mean    the third step in the left navigation pane of the setup assistant, sorry my English is poor

rhaschke, 2018-04-18:
After setting up an Ubuntu Trusty with Indigo, I can confirm this bug. I filed a simple fix in #846.

closed 2018-04-18 by rhaschke

Hymwgk, 2018-04-19:
Hi，@rhaschke   I'm a beginner to ros, I find that there is no" moveit_ros/visualization/robot_state_rviz_plugin/src/robot_state_display.cpp" file  in my computer , is that nomal?    Thank you very much !

martine1406, 2018-04-19:
Yes! I confirm this fix from @rhaschke works! Good job there :-)

rokaku, 2018-07-21:
Hello There, 
May I know how to apply the fix for current indigo release? I try to use `apt-get update `but the problem still exists. Thanks.

rokaku, 2018-07-21:
Hello @rhaschke, may I know how to apply the fix into current indigo release? I try to use `apt-get update` but there is nothing upgraded. and the problem still exists. Or I have to build it from the source? Thanks.