moveit/moveit #800: Unable to set end-effector link and There is no end-effector to get pose of in Pyhton API

nazir-ust, 2018-03-15:
### Description

I am using two UR10 arms (named 'hong' and 'kong') to perform simultaneous manipulation. My SRDF looks something like this (although I have tried different combinations for placing end-effector link inside the groups):

  <group name="hong_arm">
      <chain base_link="hong_base_link" tip_link="hong_ee_link" />
    </group>
    <group name="kong_arm">
      <chain base_link="kong_base_link" tip_link="kong_ee_link" />
    </group>
    <group name="dual_arm">
        <group name="hong_arm" />
        <group name="kong_arm" />
    </group>

    
    <end_effector name="hong_eff" parent_link="hong_ee_link" group="hong_arm" />
    <end_effector name="kong_eff" parent_link="kong_ee_link" group="kong_arm" />

The problem is that on RVIZ I can move the two arms simulatanously, but when it comes to moving through Python API, the 'group.get_end_effector_link' returns an empty string whereas 'group.set_end_effector_link' returns 'unable to set end effector link'. Needless to say, the current effector pose function also returns an error. 

And also, the current joint values for one arm are fine, but for the other arm are zero. Although sometimes both are fine. 

I am just curious as to why this works fine on RVIZ but not through Python API. 

### Your environment
* ROS Distro: [Kinetic]
* OS Version: e.g. Ubuntu 16.04
* Binary

### Expected behaviour
The group 'dual_arm' should register seamlessly.

### Actual behaviour
Works fine on RVIZ but the group doesn't seem to register properly with python API. Joint values are at times distorted. Effector links are not available for setting or for obtaining pose.

bmagyar, 2018-03-15:
Could you please provide an example we could use? The most minimal would be the best, I'd like to craft a test for this.

The Python API has been a little ignored lately, there may be a few features that are challenged by a dual arm setup.

nazir-ust, 2018-03-16:
Thanks. Btw, have the same issue using command line tool. 

I don't exactly know what would you like to see but my MoveIt configuration package is [here](https://github.com/nazir-ust/ur10_dual_arm_gazebo_moveit_config.git). 
The python code I am using is [here]( https://github.com/nazir-ust/ur10_cm/blob/master/ros_nodes/hong_kong_dual_arm_manipulation_test.py#L2 ).
And the robot execution launch file is [here](https://github.com/nazir-ust/ur10_cm/blob/master/launch/dual_arm_execution.launch)

nazir-ust, 2018-03-16:
Do you think it could be because I am using two different namespaces for the same robot description? I mean my `joint_states` topic publish joint state data for each arm at two different instances of time (Figure). 
[image]

Whereas my dual_arm group current state is more of a combined form (a single array for all joint values). I think I will try to squash the two joint states together.

nazir-ust, 2018-03-16:
`joint_states` issue seems resolved after I used joint_state_publisher(with source param) and robot_state_publisher outside the two namespaces of my arms. The end effector issue still persists.
Any help would be great! Thanks!

nazir-ust, 2018-03-16:
I think my temporary workaround would be to plan in joint space. Do you think is there any way in moveit I can set a target pose for an arm and obtain its corresponding joint values through ik?

bmagyar, 2018-03-18:
Setting from IK still requires end effector links to be available. In the C++ interface, RobotState::setFromIK does this but I'm not sure the default KDL solver can handle multiple targets.

felixvd, 2018-06-21:
Could you post an example or a more in-detail description of how you solved things along the way? I am doing something similar and I found that using multiple robots in MoveIt is not well documented online, from what I can tell. I also came across namespace issues and I wonder if you solved them differently.

felixvd, 2018-06-22:
That's unrelated to this issue and you should try posting that on [ROSanswers](https://answers.ros.org/questions/). Your robot or joint states are probably not being published correctly.

felixvd, 2018-07-02:
I tried using the Python API and had the same problem as in the original post. For some reason I cannot set the end effector link when there is more than one subgroup.

PeterMitrano, 2020-09-28:
Is this still true? I'm also not able to set the end effector of a "both_arms" group, seeing the same error msg. See the attached SRDF file for definitions of my groups (there could be an issue there I suppose)

[victor.srdf.txt](https://github.com/ros-planning/moveit/files/5289603/victor.srdf.txt)

lhzmmmm, 2024-09-30:
> Is this still true? I'm also not able to set the end effector of a "both_arms" group, seeing the same error msg. See the attached SRDF file for definitions of my groups (there could be an issue there I suppose)
> 
> [victor.srdf.txt](https://github.com/ros-planning/moveit/files/5289603/victor.srdf.txt)
do you deal with the problem?

rhaschke, 2024-09-30:
I'm afraid the python interface still doesn't support multiple end-effectors in a kinematic tree.
The whole MoveGroupInterface is limited to kinematic chains.