moveit/moveit #804: disambiguate semantics of trajectory_constraints

rhaschke, 2018-03-18:
As pointed out in the discussion of #793, the field `trajectory_constraints` of a [`MotionPlanRequest`](http://docs.ros.org/kinetic/api/moveit_msgs/html/msg/MotionPlanRequest.html) doesn't have a clear semantics yet. They could be considered as:
* a trajectory seed (e.g. for planning in STOMP), while the waypoints of the finally optimized trajectory might violate the "constraints".
* real constraints in the sense that the final solution should satisfy them with some waypoint in the given order.

While the latter interpretation actually asks for a sequential planner (as we implement it currently in https://github.com/ros-planning/moveit_task_constructor), the former one asks for a renaming of the field from trajectory_constraints into `trajectory_seed` / `initial_trajectory`. As pointed out in https://github.com/ros-planning/moveit/pull/793#issuecomment-372043457, a trajectory message would be more suitable for a seed. 
Currently, there is no planner available that actually supports the latter use case. OMPL explicitly rejects planning requests with this field set.

bmagyar, 2018-03-18:
> Currently, there is no planner available that actually supports the latter use case.

STOMP is a planner plugin for MoveIt and supports the latter use case. 
[Here](https://github.com/ros-industrial/industrial_moveit/blob/kinetic-devel/stomp_moveit/src/stomp_planner.cpp#L505)'s a line of code where it does so. It's been out there like that for over a year now. 

> real constraints in the sense that the final solution should satisfy them with some waypoint in the given order.

Shouldn't they be called `waypoints` in the message then? The name `trajectory_constraints` is not specific enough. If we want to make it more specific, then it should be done by renaming the field such that the intended use is clear. If general names are kept, all planners are up to interpret it in whatever they wish.

rhaschke, 2018-03-18:
@bmagyar I'm confused. The source code, you are referring to, uses the trajectory constraints for seeding. With _latter use case_, however, I referred to considering them as real constraints. As far as I know, there is only some benchmark code which uses this field.

bmagyar, 2018-03-19:
You are right, sorry. I'll edit the comment to remove that in order to keep the discussion clean, you could edit yours too.

bmagyar, 2018-04-12:
How can we move this forward guys? 
@rhaschke @v4hn