moveit/moveit #806: Migration to ROS Melodic

davetcoleman, 2018-03-19:
A checklist of items for the next ROS 1.0 LTS Release:

- [ ] class_loader headers migration
- [ ] pluginlib headers migration

awesomebytes, 2018-03-29:
Hello, I could give a hand with the python3 migration and the Python interface in general with @bmagyar.

130s, 2018-04-02:
- Meet dependencies
  - [ ] object_recognition_msgs -> Requested https://github.com/wg-perception/object_recognition_msgs/issues/6
  - [ ] octomap_msgs -> Requested https://github.com/OctoMap/octomap_msgs/issues/9
  - [ ] moveit_msgs

davetcoleman, 2018-04-11:
https://github.com/ros-planning/srdfdom/pull/36
Melodic development has begun! :busts_in_silhouette:

bmagyar, 2018-04-12:
#804 should be squeezed in too

davetcoleman, 2018-05-05:
I've begun working on the Docker / CI for Melodic task

130s, 2018-05-17:
- Maintainers, please hold merges (except critical ones). I'm planning to make a release soon.
- The following 2 seem to be resolved in https://github.com/ros-planning/moveit/pull/827#issuecomment-389453946, so I checked them.
  > - [ ]  class_loader headers migration @jrgnicho
  > - [ ] pluginlib headers migration @jrgnicho
- [All dependency seems to be cleared](http://repositories.ros.org/status_page/blocked_releases_melodic.html?q=moveit&s=6&r=1) so I think I'm ready to `bloom` but have a few questions:
  - There are still some check items unchecked. Are they a requirement for `bloom`? Particularly I have no idea if the following is required:
  > - Cleanup, fixes, API changes:
  >   - [ ]    remove stuff that was indicated as deprecated in Kinetic

  - Is `graph_msgs` required as it's listed above as a dependency? I do not see any mention in the code.
  ```
  $ git branch 
  * melodic-devel
  $ ack-grep -A 1 -B 1 -i graph_msgs .
  $ git log -1
  commit 6cca2555a540be43c085f2024f3cebe33fa670ea
  Author: Robert Haschke <rhaschke@users.noreply.github.com>
  Date:   Wed May 16 16:37:52 2018 +0200
  
      removing moveit_tutorials from moveit.rosinstall (#898)
      
      This partially reverts 068dcc0b82e512c7405cfa51009652516ccc9b2b, which added the tutorials just recently.
  ```

130s, 2018-05-17:
ROS Melodic release seems to be set in stone [on May 23rd](https://discourse.ros.org/t/ros-melodic-morenia-beta-has-begun/4743/2?u=130s) and I assume new release needs to be made prior to that date ([asked exact hold day](https://discourse.ros.org/t/ros-melodic-morenia-beta-has-begun/4743/2?u=130s)). I'll try my best to make the official release this time (would it be the first time for MoveIt! to make it?), so let me know anything holds up `bloom`ing.

IanTheEngineer, 2018-05-17:
I would advocate for TF2 Migration https://github.com/ros-planning/moveit/pull/830 and `boost::shared_ptr` removal https://github.com/ros-planning/moveit/pull/896 to be merged into `melodic-devel` prior to a Melodic release. They are important for ROS2 efforts, and break API. I will try to make sure they're merged soon to unblock the release.

rhaschke, 2018-05-17:
@mamoll The release of OMPL into Melodic is missing the `ompl-config.cmake` file. This blocks source builds.
```sh
> ls /opt/ros/melodic/share/ompl/
demos  ompl.conf
```
```sh
ls /opt/ros/kinetic/share/ompl/
demos  ompl.conf  ompl-config.cmake
```

130s, 2018-05-18:
@IanTheEngineer I see #830 being closed :fireworks: 

@rhaschke Can I ask you to give a "go" for OMPL issue https://github.com/ros-planning/moveit/issues/806#issuecomment-390029076? I can't make a judgement for that.

There are still several unchecked items. I think the last possible day I can bloom to make into Melodic official release is probably 21th Monday (based on the [response from Melodic manager](https://discourse.ros.org/t/ros-melodic-morenia-beta-has-begun/4743/3)), which sounds a bit unrealistic. So, no pressure folks.

rhaschke, 2018-05-18:
@130s MoveIt will not build on Melodic until OMPL is re-released by @mamoll.
For now, I added a quick hack into our Travis docker container to see remaining issues at least.
By the way, is there anybody actually **using** the Melodic branch?

rhaschke, 2018-05-21:
@130s, @mamoll Having changed `find_package(OMPL)` -> `find_package(ompl)` as suggested [here](https://github.com/ros/catkin/issues/938#issuecomment-390285366), solves the issue for Melodic too. No need to do another release.

mamoll, 2018-05-21:
Does that OMPL is treated as a pure cmake package and this is not necessary?:
https://github.com/ompl/ompl/blob/0f4f2ef283d5390a118d9bec3fbbe92efd0fe988/CMakeLists.txt#L245-L249
If so, can this file be deleted?:
https://github.com/ros-gbp/ompl-release/blob/master/melodic/package.xml

130s, 2018-05-21:
AFAIU all blocking items are checked, thanks to some of you particularly @rhaschke. If I hear no objection I'll bloom in a few hours.

130s, 2018-05-22:
Sorry for taking me awhile, but I bloomed https://github.com/ros/rosdistro/pull/17942

I couldn't finish organizing changelog...I've started editing changelog that was generated from `catkin_generate_changelog`, then I got called in for urgent work stuff. I'd rather just make a release with unorganized set of changelog than delaying the release. In the future release I'll try sorting them out.

rhaschke, 2018-05-24:
> Does that OMPL is treated as a pure cmake package and this is not necessary?:
https://github.com/ompl/ompl/blob/0f4f2ef283d5390a118d9bec3fbbe92efd0fe988/CMakeLists.txt#L245-L249
If so, can this file be deleted?:
https://github.com/ros-gbp/ompl-release/blob/master/melodic/package.xml

@mamoll At least, MoveIt uses OMPL as a cmake-only package and the package.xml could be removed.

rhaschke, 2018-05-24:
That's a valid point. But, @mamoll could decide to release OMPL through a different mechanism instead.

mamoll, 2018-05-24:
I'll stick with bloom for now. It makes releasing packages for the different ROS versions easy. Thanks, @jspricke and @rhaschke for helping me understand what's necessary and sufficient for ROS releases ;-).

130s, 2018-05-29:
> - [ ]  Make an announcement

This was suggested in today's mtg.
- We should announce that MoveIt! binary is available on Melodic
- It should NOT be expected to be stable yet. Maybe calling it "beta"?

130s, 2018-05-29:
> - [ ]  Make an announcement

This was suggested in today's mtg.
- We should announce that MoveIt! binary is available on Melodic
- It should NOT be expected to be stable yet. Maybe calling it "beta"?
- Mention about Lunar situation https://github.com/ros-planning/moveit/issues/889

I can go down for this rabbit hole.

closed 2018-08-20 by rhaschke