moveit/moveit #807: Connect rviz using moveit! to database

thisyearspairo, 2018-03-23:
### Description
Hello, I'm working with the Universal_Robot package in order to control the UR3 robot and I would like to know how to connect Rviz to a database in order to save the states I'm creating, because when I create them the program gives me: "Not connected to a database. The state will be created but not stored"
I'm using Ubuntu 16.04 and Ros Kinetic. 
Thank you very much in advance.

k-okada, 2018-03-23:
I think this is because of removing mongdb warehouse plugin, someone (including me) is tring to setup sql based plugin, if they have time ... https://github.com/ros-planning/moveit/issues/123

Krebsileinchen, 2018-05-02:
Are there any news about this topic? I started having the same issue. By now I installed MongoDB Warehouse and got a step further (http://moveit.ros.org/install/source/dependencies/). It starts connecting but then I get the error 
[ERROR] [1525254202.219795349]: Unable to connect to the database at '127.0.0.1:33829'. If you just created the database, it could take a while for initial setup.
When trying 
roslaunch MYROBOT_moveit_config default_warehouse_db.launch 
I get
[mongo_wrapper_ros_...] process has died [pid 980, exit code 1, cmd /home/../catkin_ws/src/warehouse_ros_mongo/src/mongo_wrapper_ros.py __name:=mongo_wrapper_ros_63_6095246583607931054 __log:=/home/../.ros/log/b3e2694e-4dcf-11e8-a39c-0010182ca071/mongo_wrapper_ros_sew_ws_963_6095246583607931054-1.log].
log file: /home/de6ak529/.ros/log/b3e2694e-4dcf-11e8-a39c-0010182ca071/mongo_wrapper_ros_.._6095246583607931054-1*.log
I would appreciate any ideas!