moveit/moveit #814: Request: add clang-tidy check for variable format

davetcoleman, 2018-03-29:
It would be nice to enforce variable naming convention using a clang-tidy check in our Travis build:

See [IdentifierNamingCheck](https://github.com/llvm-mirror/clang-tools-extra/commit/89b8288d86c6b9cbe2526e8b78b8c3559a34e7e4) and [Example Implementation](https://gitlab.inria.fr/openmp/clang/commit/2bdc4cd10b8d2d4ef69584b7161526006b3ca86b)

This would enforce not using ``CamelCase`` but instead ``under_scores``.

BryceStevenWilley, 2018-03-29:
#746 actually includes the `readability-identifier-naming` check, but without any options. Didn't realize that the [default](https://github.com/llvm-mirror/clang-tools-extra/commit/89b8288d86c6b9cbe2526e8b78b8c3559a34e7e4#diff-e612f120280a516d9ae1f817a17bc515R93) is actually `AnyCase`. Good catch!  I can open a new PR with this change after #746 is merged.  

I'm asuming we would stick with the current conventions of:
* Classes as `CamelCase`
* Enums as `CamelCase`
* Enum constants as `UPPER_CASE`
* Functions as `camelBack`
* Unions (are there even any in Moveit?) as `CamelCase`
* Variables as `lower_case`
* Members as `lower_case` with a `_` suffix

unless anyone has any objections.

davetcoleman, 2018-05-23:
#880 is merged :)

Those conventions look correct to me

davetcoleman, 2019-05-31:
I believe this issue has been addressed, closing. Thanks @BryceStevenWilley !

closed 2019-05-31 by davetcoleman