moveit/moveit #818: compiling error on mac

horagong, 2018-04-04:
### Description

Compiling error in link_model.h
It was ok with this change.
@@ -65,7 +65,7 @@ typedef std::map<std::string, const LinkModel*> LinkModelMapConst;
 
 /** \brief Map from link model instances to Eigen transforms */
 typedef std::map<const LinkModel*, Eigen::Affine3d, std::less<const LinkModel*>,
-                 Eigen::aligned_allocator<std::pair<const LinkModel*, Eigen::Affine3d> > >
+                 Eigen::aligned_allocator<std::pair<const LinkModel* const, Eigen::Affine3d> > >

### Your environment
* ROS Distro: lunar
* OS Version: Mac 10.13.3
* Source or Binary build? source
* If binary, which release version? 
* If source, which git commit or tag? ff2d2

### Steps to reproduce
When compiling in mac
### Expected behaviour
No failed
### Actual behaviour
failed

### Backtrace or Console output

/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/map:818:5: error: static_assert failed "Allocator::value_type must be same type as value_type"
    static_assert((is_same<typename allocator_type::value_type, value_type>::value),
    ^             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/src/moveit/moveit_core/robot_model/include/moveit/robot_model/link_model.h:257:20: note: in instantiation of template class 'std::__1::map<const moveit::core::LinkModel *, Eigen::Transform<double, 3, 2, 0>, std::__1::less<const moveit::core::LinkModel *>, Eigen::aligned_allocator<std::__1::pair<const moveit::core::LinkModel *, Eigen::Transform<double, 3, 2, 0> > > >' requested here
  LinkTransformMap associated_fixed_transforms_;

horagong, 2018-04-04:
The following files got erros in logXXX.
I changed it to CONSOLE_BRIDGE_logXXX and it was ok.

moveit/moveit_experimental/collision_distance_field/include/moveit/collision_distance_field/collision_distance_field_types.h
moveit/moveit_experimental/collision_distance_field/src/collision_distance_field_types.cpp
moveit/moveit_experimental/collision_distance_field/include/moveit/collision_distance_field/collision_robot_distance_field.h
moveit/moveit_experimental/collision_distance_field/src/collision_robot_distance_field.cpp

horagong, 2018-04-04:
The following files got errors when I have no OpenMP.
Is it necessary? I compiled it with this.

+if (OpenMP_FOUND OR OpenMP_CXX_FLAGS)
+       set_target_properties(${MOVEIT_LIB_NAME}_core PROPERTIES COMPILE_FLAGS "${CMAKE_CXX_FLAGS} ${OpenMP_CXX_FLAGS}")
+       set_target_properties(${MOVEIT_LIB_NAME}_core PROPERTIES LINK_FLAGS "${CMAKE_CXX_FLAGS} ${OpenMP_CXX_FLAGS}")
+endif (OpenMP_FOUND OR OpenMP_CXX_FLAGS)

moveit/moveit_ros/perception/lazy_free_space_updater/CMakeLists.txt
moveit/moveit_ros/perception/pointcloud_octomap_updater/CMakeLists.txt
moveit/moveit_ros/perception/point_containment_filter/CMakeLists.txt

moveit_ros_perception:make 
clang: error: no such file or directory: 'NOTFOUND'
make[2]: *** [point_containment_filter/CMakeFiles/moveit_point_containment_filter.dir/src/shape_mask.cpp.o] Error 1
make[1]: *** [point_containment_filter/CMakeFiles/moveit_point_containment_filter.dir/all] Error 2

horagong, 2018-04-04:
moveit/moveit_setup_assistant/src/tools/moveit_config_data.cpp:697:31: error: no matching member function for call to 'assign'
  srdf_->disabled_collisions_.assign(disabled_collisions.begin(), disabled_collisions.end());
  ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/vector:588:10: note: candidate function not viable: no known conversion from 'std::__1::set<moveit_setup_assistant::SortableDisabledCollision, std::__1::less<moveit_setup_assistant::SortableDisabledCollision>, std::__1::allocator<moveit_setup_assistant::SortableDisabledCollision> >::iterator' (aka '__tree_const_iterator<moveit_setup_assistant::SortableDisabledCollision, std::__1::__tree_node<moveit_setup_assistant::SortableDisabledCollision, void *> *, long>') to 'std::__1::vector<srdf::Model::DisabledCollision, std::__1::allocator<srdf::Model::DisabledCollision> >::size_type' (aka 'unsigned long') for 1st argument
    void assign(size_type __n, const_reference __u);
         ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/vector:576:9: note: candidate template ignored: requirement '!__is_forward_iterator<__tree_const_iterator<SortableDisabledCollision, __tree_node<SortableDisabledCollision, void *> *, long> >::value' was not satisfied [with _InputIterator = std::__1::__tree_const_iterator<moveit_setup_assistant::SortableDisabledCollision, std::__1::__tree_node<moveit_setup_assistant::SortableDisabledCollision, void *> *, long>]
        assign(_InputIterator __first, _InputIterator __last);
        ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/vector:586:9: note: candidate template ignored: requirement 'is_constructible<value_type, typename iterator_traits<__tree_const_iterator<SortableDisabledCollision, __tree_node<SortableDisabledCollision, void *> *, long> >::reference>::value' was not satisfied [with _ForwardIterator = std::__1::__tree_const_iterator<moveit_setup_assistant::SortableDisabledCollision, std::__1::__tree_node<moveit_setup_assistant::SortableDisabledCollision, void *> *, long>]
        assign(_ForwardIterator __first, _ForwardIterator __last);
        ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/vector:592:10: note: candidate function not viable: requires single argument '__il', but 2 arguments were provided
    void assign(initializer_list<value_type> __il)

I changed it like this for compiling.

  //XXX srdf_->disabled_collisions_.assign(disabled_collisions.begin(), disabled_collisions.end());
  std::set<SortableDisabledCollision>::iterator sdc;
  for (sdc = disabled_collisions.begin(); sdc != disabled_collisions.end(); sdc++) { 
    const srdf::Model::DisabledCollision dc = *sdc;
    srdf_->disabled_collisions_.push_back(dc);
  }

horagong, 2018-04-04:
With these changes, I compiled it successfully!

v4hn, 2018-04-05:
First of all thanks for looking into these problems.

Please provide a pull-request with your patches for `kinetic-devel`,
so that we can review and merge them properly.