moveit/moveit #819: setFromDiffIK changes robot state before state validity callback is called

thebarnable, 2018-04-04:
According to the [documentation](http://docs.ros.org/jade/api/moveit_core/html/namespacemoveit_1_1core.html#a9503536c23482749230ae2a0af36d5f0), the callback function given to setFromIK respectively setFromDiffIK has the arguments robot_state, joint_group and joint_group_variable_values. If I understand correctly, the former two should represent the _current_ state of the robot, the latter contains the calculated goal positions from the IK solver. Callbacks called from _setFromIK_ show this behaviour. However, callbacks called from _setFromDiffIK_ don't. In these, robot_state & joint_group seem to have already been set to the calculated positions - they contain the same values as joint_group_variable_values.

### Your environment
* ROS Distro: Kinetic
* OS Version: Ubuntu 16.04
* Source or Binary build? Binary
* If binary, which release version? 0.9.11

### Steps to reproduce
I do not have a MWE handy right now, but the behavior can be explained with the source code (I hope this is enough for now). In [setFromIK](http://docs.ros.org/jade/api/moveit_core/html/robot__state_8cpp_source.html#l01645), setJointGroupPositions is called _after_ solving the IK and calling the callback. [setFromDiffIK](http://docs.ros.org/jade/api/moveit_core/html/robot__state_8cpp_source.html#l01200) directly returns the return value of the integrateVariableVelocity, in which setJointGroupPositions is called _before_ the callback.

v4hn, 2018-04-05:
Please don't link to jade-documentation anymore. Nothing changed in the relevant parts of the code, but still this is confusing.

Your analysis seems to be correct.
Would you propose to change the behavior in some way or improve documentation?

I am not sure anything has to be changed in the code.
As far as I can see it is left unspecified what the values of the `RobotState` are that is passed in.

thebarnable, 2018-04-09:
Sorry for that, I'll fix the links.

I would actually propose to change the behavior. Firstly, the documentation for both setFromIK and setFromDiffIK links to the same documentation for the callback - even though, as you said, the exact behavior is left unspecified, I would expect the callbacks for both functions to have the _same_  behavior. This is not only more intuitive (as both IK functions have very similar names and headers), but it can also make the user's life easier, e.g. if both setFromIK and setFromDiffIK are used and shall point to the same callback. 

Secondly, if the values in RobotState are already set to the new ones, passing in 'joint_group_variable_values' as well would be redundant, because copyJointGroupPositions would yield the exact same values..

v4hn, 2018-04-10:
On Mon, Apr 09, 2018 at 07:02:06AM +0000, Tim Stadtmann wrote:
> I would actually propose to change the behavior.

In this case please send a pull-request and we can discuss there.

Please be aware that either way you should not ascribe fixed semantics to the values that are set in the input `RobotState`.
The semantics depend on the IK plugin that is currently used.