moveit/moveit #840: allowed_start_tolerance_ parameter is used incorrectly in waitForRobotToStop method in the trajectory_execution_manager

srsidd, 2018-04-13:
### Description
PR #350  added a method `TrajectoryExecutionManager::waitForRobotToStop` in `moveit_ros/planning/trajectory_execution_manager/src/trajectory_execution_manager.cpp` which uses the same parameter `allowed_start_tolerance` to check if the robot has stopped executing it's plan or not. However the `allowed_start_tolerance` parameter is to verify that trajectory's first point matches current robot state. We need a different parameter in the `TrajectoryExecutionManager::waitForRobotToStop` so that we can bypass it if need from the launch file without affecting the pre-execution check.

### Your environment
* ROS Distro: [|Kinetic]
* Ubuntu 16.04
* Source - kinetic-devel from PR #350

rhaschke, 2018-04-15:
> We need a different parameter in `TrajectoryExecutionManager::waitForRobotToStop` so that we can bypass it if needed without affecting the pre-execution check.

I'm not sure what you want to bypass: The stopping criterium should not be bypassed! However, if `allowed_start_tolerance` is zero (to bypass the start sanity check), we should use here a non-zero default fallback. However, in general both thresholds should be identical.

v4hn, 2018-04-17:
> The stopping criterium should not be bypassed!

Why not? If user knows what they are doing, there is no need for MoveIt to trip them up
with additional sleep time.

I think separating these concepts into two parameters (possibly with one defaulting to the other)
is perfectly justified.

v4hn, 2018-04-19:
Please clear up my misunderstanding here.
If I would like to use this interface for continuous control (i.e. not stopping the robot between trajectories),
one way to do this is to use an action server that can interpolate to new incoming trajectory goals
(ros-control's action server can do that), make it return shortly before it arrives (or with non-zero goal velocities)
and execute the next trajectory in time to continue smoothly.
This setup might require some work in other places, but I don't see the need to render it impossible here.
Also, it's just another ROS parameter with a default 1 in 50 users might change.
It's not like we don't have plenty of them already.

I remember talking to someone at ROSCon 2017 who does something similar with ROS actions already. (maybe @gavanderhoorn ?)

rhaschke, 2018-04-19:
I see your point. However, for such a use case the present API is not well-suited. How do you "make it return shortly before it arrives"? In the general case, the low-level controller will not know whether there is another sub trajectory pending or not. Hence, it cannot know whether to early return or to finish the movement.
I think, the execution manager should monitor the trajectory progress himself and send a new a sub trajectory to the controller in time. To this end, the execution manager should have an appropriate parameter that defines the switching tolerance. However, this also heavily relies on a nice interplay between low-level controller and execution manager.

v4hn, 2018-04-20:
> However, for such a use case the present API is not well-suited.

Maybe not, but it *is* the API we use when someone wants to have online collision-checking during execution.
Of course we can always write a new API. I just don't think a single parameter should be the reason. :-)

> the low-level controller will not know whether there is another sub trajectory pending or not.

This still leaves trajectories that end in non-zero velocity (which is the better approach anyway).

> I think, the execution manager should monitor the trajectory progress himself and send a new a sub trajectory to the controller in time.

This only supports a finite list of subtrajectories, which is usually not what I am interested in if I want to work with collision-aware execution.

Anyway, this discussion went way beyond this issue.
I still encourage and support an upstream change that allows to set the value differently.

srsidd, 2018-10-17:
This is no longer an issue since #883 addressed it and has been merged.

closed 2018-10-17 by srsidd