moveit/moveit #852: Returning dirty collision body transforms warning

RhysMcK, 2018-04-20:
### Description
I have just installed Moveit! from the latest source commit and i am now receiving the warning "Returning dirty collision body transforms" whenever i run my program. This warning was not showing up moments ago when i had Moveit installed from binary. Any ideas on why i might be getting this warning ?

Thanks!

### Environment
* ROS Distro:Kinetic
* OS Version: Ubuntu 16.04
* Source  https://github.com/ros-planning/moveit/commit/78d862a5d942da3b4a0b91759bf698b096fa62a8

rhaschke, 2018-04-20:
In order to answer your question, it would be helpful if you refer to the program you are using.
In general you receive those message, if you updated the robot joint state, but didn't called update() on your robot state. If you then access any transforms (e.g. for collision checking) via a const RobotState, you will receive those errors.

RhysMcK, 2018-04-22:
Sorry, yes, I am just currently using Moveit!'s MotionPlanning plugin with Rviz. I am not running any of my own code at the moment (apart from the URDF).

closed 2019-02-15 by RhysMcK

gavanderhoorn, 2019-02-15:
@RhysMcK: was this resolved? Has the error stopped popping up for you?

Jus tasking as I'm also running into many "dirty collision" warnings just using the RViz plugin.

rhaschke, 2019-02-15:
@gavanderhoorn Can you provide more background on your issue. 
Which MoveIt version? Which planning pipeline?

gavanderhoorn, 2019-02-15:
Kinetic, Xenial, current binaries (non shadow), OMPL.

Only really "different" thing I can think of is I'm using the spline based time parameterisation adapter, instead of the regular IPTP.

rhaschke, 2019-02-15:
If this one modifies the RobotStates in the trajectory, but doesn't call RobotState::update(), that might be the reason. Could you explain when exactly (i.e. which phase of planning / execution) you get the errors?
Best provide the log :wink:

gavanderhoorn, 2019-02-15:
> If this one modifies the RobotStates in the trajectory, but doesn't call RobotState::update(), that might be the reason. 

yes, I remembered a PR where that was fixed some time ago, but couldn't remember whether that was only for IPTP or also for the new spline based adapter.

> Could you explain when exactly (i.e. which phase of planning / execution) you get the errors? Best provide the log :wink:

I realised the minute I posted that comment that I was "violating" all conventions I normally require others to follow when reporting issues, but I just happened to notice @RhysMcK closing this issue and acted on impulse.

I'll see if I can try to narrow it down today.

rhaschke, 2019-02-15:
The PR https://github.com/ros-planning/moveit/pull/1298 you mentioned only addressed the default IPTP.

gavanderhoorn, 2019-02-15:
Could well be that the spline based one isn't doing the right thing and that's why I'm seeing those errors.

I'll report back once I confirm it.