moveit/moveit #855: collision_detection causes crash at runtime

ToRaVe, 2018-04-24:
### Description

I'd like to alter my robot's behaviour when it collides with itself. In order to do so I utilize the function "checkSelfCondition(colReq, colRes)" which tends to crash as soon as a there is a collision (I checked this via RViz).

### Your environment
* ROS Distro: Kinetic
* OS Version: Ubuntu 16.04
* Source build
* If source, which git commit or tag? Not sure. How can I look it up?

### Steps to reproduce
I use checkSelfCollision inside a node. My code there looks like this:
```
planning_scene::PlanningScene planning_scene(kinematic_model);
    collision_detection::CollisionRequest colReq;
    collision_detection::CollisionResult colRes;
    colReq.contacts = true;
    colReq.max_contacts = 1000;
    planning_scene.setCurrentState(rstate); // rstate being a robot_state::RobotState

    joint_state.header.stamp = ros::Time::now();
    jPub.publish(joint_state);

    planning_scene.checkSelfCollision(colReq, colRes);
```

### Expected behaviour
colRes.collision = false  if there is no collision
colRes.collision = true if there is a collision

### Actual behaviour
When there is no collision, everything works fine. But when the bot collides with itself, ros crashes.

### Backtrace or Console output
*** stack smashing detected ***: /home/tvetter/catkin_ws/devel/lib/analyzews/manipulability_new terminated
[my-4] process has died [pid 10962, exit code -6, cmd /home/ToRaVe/catkin_ws/devel/lib/anal/my_new __name:=my __log:=/home/ToRaVe/.ros/log/a5158c2c-47ab-11e8-8c5f-080027f17e31/manipulability-4.log].
log file: /home/tvetter/.ros/log/a5158c2c-47ab-11e8-8c5f-080027f17e31/manipulability-4*.log

Thank you for your help (:

rhaschke, 2018-04-24:
Please provide more details:
* Your source version, run `git log` in your source folder and report the top-most commit ID.
* Your full source code of the node.
* A backtrace of your crashing node.

ToRaVe, 2018-04-25:
### No source, I'm sorry
Comming to think of it, I must admit that my information concerning the source build was wrong. Moveit has in fact been installed via apt-get.

### Source Code

`#include "ros/ros.h"
#include <moveit/robot_state/robot_state.h>
#include <moveit/robot_model_loader/robot_model_loader.h>
#include <moveit/robot_model/robot_model.h>
#include <moveit/robot_state/conversions.h>
#include <moveit/kinematics_metrics/kinematics_metrics.h>
#include <moveit/collision_detection/collision_robot.h>

#include <math.h>

#include <fstream>
#include <iterator>
#include <string>

#include <chrono>
#include <ctime>

class Manipulat0r{
public:
  Manipulat0r(){
    initialize();  // Initialization of several variables and retrieving variables from the environment set by the launch-file
  }

  bool work(){
    count++;

    if(!generateNextJointState(joint_state.position)){  // If no new joint_state can be found, publish the results and finish the process
      outputmsg();                // Display the results
      maPub.publish(markerArray); // Publish the markers
      joint_state.header.stamp = ros::Time::now();
      jPub.publish(joint_state);  // Publish the joints
      return false;
    }

    robot_state::RobotState rstate(kinematic_model);
    moveit::core::jointStateToRobotState(joint_state, rstate);

    // TODO
    //robot_model_loader = robot_model_loader::RobotModelLoader("robot_description");
    //robot_model::RobotModelPtr kinematic_model = robot_model_loader.getModel();
    planning_scene::PlanningScene planning_scene(kinematic_model);
    collision_detection::CollisionRequest colReq;
    collision_detection::CollisionResult colRes;
    colReq.contacts = true;
    colReq.max_contacts = 1000;
    planning_scene.setCurrentState(rstate);
    robot_state::RobotState test_state = planning_scene.getCurrentState();
    test_state.printStatePositions();

    joint_state.header.stamp = ros::Time::now();
    jPub.publish(joint_state);

    planning_scene.checkSelfCollision(colReq, colRes); //!!! Here it crashes
    /*if(colRes.collision || planning_scene.isStateColliding()){
      std::cout << "Kollision detektiert" << std::endl;
      ros::Duration(60).sleep();
    }*/
    //std::cout << colRes.distance << std::endl;

    kinematics_metrics::KinematicsMetrics metrics(kinematic_model);
    metrics.getManipulabilityIndex(rstate, "arm", manipulIndex);

    cartesianPosition = getCartesianPosition(robotName, rstate);
    for(int ii=0; ii<3; ii++)
      tempPosition[ii] = cartesianPosition[ii];
[... 256 lines cut ...]
};

int main(int argc, char **argv){

  ros::init(argc, argv, "manip_calc");    // Initialize ROS

  Manipulat0r m;                          // Create object of Manipulat0r class, which contains all necessary variables and does all the work
  
  bool positionsAvailable = true;         // Determines if there are joint-space positions that have not been visited yet

  ros::Duration(1).sleep();

  while(ros::ok() && positionsAvailable)  // Go through the joint space and calculate manipulability indices until all allowed positions have been visited
    positionsAvailable = m.work();

  ros::spin();  // Keep ROS running.

  return 0;
}
`

### Backtrace
The error message at the bottom of my first post is all I get inside the terminal. Is there another location where a backtrace can be found?

I'm quite new to ros so I'm sorry if I'm asking dumb questions or making stupid mistakes. Thank you for helping (:

ToRaVe, 2018-04-25:
The code-tags seem to have somehow removed the float specification. I try again without the tags:

#include "ros/ros.h"
#include <moveit/robot_state/robot_state.h>
#include <moveit/robot_model_loader/robot_model_loader.h>
#include <moveit/robot_model/robot_model.h>
#include <moveit/robot_state/conversions.h>
#include <moveit/kinematics_metrics/kinematics_metrics.h>
#include <moveit/collision_detection/collision_robot.h>

#include <math.h>

#include <fstream>
#include <iterator>
#include <string>

#include <chrono>
#include <ctime>

class Manipulat0r{
public:
  Manipulat0r(){
    initialize();  // Initialization of several variables and retrieving variables from the environment set by the launch-file
  }

  bool work(){
    count++;

    if(!generateNextJointState(joint_state.position)){  // If no new joint_state can be found, publish the results and finish the process
      outputmsg();                // Display the results
      maPub.publish(markerArray); // Publish the markers
      joint_state.header.stamp = ros::Time::now();
      jPub.publish(joint_state);  // Publish the joints
      return false;
    }

    robot_state::RobotState rstate(kinematic_model);            // Generate robotstate depending on kinematic model
    moveit::core::jointStateToRobotState(joint_state, rstate);  // Apply the updated joint state to receive the corresponding cartisian state

    // TODO: Stuff for collision detection
    planning_scene::PlanningScene planning_scene(kinematic_model);
    collision_detection::CollisionRequest colReq;
    collision_detection::CollisionResult colRes;
    colReq.contacts = true;       // Make it collect contacts
    colReq.max_contacts = 1000;
    planning_scene.setCurrentState(rstate); // Give the current state to the planning_scene object

    /* Optional
    joint_state.header.stamp = ros::Time::now();
    jPub.publish(joint_state);*/

    planning_scene.checkSelfCollision(colReq, colRes); // Here the program crashes
    /*if(colRes.collision || planning_scene.isStateColliding()){
      std::cout << "Kollision detektiert" << std::endl;
      ros::Duration(60).sleep();
    }*/

    // Stuff for manipulability
    kinematics_metrics::KinematicsMetrics metrics(kinematic_model);
    metrics.getManipulabilityIndex(rstate, "arm", manipulIndex);

    cartesianPosition = getCartesianPosition(robotName, rstate);
    for(int ii=0; ii<3; ii++) // Transport from Eigen::Matrix to std::Vector
      tempPosition[ii] = cartesianPosition[ii];

    int idx = 0;
    if(!overlapping(tempPosition, positions, idx)){
      createNewMarker(marker, manipulIndex);

      // For storage in file
[... 266 lines cut ...]
  moveit::core::RobotModelConstPtr robot_model;

  visualization_msgs::Marker marker;
  visualization_msgs::MarkerArray markerArray;

  int count, countM;
};

int main(int argc, char **argv){

  ros::init(argc, argv, "manip_calc");    // Initialize ROS

  Manipulat0r m;                          // Create object of Manipulat0r class, which contains all necessary variables and does all the work
  
  bool positionsAvailable = true;         // Determines if there are joint-space positions that have not been visited yet

  ros::Duration(1).sleep();

  while(ros::ok() && positionsAvailable)  // Go through the joint space and calculate manipulability indices until all allowed positions have been visited
    positionsAvailable = m.work();

  ros::spin();  // Keep ROS running.

  return 0;
}

rhaschke, 2018-04-25:
@ToRaVe Instead of pasting your source code here, you should use https://gist.github.com.

ToRaVe, 2018-04-25:
@rhaschke 
Alright, I hope it's working now and nothing is missing. [LinktoCode](https://gist.github.com/ToRaVe/6945dc36bd9f2cddf9500a6fc071fde3#file-man-cpp)

Thanks for your help

ToRaVe, 2018-04-25:
I just tested my code with the UR5 robot. In this case, the behavior is somewhat different from what I described above (where I used the lwa-bot).
In this case, several collisions are detected before there is a crash. The crash however occurs at the same line.

### Console output
*** stack smashing detected ***: /home/tvetter/catkin_ws/devel/lib/analyzews/manipulability_new terminated
[manipulability-4] process has died [pid 21073, exit code -6, cmd /home/tvetter/catkin_ws/devel/lib/analyzews/manipulability_new __name:=manipulability __log:=/home/tvetter/.ros/log/16891a46-485f-11e8-9468-080027f17e31/manipulability-4.log].
log file: /home/tvetter/.ros/log/16891a46-485f-11e8-9468-080027f17e31/manipulability-4*.log
================================================================================REQUIRED process [rviz-3] has died!
process has finished cleanly
log file: /home/tvetter/.ros/log/16891a46-485f-11e8-9468-080027f17e31/rviz-3*.log
Initiating shutdown!
================================================================================

gavanderhoorn, 2018-04-25:
I would suggest to build a debug build of MoveIt, run `move_group` in gdb and use the `bt` command to get a backtrace. An `exit code -6` doesn't say much, and is not enough to start diagnosing this.

> The crash however occurs at the same line.

Which line? I don't see any line mentioned in the output you show.

ToRaVe, 2018-04-25:
@gavanderhoorn
Thanks for your suggestion, I am going to try it.

I identified the line by putting some text-output before and after it. If the text before the line is printed in the terminal, but the one after the line is not, then the line in between seems to cause trouble.

gavanderhoorn, 2018-04-25:
Could you please use three backticks (```) to format code or console copy-pastes properly?

> I identified the line by putting some text-output before and after it. If the text before the line is printed in the terminal, but the one after the line is not, then the line in between seems to cause trouble.

"printf debugging" can give a good first indication, but due to buffering and other effects it may not actually always lead to accurate results. A backtrace in GDB is always preferable.

ToRaVe, 2018-04-25:
@gavanderhoorn 
Another indicator is that the code runs properly, when said line is commented out. 
I'm actually not a professional programmer and have never worked with GDB, but I'm trying it.

ToRaVe, 2018-04-25:
@rhaschke @gavanderhoorn I managed to run my node with gdb and got a backtrace. I hope I have done everything right so far.

```#0  0x00007ffff5a26428 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
#1  0x00007ffff5a2802a in __GI_abort () at abort.c:89
#2  0x00007ffff5a687ea in __libc_message (do_abort=do_abort@entry=1, fmt=fmt@entry=0x7ffff5b8049f "*** %s ***: %s terminated\n")
    at ../sysdeps/posix/libc_fatal.c:175
#3  0x00007ffff5b0a15c in __GI___fortify_fail (msg=<optimized out>, msg@entry=0x7ffff5b80481 "stack smashing detected") at fortify_fail.c:37
#4  0x00007ffff5b0a100 in __stack_chk_fail () at stack_chk_fail.c:28
#5  0x00007fffed61b709 in refinePortal () from /usr/local/lib/libccd.so.2
#6  0x00007fffed61b853 in ccdMPRPenetration () from /usr/local/lib/libccd.so.2
#7  0x00007fffefcc515f in fcl::details::GJKCollide(void*, void (*)(void const*, _ccd_vec3_t const*, _ccd_vec3_t*), void (*)(void const*, _ccd_vec3_t*), void*, void (*)(void const*, _ccd_vec3_t const*, _ccd_vec3_t*), void (*)(void const*, _ccd_vec3_t*), unsigned int, double, fcl::Vec3fX<fcl::details::Vec3Data<double> >*, double*, fcl::Vec3fX<fcl::details::Vec3Data<double> >*) () from /usr/lib/x86_64-linux-gnu/libfcl.so.0.5
#8  0x00007fffefb34a97 in bool fcl::GJKSolver_libccd::shapeTriangleIntersect<fcl::Box>(fcl::Box const&, fcl::Transform3f const&, fcl::Vec3fX<fcl::details::Vec3Data<double> > const&, fcl::Vec3fX<fcl::details::Vec3Data<double> > const&, fcl::Vec3fX<fcl::details::Vec3Data<double> > const&, fcl::Transform3f const&, fcl::Vec3fX<fcl::details::Vec3Data<double> >*, double*, fcl::Vec3fX<fcl::details::Vec3Data<double> >*) const () from /usr/lib/x86_64-linux-gnu/libfcl.so.0.5
#9  0x00007fffefb35804 in ?? () from /usr/lib/x86_64-linux-gnu/libfcl.so.0.5
#10 0x00007fffefb359f6 in fcl::MeshShapeCollisionTraversalNodeOBBRSS<fcl::Box, fcl::GJKSolver_libccd>::leafTesting(int, int) const ()
   from /usr/lib/x86_64-linux-gnu/libfcl.so.0.5
#11 0x00007fffefcd4786 in fcl::collisionRecurse(fcl::CollisionTraversalNodeBase*, int, int, std::__cxx11::list<fcl::BVHFrontNode, std::allocator<fcl::BVHFrontNode> >*) () from /usr/lib/x86_64-linux-gnu/libfcl.so.0.5
#12 0x00007fffefcd4786 in fcl::collisionRecurse(fcl::CollisionTraversalNodeBase*, int, int, std::__cxx11::list<fcl::BVHFrontNode, std::allocator<fcl::BVHFrontNode> >*) () from /usr/lib/x86_64-linux-gnu/libfcl.so.0.5
#13 0x00007fffefcd4786 in fcl::collisionRecurse(fcl::CollisionTraversalNodeBase*, int, int, std::__cxx11::list<fcl::BVHFrontNode, std::allocator<fcl::BVHFrontNode> >*) () from /usr/lib/x86_64-linux-gnu/libfcl.so.0.5
#14 0x00007fffefcd4786 in fcl::collisionRecurse(fcl::CollisionTraversalNodeBase*, int, int, std::__cxx11::list<fcl::BVHFrontNode, std::allocator<fcl::BVHFrontNode> >*) () from /usr/lib/x86_64-linux-gnu/libfcl.so.0.5
#15 0x00007fffefb0c587 in unsigned long fcl::details::orientedBVHShapeCollide<fcl::MeshShapeCollisionTraversalNodeOBBRSS<fcl::Box, fcl::GJKSolver_libccd>, fcl::OBBRSS, fcl::Box, fcl::GJKSolver_libccd>(fcl::CollisionGeometry const*, fcl::Transform3f const&, fcl::CollisionGeometry const*, fcl::Transform3f const&, fcl::GJKSolver_libccd const*, fcl::CollisionRequest const&, fcl::CollisionResult&) () from /usr/lib/x86_64-linux-gnu/libfcl.so.0.5
#16 0x00007fffefcee260 in unsigned long fcl::collide<fcl::GJKSolver_libccd>(fcl::CollisionGeometry const*, fcl::Transform3f const&, fcl::CollisionGeometry const*, fcl::Transform3f const&, fcl::GJKSolver_libccd const*, fcl::CollisionRequest const&, fcl::CollisionResult&) ()
   from /usr/lib/x86_64-linux-gnu/libfcl.so.0.5
#17 0x00007fffefcedf33 in fcl::collide(fcl::CollisionObject const*, fcl::CollisionObject const*, fcl::CollisionRequest const&, fcl::CollisionResult&) ()
   from /usr/lib/x86_64-linux-gnu/libfcl.so.0.5
#18 0x00007ffff3c824c3 in collision_detection::collisionCallback(fcl::CollisionObject*, fcl::CollisionObject*, void*) ()
   from /opt/ros/kinetic/lib/libmoveit_collision_detection_fcl.so.0.9.11
#19 0x00007fffeff454ed in fcl::details::dynamic_AABB_tree::selfCollisionRecurse(fcl::NodeBase<fcl::AABB>*, void*, bool (*)(fcl::CollisionObject*, fcl::CollisionObject*, void*)) () from /usr/lib/x86_64-linux-gnu/libfcl.so.0.5
#20 0x00007fffeff4550f in fcl::details::dynamic_AABB_tree::selfCollisionRecurse(fcl::NodeBase<fcl::AABB>*, void*, bool (*)(fcl::CollisionObject*, fcl::CollisionObject*, void*)) () from /usr/lib/x86_64-linux-gnu/libfcl.so.0.5
#21 0x00007ffff3c8eda3 in collision_detection::CollisionRobotFCL::checkSelfCollisionHelper(collision_detection::CollisionRequest const&, collision_detection::CollisionResult&, moveit::core::RobotState const&, collision_detection::AllowedCollisionMatrix const*) const ()
   from /opt/ros/kinetic/lib/libmoveit_collision_detection_fcl.so.0.9.11
#22 0x000000000042c6f8 in Manipulat0r::work (this=0x7fffffffd290) at /home/tvetter/catkin_ws/src/analyzews/src/manipulability_new.cpp:51
#23 0x000000000042af9c in main (argc=1, argv=0x7fffffffd988) at /home/tvetter/catkin_ws/src/analyzews/src/manipulability_new.cpp:355

gavanderhoorn, 2018-04-25:
Not an expert here, but I notice that about 75% of the backtrace shows calls inside `libfcl` and `libccd`. MoveIt is only a user of those libraries. They are not maintained or developed here.

It could be that `collision_detection` is not setting things up correctly before passing the work on to FCL, but at this point it seem more likely the issue is in FCL and/or `libccd`.

rhaschke, 2018-04-25:
This segfault occurs deep in FCL and it doesn't occur on my side when running your code.
Hence, I'm afraid that you are maybe mixing up different library versions (FCL or MoveIt) with non-matching headers. However, it doesn't seem to be MoveIt related. Hence, I will close this issue.

closed 2018-04-25 by rhaschke