moveit/moveit #856: error through catkin_make

kashyapm94, 2018-04-24:
### Description

I am trying to do trajectory planning for a robot arm and for that I have a cpp code. When I try to build it I receive an error. 
Looks like there is an error in conversion of moveit::planning_interface::MoveItErrorCode to bool form.

I am new to ROS and I would be grateful for your help.

Thanks in advance!

### Your environment
* ROS Distro: [Kinetic]
* Ubuntu 16.04

### Steps to reproduce
Created a catkin workspace, placed the igus_trajectory_planning_3 packages in src directory. Used catkin_make install which reached 100% then failed to finish.

### Backtrace or Console output

Base path: /home/kashyap/catkin_ws
Source space: /home/kashyap/catkin_ws/src
Build space: /home/kashyap/catkin_ws/build
Devel space: /home/kashyap/catkin_ws/devel
Install space: /home/kashyap/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/kashyap/catkin_ws/build"
####
####
#### Running command: "make -j8 -l8" in "/home/kashyap/catkin_ws/build"
####
[ 66%] Built target media_files
[ 66%] Built target _pr2_msgs_generate_messages_check_deps_LaserTrajCmd
[ 66%] Built target _pr2_msgs_generate_messages_check_deps_PressureState
[ 66%] Built target _pr2_msgs_generate_messages_check_deps_LaserScannerSignal
[ 66%] Built target _pr2_msgs_generate_messages_check_deps_PeriodicCmd
[ 66%] Built target _pr2_msgs_generate_messages_check_deps_AccelerometerState
[ 66%] Built target _pr2_msgs_generate_messages_check_deps_BatteryState
[ 66%] Built target _pr2_msgs_generate_messages_check_deps_AccessPoint
[ 66%] Built target _pr2_msgs_generate_messages_check_deps_BatteryServer2
[ 66%] Built target _pr2_msgs_generate_messages_check_deps_DashboardState
[ 66%] Built target _pr2_msgs_generate_messages_check_deps_BatteryState2
[ 66%] Built target _pr2_msgs_generate_messages_check_deps_PowerState
[ 66%] Built target _pr2_msgs_generate_messages_check_deps_SetPeriodicCmd
[ 66%] Built target _pr2_msgs_generate_messages_check_deps_BatteryServer
[ 66%] Built target _pr2_msgs_generate_messages_check_deps_SetLaserTrajCmd
[ 66%] Built target _pr2_msgs_generate_messages_check_deps_PowerBoardState
[ 66%] Built target geometry_msgs_generate_messages_lisp
[ 66%] Built target std_msgs_generate_messages_lisp
[ 66%] Built target geometry_msgs_generate_messages_nodejs
[ 66%] Built target geometry_msgs_generate_messages_eus
[ 66%] Built target std_msgs_generate_messages_nodejs
[ 66%] Built target std_msgs_generate_messages_eus
[ 66%] Built target geometry_msgs_generate_messages_py
Scanning dependencies of target set_pose1_cpp
[ 66%] Built target _pr2_msgs_generate_messages_check_deps_GPUStatus
[ 66%] Built target geometry_msgs_generate_messages_cpp
[ 66%] Built target std_msgs_generate_messages_cpp
[ 66%] Built target std_msgs_generate_messages_py
[ 66%] Built target kinematic_model_tutorial
[ 67%] Built target ros_api_tutorial
[ 67%] Building CXX object igus_trajectory_planning_3/CMakeFiles/set_pose1_cpp.dir/src/set_pose1_cpp.cpp.o
[ 67%] Built target planning_pipeline_tutorial
[ 68%] Built target planning_scene_ros_api_tutorial
[ 69%] Built target planning_scene_tutorial
[ 70%] Built target motion_planning_api_tutorial
[ 70%] Built target move_group_interface_tutorial
[ 71%] Built target state_display_tutorial
[ 76%] Built target pr2_msgs_generate_messages_lisp
[ 80%] Built target pr2_msgs_generate_messages_nodejs
[ 82%] Built target interactivity_tutorial
[ 87%] Built target pr2_msgs_generate_messages_eus
[ 88%] Built target pick_place_tutorial
[ 93%] Built target pr2_msgs_generate_messages_py
[ 94%] Built target collision_contact_tutorial
[ 98%] Built target pr2_msgs_generate_messages_cpp
[100%] Built target attached_body_tutorial
[100%] Built target pr2_msgs_generate_messages
/home/kashyap/catkin_ws/src/igus_trajectory_planning_3/src/set_pose1_cpp.cpp: In function ‘int main(int, char**)’:
/home/kashyap/catkin_ws/src/igus_trajectory_planning_3/src/set_pose1_cpp.cpp:15:1: error: ‘PlanningSceneInterface’ is not a member of ‘moveit::planning_interface’
 moveit::planning_interface::PlanningSceneInterface planning_scene_interface;
 ^
/home/kashyap/catkin_ws/src/igus_trajectory_planning_3/src/set_pose1_cpp.cpp:41:34: error: cannot convert ‘moveit::planning_interface::MoveItErrorCode’ to ‘bool’ in initialization
 bool success = group.plan(my_plan);
                                  ^
/home/kashyap/catkin_ws/src/igus_trajectory_planning_3/src/set_pose1_cpp.cpp:52:9: error: cannot convert ‘moveit::planning_interface::MoveItErrorCode’ to ‘bool’ in assignment
 success = group.plan(my_plan);
         ^
igus_trajectory_planning_3/CMakeFiles/set_pose1_cpp.dir/build.make:62: recipe for target 'igus_trajectory_planning_3/CMakeFiles/set_pose1_cpp.dir/src/set_pose1_cpp.cpp.o' failed
make[2]: *** [igus_trajectory_planning_3/CMakeFiles/set_pose1_cpp.dir/src/set_pose1_cpp.cpp.o] Error 1
CMakeFiles/Makefile2:2591: recipe for target 'igus_trajectory_planning_3/CMakeFiles/set_pose1_cpp.dir/all' failed
make[1]: *** [igus_trajectory_planning_3/CMakeFiles/set_pose1_cpp.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

jrgnicho, 2018-04-24:
The **plan** method returns a [MoveItErrorCode ](http://docs.ros.org/api/moveit_msgs/html/msg/MoveItErrorCodes.html) object and not a boolean.  In the future please post these type of questions on [ros answers](https://answers.ros.org/questions/) :)

closed 2018-04-24 by jrgnicho

v4hn, 2018-04-25:
This change happened only rather recently in kinetic.
It would be nice to document this (with an explanation) in the new `MIGRATION.md` proposed in #833 .

davetcoleman, 2018-05-14:
I've just added this to the MIGRATION.md here: https://github.com/ros-planning/moveit/pull/833