moveit/moveit #860: Remove console_bridge from MoveIt! and standardize on ROS_LOGGER

davetcoleman, 2018-04-25:
As I've said before, I think its silly we have two different logging approaches in the MoveIt! project:

In ``moveit_core``:
> CONSOLE_BRIDGE_logInform("Hello world");

In the rest of ``moveit``:
> ROS_INFO_NAMED("moveit_ros", "Hello world");

I am seriously proposing we just do away with console_bridge in Melodic rather than all these [weird work arounds](https://github.com/ros-planning/moveit/pull/859) for the following reasons:

- Its confusing to developers that we have two standards. ROS_INFO is the standard all ROS devs learn
- The ideal of "no ROS in moveit_core" is false since it already depends on lots of ROS including lots of ROS messages, roslib, rostime, and URDF (see [package.xml](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_core/package.xml))
- We should still prohibit putting the ROS middleware in moveit_core but leverage basic utilities like logging
- ROS_LOGGER is *way* more powerful that console_bridge - the throttling, namespacing, and "once" feature are really useful for low level libraries
- ROS_LOGGER allows for piping strings together ("Value: " << value) while console does not
- console_bridge outputs very trashy info to the console and is hard to limit various types of output, unlike ROS_LOGGER
- I've never heard of anyone using moveit_core without ROS anyway. MoveIt! really is just a ROS "glue" framework for various third party plugins

I propose we create a pull request in Melodic that replaces all the console_bridge using a quick find-replace

IanTheEngineer, 2018-04-25:
+1 toward removing the console logger. I agree that there really isn't a need to keep `moveit_core` "ROS agnostic".

v4hn, 2018-04-26:
Fine by me.

@davetcoleman you proposed to use `_NAMED` for replacement.
Should all of these be `moveit_core` (for the core package)
or do you want to have more fine-grained namespaces?
I would be fine with the simple namespace(s).

davetcoleman, 2018-04-26:
I would prefer the namespace be each sub-folder or class name, e.g. 

```
ROS_INFO_NAMED("robot_model", "Loading URDF");
ROS_ERROR_NAMED("kinematic_base", "Invalid number of joints");
```

davetcoleman, 2018-05-01:
I believe this has been completed by @jeasinema, closing :)

closed 2018-05-01 by davetcoleman