moveit/moveit #865: test regression on ROS build farm

rhaschke, 2018-04-26:
#835 caused several test regressions on the ROS build farm:
http://build.ros.org/job/Kdev__moveit__ubuntu_xenial_amd64/257/
http://build.ros.org/job/Ldev__moveit__ubuntu_xenial_amd64/184/

@willcbaker Please have a look and fix them asap. Otherwise, we should revert those changes.
Obviously Travis doesn't catch these test failures. @davetcoleman Do you have an idea why?

rhaschke, 2018-04-26:
Why do you think so? How do your changes depend on `moveit_resources` changes?
I rather think, it's related to different namespace setups. The failure is in all cases in the same line of code (however you duplicated the original unittest code two times, which isn't nice as well).
The failure is related to the fact, that the trajectory wasn't actually executed. But I don't see why yet.

willcbaker, 2018-04-26:
@rhaschke seen that exact behavior; is fixed due to forcing a global namespace in the launch files

rhaschke, 2018-04-27:
I see. I triggered new releases for [Kinetic](https://github.com/ros/rosdistro/pull/17647) and [Lunar](https://github.com/ros/rosdistro/pull/17646).
@willcbaker Please consider my remaining remarks in #835.

closed 2018-04-27 by rhaschke