moveit/moveit #867: soft vs. hard joint limits

rhaschke, 2018-04-27:
We currently run into issues, because our robot essentially has two kinds of joint limits: hard and soft:
By applying some external forces (e.g. grasping an object or deliberately touching the environment), joints can be pushed up to the hard limits. However, in free motion the joints can only be actuated up to the soft limits.
As far as I can see, MoveIt only supports a single type of bounds. However, I think one should use soft bounds for sampling (i.e. potential motion generation) and hard bounds for actual collision checking.
@davetcoleman @v4hn What do you think?

v4hn, 2018-04-27:
I'm pretty sure you are talking about the Shadow Hand here.
I just noticed the same with a hand the last 2-3 days in a small project I'm working on and wanted to propose the same. :-)

Short answer: Yes! We need support for soft-limits in MoveIt.

Longer answer:

There is already a way to specify such limits in the URDF format:
http://wiki.ros.org/urdf/XML/joint
I'm not sure I like the overall `<safety_controller>` tag too much though.

Alternatively we could include soft limits in `joint_limits.yaml`.
But as long as we don't pass in the `RobotModel` to our IK plugins,
the plugins will still ignore them because they only look at the URDF.

Note that we have a very crude kind of support for start states that are out of limits right now:
If the joints are "close enough" to the limits we just pretend they are *at* the limits instead.

Some places to consider:

- state validity checks should use hard limits
- start states should allow planning from hard limits, but only to soft-limit targets
- IK should use soft limits

I'm sure there are more.

mmoerdijk, 2018-05-02:
Just to illustrate that is not only a problem with the Shadow Hand, we have the same problem with our robot. Wenn in gravity compensation mode, it can be moved by hand to teach positions and it is possible to teach positions that are inside the soft limits. Most of the time these positions are just inside the soft limits and it would not be a problem to let the robot move to this position.  

So I would like to propose that allowing soft-limit targets should be configurable together with a allowed margin. This would have the advantage that during the motion the robot doesn't hit the softjoint limits but it would be able to reach the teached end position

bmagyar, 2018-05-02:
In my view soft limits can the interpreted as the recommended margin before
hard limits. Not sure if replacing one for the other is a complete
solution. How about using both and inferring margins from that?

guihomework, 2018-05-02:
@v4hn we had seen the soft_limits in the safety controller, but are not sure how this will be handled by all the software depending on the joint limit. 

Not only MoveIt! has to handle both soft and hard limits, but controllers too, https://github.com/ros-controls/ros_controllers/blob/kinetic-devel/effort_controllers/src/joint_position_controller.cpp#L258 because hard limits might not be reachable by the controller, or should be clamped to them.

Seems like the joint_trajectory_controller GUI already reads this soft limit values by default
https://github.com/ros-controls/ros_controllers/blob/a9e650737d6d04d8a768c84a48763bdbc5e05800/rqt_joint_trajectory_controller/src/rqt_joint_trajectory_controller/joint_limits_urdf.py#L14

I agree that IK should not give a valid result out of soft limits either, but FK should be possible using values within hard limit range (to know where the end effector really is)