moveit/moveit #872: Compile error in kdl_kinematics_plugin.cpp: cannot declare variable ‘ik_solver_pos’ to be of abstract type.

DataCorrupted, 2018-04-28:
### Description

I was trying to install a ros on tinker board(Arch: Armhf). Not much support for that though, so I have to compile almost everything from scratch. Moveit was one of them, then I got this error(Full consel output is listed below):

/home/linaro/tmp_ws/moveit_ws/src/moveit/moveit_kinematics/kdl_kinematics_plugin/src/kdl_kinematics_plugin.cpp:468:37: error: cannot declare variable ‘ik_solver_pos’ to be of abstract type ‘KDL::ChainIkSolverPos_NR_JL_Mimic’
   KDL::ChainIkSolverPos_NR_JL_Mimic ik_solver_pos(kdl_chain_, joint_min_, joint_max_, fk_solver, ik_solver_vel,

### Your environment
* ROS Distro: Kinetic
* OS Version: Debian Jessis
* Source build
* Using kinetic-devel branch.

### Steps to reproduce
I was using 

catkin_make --make-args -j1

### Backtrace or Console output
[ 75%] Building CXX object moveit/moveit_kinematics/kdl_kinematics_plugin/CMakeFiles/moveit_kdl_kinematics_plugin.dir/src/kdl_kinematics_plugin.cpp.o
/home/linaro/tmp_ws/moveit_ws/src/moveit/moveit_kinematics/kdl_kinematics_plugin/src/kdl_kinematics_plugin.cpp: In member function ‘bool kdl_kinematics_plugin::KDLKinematicsPlugin::searchPositionIK(const Pose&, const std::vector<double>&, double, std::vector<double>&, const IKCallbackFn&, moveit_msgs::MoveItErrorCodes&, const std::vector<double>&, const kinematics::KinematicsQueryOptions&) const’:
/home/linaro/tmp_ws/moveit_ws/src/moveit/moveit_kinematics/kdl_kinematics_plugin/src/kdl_kinematics_plugin.cpp:466:36: error: cannot declare variable ‘ik_solver_vel’ to be of abstract type ‘KDL::ChainIkSolverVel_pinv_mimic’
   KDL::ChainIkSolverVel_pinv_mimic ik_solver_vel(kdl_chain_, joint_model_group_->getMimicJointModels().size(),
                                    ^~~~~~~~~~~~~
In file included from /home/linaro/tmp_ws/moveit_ws/src/moveit/moveit_kinematics/kdl_kinematics_plugin/include/moveit/kdl_kinematics_plugin/kdl_kinematics_plugin.h:57:0,
                 from /home/linaro/tmp_ws/moveit_ws/src/moveit/moveit_kinematics/kdl_kinematics_plugin/src/kdl_kinematics_plugin.cpp:37:
/home/linaro/tmp_ws/moveit_ws/src/moveit/moveit_kinematics/kdl_kinematics_plugin/include/moveit/kdl_kinematics_plugin/chainiksolver_vel_pinv_mimic.hpp:47:7: note:   because the following virtual functions are pure within ‘KDL::ChainIkSolverVel_pinv_mimic’:
 class ChainIkSolverVel_pinv_mimic : public ChainIkSolverVel
       ^~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/linaro/work_space/install_isolated/include/kdl/chainiksolvervel_pinv.hpp:25:0,
                 from /home/linaro/tmp_ws/moveit_ws/src/moveit/moveit_kinematics/kdl_kinematics_plugin/include/moveit/kdl_kinematics_plugin/kdl_kinematics_plugin.h:53,
                 from /home/linaro/tmp_ws/moveit_ws/src/moveit/moveit_kinematics/kdl_kinematics_plugin/src/kdl_kinematics_plugin.cpp:37:
/home/linaro/work_space/install_isolated/include/kdl/chainiksolver.hpp:92:22: note: 	virtual void KDL::ChainIkSolverVel::updateInternalDataStructures()
         virtual void updateInternalDataStructures()=0;
                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/linaro/tmp_ws/moveit_ws/src/moveit/moveit_kinematics/kdl_kinematics_plugin/src/kdl_kinematics_plugin.cpp:468:37: error: cannot declare variable ‘ik_solver_pos’ to be of abstract type ‘KDL::ChainIkSolverPos_NR_JL_Mimic’
   KDL::ChainIkSolverPos_NR_JL_Mimic ik_solver_pos(kdl_chain_, joint_min_, joint_max_, fk_solver, ik_solver_vel,
                                     ^~~~~~~~~~~~~
In file included from /home/linaro/tmp_ws/moveit_ws/src/moveit/moveit_kinematics/kdl_kinematics_plugin/include/moveit/kdl_kinematics_plugin/kdl_kinematics_plugin.h:56:0,
                 from /home/linaro/tmp_ws/moveit_ws/src/moveit/moveit_kinematics/kdl_kinematics_plugin/src/kdl_kinematics_plugin.cpp:37:
/home/linaro/tmp_ws/moveit_ws/src/moveit/moveit_kinematics/kdl_kinematics_plugin/include/moveit/kdl_kinematics_plugin/chainiksolver_pos_nr_jl_mimic.hpp:46:7: note:   because the following virtual functions are pure within ‘KDL::ChainIkSolverPos_NR_JL_Mimic’:
 class ChainIkSolverPos_NR_JL_Mimic : public ChainIkSolverPos
       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/linaro/work_space/install_isolated/include/kdl/chainiksolvervel_pinv.hpp:25:0,
                 from /home/linaro/tmp_ws/moveit_ws/src/moveit/moveit_kinematics/kdl_kinematics_plugin/include/moveit/kdl_kinematics_plugin/kdl_kinematics_plugin.h:53,
                 from /home/linaro/tmp_ws/moveit_ws/src/moveit/moveit_kinematics/kdl_kinematics_plugin/src/kdl_kinematics_plugin.cpp:37:
/home/linaro/work_space/install_isolated/include/kdl/chainiksolver.hpp:57:22: note: 	virtual void KDL::ChainIkSolverPos::updateInternalDataStructures()
         virtual void updateInternalDataStructures()=0;
                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
moveit/moveit_kinematics/kdl_kinematics_plugin/CMakeFiles/moveit_kdl_kinematics_plugin.dir/build.make:62: recipe for target 'moveit/moveit_kinematics/kdl_kinematics_plugin/CMakeFiles/moveit_kdl_kinematics_plugin.dir/src/kdl_kinematics_plugin.cpp.o' failed
make[2]: *** [moveit/moveit_kinematics/kdl_kinematics_plugin/CMakeFiles/moveit_kdl_kinematics_plugin.dir/src/kdl_kinematics_plugin.cpp.o] Error 1
CMakeFiles/Makefile2:18513: recipe for target 'moveit/moveit_kinematics/kdl_kinematics_plugin/CMakeFiles/moveit_kdl_kinematics_plugin.dir/all' failed
make[1]: *** [moveit/moveit_kinematics/kdl_kinematics_plugin/CMakeFiles/moveit_kdl_kinematics_plugin.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2

DataCorrupted, 2018-04-28:
Basic assumption: This code must be working, or it won't be a distro used by so many users. Is there any possibility that I am using the wrong compiler?

rhaschke, 2018-04-28:
You are right, this code compiles nicely on all continuous integration servers. Probably you build chain is broken.

rhaschke, 2018-05-17:
I got this issue on Ubuntu Bionic too. The actual reason is, that KDL 1.4 changed API. There will be a fix for this in MoveIt Melodic: #902.

closed 2018-05-17 by rhaschke