moveit/moveit #873: Moveit Commander: turning off self collision

faseehahmad, 2018-04-28:
I am using moveit commander to perform simultaneous two handed manipulation using baxter robot. Currently, the problem I am facing is that, whenever the I command both arms to reach to a position where they are close to each other, the planner fails to give a plan. I think this is because, moveit plans for both the  arms together and since the arms are really close to each other, it fails to obtain a plan.
Any ideas what I should do? How can I turn off self collision so that the planner ignores the collision between arms?
I am doing this is simulation.
Thanks

BryceStevenWilley, 2018-04-29:
> How can I turn off self collision so that the planner ignores the collision between arms?

This is a very common thing, but usually is only done between adjacent links or links that will never collide. It can be done by adding a `disable_collisions` tag in the robot's srdf, see [this example](https://github.com/ros-industrial/universal_robot/blob/kinetic-devel/ur5_moveit_config/config/ur5.srdf#L40) in the ur5 as an example. 

Personally, I wouldn't recommend turning off self collision for this particular application, and instead trying to use a longer planning time, or a different planner than the default, like RRT*, something that can give you a more detailed/optimal motion.

faseehahmad, 2018-04-29:
Hello Bryce, I tried as you suggested by adding the link names. But, the
problem is that I am doing planning for both arms together. Although, i
have disabled the collisions but the collision model of the robot is still
the same and while planning it would still consider the arms as obstacles
right?

BryceStevenWilley, 2018-04-30:
So adding the `disable_collisions` tag in the srdf only disables collisions between certain pairs of links, say link A and link B. The collision model of the robot is still the same (link A and link B still are a part of the robot) but just those pairs specified (A and B) won't be checked. The links will still collide with everything else in the scene and with other links, i.e. A and B will still collide with link C.

faseehahmad, 2018-04-30:
Ok got it. Thanks for you help