moveit/moveit #886: Next Kinetic Release

davetcoleman, 2018-05-05:
Its been 5 months since our last Kinetic release and a lot of changes have been happening. @130s any thoughts on when we should make the next release? Seems like a good time to me.

BoltzmannBrain, 2018-05-09:
What can I do to help speed this along??

130s, 2018-05-09:
I'll save the time this weekend. I'm willing to make releases for Indigo too as it's easier to make releases at the same time across distros, if that's okay.

v4hn, 2018-05-15:
I very much agree that we need to release very soon - best this month.

We have at least one open regression with #868 that should be fixed before the next release.
In my opinion this blocks the release.
If it's not fixed until the end of the month, we can release anyway though because it really has been a long time and we introduced the regression in the last release already.

rhaschke, 2018-05-15:
@v4hn I already suggested two solutions and was asking for feedback to resolve #868. Would be great to get some soon ;-)

130s, 2018-05-17:
- Maintainers, please hold merges for Indigo and Kinetic (except critical ones).
- MoveIt!'s release into Melodic is approaching to realization https://github.com/ros-planning/moveit/issues/806#issuecomment-389760536. It'd be very helpful for release task if the release for all distros can be done at once. But I can wait until end of May (as I just asked for holding merge, release remains easy).

gavanderhoorn, 2018-05-17:
It would be very nice if we could get #853 in Indigo. I know I opened it, and it's currently failing, but the new parameterisation plugin creates some really smooth paths.

I'll see if I can find some time (very) soon to fix the tests (because those are failing right now).

130s, 2018-05-18:
@v4hn @gavanderhoorn ok, I took the tickets you mentioned as a "requirement" for the next release. Ping when they are done.

Maintainers, merge-hold https://github.com/ros-planning/moveit/issues/886#issuecomment-389766585 is raised (well there have been already some merges anyways. Clearly I'm not heard :/ ).

rhaschke, 2018-05-18:
As far as I can see, now new commits on Indigo or Kinetic since https://github.com/ros-planning/moveit/issues/886#issuecomment-389766585. However, there is activity on Melodic, as it was (and still is) not yet ready for release...

mikaelarguedas, 2018-05-29:
Will there be a Lunar release along the Kinetic one ?

MoveIt! is currently broken in Lunar due to a change in OMPL: see https://github.com/ros-gbp/ompl-release/issues/14 for details

I was planning on syncing lunar later this week.
I would like to know what your preference is between (sorted from the most appealing to me to the least appealing):
- getting a new Lunar release out by then including https://github.com/ros-planning/moveit/pull/907 (I can postpone the sync for a couple days if this allows you to release)
- revert the OMPL release for now to keep moveit building and let you synchronize with @mamoll for the next release
- letting: moveit, moveit_planners_ompl, moveit_planners, moveit_runtime regress and be out of the repos for a cycle

rhaschke, 2018-05-29:
@mikaelarguedas Yes, we just agreed in the maintainer meeting to have a Kinetic and Lunar release prepared asap. I will do so tomorrow latest. There is one regression that needs a fix...

mikaelarguedas, 2018-05-29:
awesome thanks @rhaschke !

rhaschke, 2018-05-31:
MoveIt 0.9.12 was successfully released into [Kinetic](http://repositories.ros.org/status_page/ros_kinetic_default.html?q=%5Emoveit_) and [Lunar](http://repositories.ros.org/status_page/ros_lunar_default.html?q=%5Emoveit_)!

gavanderhoorn, 2018-05-31:
I don't have the time to look at #853 before the end of next week, and I don't want to hold up a release into Indigo. So please go ahead with an Indigo release without that PR merged.

The CI is failing on a method in `robot_trajectory::RobotTrajectory` not being found. A quick look didn't lead me to any solution though.

troizky, 2018-07-18:
There is a bug in method `collision_detection::distanceCallback` in file moveit_core/collision_detection_fcl/src/collision_common.cpp #988
In the following lines:
```
    dist_result.link_names[0] = cd1->ptr.obj->id_;
    dist_result.link_names[1] = cd2->ptr.obj->id_;
```
method CollisionGeometryData::getID() should be used for `cd1` and `cd2`.

BoltzmannBrain, 2018-07-18:
@troizky thanks for bringing this up. What implies this is a bug? Did you run into an error?

troizky, 2018-07-19:
Yes, I did.
I got a crash with exception 'bad_alloc' and provided backtrace in open issue #988  https://github.com/ros-planning/moveit/issues/988  
because of access to object of type AttachedBody as World::Object which is incorrect.
And my fix definitely solved this problem.

>
>@troizky thanks for bringing this up. What implies this is a bug? Did you run into an error?

closed 2018-08-20 by rhaschke