moveit/moveit #889: Continue ROS Lunar release?

130s, 2018-05-13:
Your voice is appreciated.

May or may not be related to [this on Discourse](https://discourse.ros.org/t/proposed-changes-to-the-ros-releases/4736/1) where dropping non-LTS ROS distro is suggested for the future release, which is based on the fact that very few people use non-LTS ROS, do we still want to make releases for Lunar? It'll soon be no longer the latest distro once Melodic release is done #866. FYI its EOL is May 2019.

Making a new release for a ROS distro takes about half an hour person-hour wise (assumption *1) and a few hours lead-time, so it's not terribly costly. It's more of a mental burden (to me). But I don't want to disappoint existing usecases.

*1. If it's done at the same time subsequently from a release for other distro so that common processes, e.g. creating changelog, can be skipped)

rhaschke, 2018-05-13:
As the Lunar and Kinetic releases are currently identical, I suggest to still do a Lunar release. The extra effort is rather limited. I guess, it's less than half an hour in this case.
However, you are right, probably there are not so many people out there, using a Lunar release...

130s, 2018-05-13:
Thanks for the input.

> As the Lunar and Kinetic releases are currently identical, 

As far as the release goes, this isn't always the case, since the depended packages may vary. So it's always possible to run into an issue during release that no one has ever seen (I've updated my OP to note the risk regarding release just FYI).

davetcoleman, 2018-05-14:
My vote is to make one final Lunar release before we drop it in favor of Melodic

130s, 2018-05-29:
Also made [a callout on discourse](https://discourse.ros.org/t/moveit-future-release-plan-for-ros-lunar/4807).

Though closing as I think we've agreed as following, feel free to add comments:
- Make one last release and make an announcement to recommend Melodic.
- Bugfix release may still occur if needed.

closed 2018-05-29 by 130s