moveit/moveit #894: Setup Assistant 2

MohmadAyman, 2018-05-14:
### Description

Hello there, I'm Mohamad Ayman a Computer Engineering Masters student and a Google Summer of Code student.

My proposal this summer focuses on improving the Setup Assistant by automating the generation of some configurations and adding more GUIs to make the generation of some easier and more straight forward, along with improving the integration with Gazebo where we one wont have to add configs or edit the robot file manually for it to work with gazebo.

Below are the description of the improvements broken into tasks, I will update the tasks each time a related PR is opened, any suggestions or feedback are highly appreciated.

- [ ] New screen for automatically generating interfaces to low level controllers.

- [ ]  Improving gazebo integration.

- [ ] Perception screen for using laser scanner point clouds.

- [ ] New grasping screen for setting up end effectors.

- [ ] Parse existing text files [launch/config/package] and adding more GUIs for loading and saving settings.

simonschmeisser, 2018-06-11:
We are about to start working on our own "mikado_setup_assistant" (name not final) for our software, sharing plugins either way sounds like a great idea! (I.e. having a plugin system should hopefully also allow loading these in a different host application)

davetcoleman, 2018-06-15:
@MohmadAyman I would say the plugins should be a lower priority than the other features we are focusing on, but if there was time at the end of the summer then of course it sounds great

MohmadAyman, 2018-07-14:
Hi @simonschmeisser , have you guys started working on your "mikado_setup_assistant"?

davetcoleman, 2018-08-20:
@MohmadAyman please update this issue with your final updates, then we should consider closing it

MohmadAyman, 2018-08-20:
Done,

Can we leave it until the tutorials have been updated? so they don't get forgotten.

davetcoleman, 2018-10-17:
I believe this can now be closed. Congrats @MohmadAyman for a very successful GSoC project! Your continued involvement in MoveIt! including code review and code contributions are always welcomed :100:

closed 2018-10-17 by davetcoleman

MohmadAyman, 2018-12-08:
I've opened a PR for the Plugin based SA #1243 , would be nice to have your input.