moveit/moveit #900: rework of python interface

rhaschke, 2018-05-17:
Sorry for the lengthy issue. @bmagyar @davetcoleman @v4hn I'm asking for feedback on a proposal  formulated in the very end.

Working on a python interface for [MoveIt Task Constructor](https://github.com/ros-planning/moveit_task_constructor), I investigated some time to review and evaluate existing methods to pass data between C++ and Python. Currently, `moveIt_commander` relies on serialization/deserialization of ROS message types, both in python and C++ as follows:

  `Python msg type <--> std::string <--> C++ msg type`

However, this requires some boiler-plate code both, in C++ and Python, to call serialization/deserialization like this:

[C++](https://github.com/ros-planning/moveit/blob/ab0f70658c0da0c20e63d42d38df5f84acfe22af/moveit_ros/planning_interface/robot_interface/src/wrap_python_robot_interface.cpp#L230-L233):
```cpp
robot_state::RobotStatePtr s = current_state_monitor_->getCurrentState();
moveit_msgs::RobotState msg;
robot_state::robotStateToRobotStateMsg(*s, msg);
return py_bindings_tools::serializeMsg(msg);
```
[Python](https://github.com/ros-planning/moveit/blob/ab0f70658c0da0c20e63d42d38df5f84acfe22af/moveit_commander/src/moveit_commander/robot.py#L224-L228)
```python
from moveit_ros_planning_interface import _moveit_robot_interface
from moveit_msgs.msg import RobotState

ri = _moveit_robot_interface.RobotInterface(robot_description)
state = RobotState()
state.deserialize(ri.get_current_state())
```
Using `boost::python's` type conversion capabilities, I managed to get rid of explicit serialization/deserialization, which essentially will render most of the code in moveit_commander obsolete:

C++
```cpp
robot_state::RobotStatePtr s = current_state_monitor_->getCurrentState();
moveit_msgs::RobotState msg;
robot_state::robotStateToRobotStateMsg(*s, msg);
return msg;
```
Python
```python
from moveit_ros_planning_interface import _moveit_robot_interface
from moveit_msgs.msg import RobotState

ri = _moveit_robot_interface.RobotInterface(robot_description)
state = ri.get_current_state()
```

Additionally, I noticed that the python wrapping code is current spread across multiple packages: C++ code is located in `moveit_ros`:
* [py_bindings_tools](https://github.com/ros-planning/moveit/tree/ab0f70658c0da0c20e63d42d38df5f84acfe22af/moveit_ros/planning_interface/py_bindings_tools/src)
* [move_group_interface/src/wrap_python_move_group.cpp](https://github.com/ros-planning/moveit/blob/ab0f70658c0da0c20e63d42d38df5f84acfe22af/moveit_ros/planning_interface/move_group_interface/src/wrap_python_move_group.cpp)
* [robot_interface/src/wrap_python_robot_interface.cpp](https://github.com/ros-planning/moveit/blob/ab0f70658c0da0c20e63d42d38df5f84acfe22af/moveit_ros/planning_interface/robot_interface/src/wrap_python_robot_interface.cpp)
* [planning_scene_interface/src/wrap_python_planning_scene_interface.cpp](https://github.com/ros-planning/moveit/blob/ab0f70658c0da0c20e63d42d38df5f84acfe22af/moveit_ros/planning_interface/planning_scene_interface/src/wrap_python_planning_scene_interface.cpp)

... while corresponding python code is located in `moveit_commander`. 
To facilitate maintenance, **I suggest to join corresponding C++ and python code** within the package that provides the corresponding functionality, using a common folder structure (which could be enforced using appropriate cmake convenience functions) like this:

```
* <package name>/python
  * wrapper
    * CMakeLists.txt
    * include/moveit/python/<package name>/*.h
    * src/*.cpp
  * src/moveit/<package name>
    * __init__.py
    * *.py
```
The generic conversion code, I would suggest to place either into `moveit_core/python` or into a new repo `moveit_python_tools`. **What do you think about this idea?**

bmagyar, 2018-05-17:
+100 for packages providing their own python interfaces. It also keeps them
in sight so they are less likely to become dead code.

Sorry for the brevity, I'm on the go.
One thing I really like about the current python interfaces is that while
it's a wrapper and does mostly forwarding to C++, it uses native Python ROS
message types so when working with python you don't have to convert
messages to feed C++types to the python wrappers. Is this functionality
maintained with the boost alternative?

rhaschke, 2018-05-17:
Conceptually we keep the same serialization/deserialization approach with ROS msgs, but w/o explicit need to call serialization/deserialization functions. That's done magically under the hood.

davetcoleman, 2018-05-29:
I can't speak too much to python as I've done a pretty good job of avoiding it over the years in favor of always C++. Still my thoughts:

> but w/o explicit need to call serialization/deserialization functions. That's done magically under the hood.

Is there a performance improvement then, or does the same serialization still occur but just more obfuscated? If no performance difference, is the slightly improved syntax (and just 1 fewer lines of code) warrant the effort?

I like your thoughts about cleaning up the location of python wrapper code a lot! I'd be interested to hear @wjwwood's thoughts on best practices of ROS packages with python wrappers around C++

v4hn, 2018-05-29:
@rhaschke The issue is not that lengthy really :)

:+1: to implicit boost::python conversion ... if applicable without making it much harder to debug problems.

:+1: to moving python and C++ code in together.

basically :+1: to anything that involves useful and easier Python module structure...

Where possible, official API should be kept.
If we move the code, we can still provide forward-imports from the commander.

For "generic conversion code" I would prefer a new *package* (not repo) `moveit_python_tools`.
As far as I know `moveit_core` has no python dependency as of yet and I don't see the need to change that.

mlautman, 2018-05-30:
+1 I am very interested in being part of this effort. 

Working with the python move_group, I've noticed that a lot of interfaces are hidden from the MoveGroupCommander class. For instance, `get_named_targets` and `get_named_target_values` are not exposed. To access those methods you have to call them through the internal variable ie: `group._g.get_named_targets`. Is there any reason for this? If not I'll open a small PR to expose those in the MoveGroupCommander class

wjwwood, 2018-05-30:
> I'd be interested to hear @wjwwood's thoughts on best practices of ROS packages with python wrappers around C++

Ideally (in my opinion), Python bindings would:

- not affect the C++ code at all (C++ code has no knowledge they exist)
  - and therefore are not required for C++ libraries to work
- be in a separate package, but in only one extra package if possible (not spread across multiple packages without good reason)
  - but it's also ok to have them in the same package as the C++ code, if it's not possible/realistic to use moveit without the Python interface

I don't think the layout across repositories matters that much, and it seems you guys have a preference for one large repository anyways.

The idea being that at some point someone might want to use moveit without using Python, and in that case they'd like to not pay for building, installing, depending on extra stuff to make the Python bindings work.

As for the particulars on how to do it, I don't really have time to dig into the details and make any suggestions

wjwwood, 2018-05-30:
Also, we're looking at using `pybind11` as a new way to make bindings for Python on top of C/C++ code, and I believe it's based on (if not the spiritual successor to) Boost.Python.

https://github.com/pybind/pybind11

rhaschke, 2018-05-30:
@wjwwood I agree to your argument that one might want to disable the python bindings, which calls for a separate package. Definitely, MoveIt can be used without python bindings.

davetcoleman, 2018-05-30:
``moveit_python_tools`` is a bit of a generic name that might fall under [REP 144](http://www.ros.org/reps/rep-0144.html) guidelines:

> using catchall names such as utils should be avoided as they do not scope what goes into the package or what should be outside the package

Ironically that is what I called ``moveit_visual_tools``, but that was many years ago and pre-dates REP 144.

I'd be more in favor of bringing in @mikeferguson  ``moveit_python`` instead of creating ``moveit_python_tools``

rhaschke, 2018-05-30:
@davetcoleman I think, `moveit_python` is quite different from current `moveit_commander` that I propose to refactor.
While `moveit_python` is a purely python-based library to interface the move_group via ROS msgs, the `moveit_commander` is a true wrapper of (some) C++ code. In principle this allows for using MoveIt libraries directly from python - without going over ROS. That's what the Task Constructor framework essentially does as well.
Hence, both python libraries are useful. They have partially overlapping use cases.

janEbert, 2018-06-21:
I have been working on this issue but am wondering how backwards compatibility is supposed to be handled.
Should all serialization methods and their counterpart methods that take and return `std::string`s instead of (in the future) `moveit_msgs` types be kept in favor of compatibility or removed in favor of readability and cleanliness? The same goes for the Python methods where we would have to check if each possibly serialized type is `str`.

janEbert, 2018-07-19:
If the `moveit_commander` package is supposed to be deprecated after the cleanup, where should the folder's Python code that relies on other interfaces be moved?
For example, [`interpreter.py`](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_commander/src/moveit_commander/interpreter.py) - which imports all the interfaces from `moveit_ros/planning_interface` - would either need to remain in `moveit_commander` or be placed in a new package, different from `moveit_python_tools`, as `moveit_ros/planning_interface` already depends on `moveit_python_tools`.

This could either mean that `moveit_commander` stays the way it is (except for low-level stuff such as [`conversions.py`](https://github.com/ros-planning/moveit/blob/kinetic-devel/moveit_commander/src/moveit_commander/conversions.py)) or that a new package specifically for non-deprecated high-level Python must be created.

How should this be handled?

felixvd, 2020-10-09:
I'll expand on this old issue, since I just read so many dealing with the Python interfaces ( #1947, #1966 (#2356), #2356, #2271...): Does anyone have insight about the design choices behind the interfaces, and if something in the last 8 years has changed in this regard?

I also saw that [Tesseract uses](https://github.com/ros-industrial-consortium/tesseract_ros/tree/master/tesseract_ros_python) [swig](http://www.swig.org/) to auto-generate (?) their interfaces. Maybe @Levi-Armstrong has something to share about their experience with it.

If anyone has background information please feel free to share. Pinging @mikeferguson because his `moveit_python` package was mentioned in the thread.

PeterMitrano, 2020-10-09:
In my experience, pybind11 is the way to go. Perhaps @mamoll can say a bit about his experience switching from boost python to pybind11. I have used pybind11 and rely on moveit python for my research, so I could help.

I agree with everything William said. Adding or changing bindings should not affect C++, and should not require changing things in multiple places.

One challenge is that uses will want to use Python ROS message objects to interact, as we do now, but those will not be converted to C++ ROS message objects without either (1) serialization like we have now, which is slow (2) or hand-written bindings for every message type we pass across the python/C++ boundary. See [https://github.com/dimatura/pyrosmsg](https://github.com/dimatura/pyrosmsg) for an investigation of these two options.

Total-auto-generating is probably not the way to go. Some things would not feel "pythonic" if we did it that way, I think.

Levi-Armstrong, 2020-10-09:
The nice thing about leveraging swig is it not only provide python but also other language like matlab, C#, etc.. Since ROS is now supported on Windows there may be future interest in these other language going forward. @johnwason was the one who created the ones for Tesseract and may have addition context related to swig advantages.

rhaschke, 2020-10-09:
I can only second @PeterMitrano, pybind11 is the spiritual successor of `boost::python` and provides a very pythonic API experience. I have already implemented a pybind11-based wrapper for MTC, which has all of the core features implemented that were proposed here initially. I looked into `swig` back then as well and wasn't that convinced...
I see, though, the benefits of handling multiple languages and automatic wrapper generation.

johnwason, 2020-10-09:
SWIG is a faster to use and more flexible than solutions like PyBind11, but it has a much higher learning curve. Both solutions will probably be obsolete once C++20 modules become more common. The modules should contain enough metadata for Python bindings to be generated with much less work.

PeterMitrano, 2020-10-10:
FWIW I just read that tensorflow is switching from swig to pybind11, no idea why though

[https://github.com/tensorflow/tensorflow/releases/tag/v2.2.0](https://github.com/tensorflow/tensorflow/releases/tag/v2.2.0https://github.com/tensorflow/tensorflow/releases/tag/v2.2.0)
> Export C++ functions to Python using pybind11 as opposed to SWIG as a part of our deprecation of swig efforts.

[https://github.com/tensorflow/community/blob/master/rfcs/20190208-pybind11.md](https://github.com/tensorflow/community/blob/master/rfcs/20190208-pybind11.md)

felixvd, 2020-10-10:
Interesting. Specifically this quote:

> ### Cons with SWIG
> 
> 1. SWIG auto-generated code is not optimal for performance.
> 2.  SWIG needs the swig toolchain.
> 3.  SWIG requires .i or .swig files which require knowledge of SWIG directives.
>     1.  We've changed .swig to .i files to get around readability issues
> 4.  SWIG combined with C++ is relatively complicated to understand.

If I understand correctly, the cons for both `boost::python` and pybind11 are the manual maintenance (new/updated functions are not exposed automatically) and being limited to Python. Auto-generating code for MATLAB and other languages (Julia, Scratch...) is definitely charming though.

I also agree that our current Python code being spread over multiple packages is confusing. Can we move all the files including the `wrap_python_move_group.cpp`, `wrap_python_planning_scene_interface.cpp` wrappers into one package such as `moveit_python_bindings`, or is there a problem I am overlooking?

v4hn, 2020-10-10:
> I also agree that our current Python code being spread over multiple packages is confusing.
> Can we move all the files [...] into one package such as `moveit_python_bindings`, or is there a problem I am overlooking?

I don't know any strong reason for either approach.
It makes sense for the bindings to stay with their respective C++ module.

But given that all our official Python interfaces are exposed through `moveit_commander`, we could keep the bindings there as well..

All assuming we had sufficient test coverage for the python bindings to notice breakage in a central python-interface package.

PeterMitrano, 2020-10-10:
If we did distributed, I _think_ that would mean the python user would be writing `from moveit_ros_planning import ` and `from moveit_other_module import`. That might obey the principle of least surprise for new users, although existing users probably expect everything under `moveit_commander`.

Tensorflow seems to do it under `tensorflow/python` which seems akin to `moveit/moveit_commander` here.

felixvd, 2021-02-28:
> I have already implemented a pybind11-based wrapper for MTC, which has all of the core features implemented that were proposed here initially. I looked into `swig` back then as well and wasn't that convinced...
@rhaschke Could you point me to the wrapper? Also, how much work do you reckon it would be to write the Python bindings with pybind11 in the same manner? About 175 hours, by any chance?

Our current Python bindings only cover the `MoveGroupInterface`, a tiny fraction of the `PlanningSceneInterface` and (strangely) the `RobotState`, and I think the latter doesn't use the Pimpl pattern unlike the other two. That leaves the majority of `moveit::core` inaccessible from Python - most notably `PlanningScene`, but also `KinematicsBase`, `RobotModel` etc.

I have a few prototyping workflows that need more bindings, and I was tempted to write them for the `PlanningScene`, but our existing method is evidently outdated, so I'd rather try my hand with pybind11, but I think I would take too long without a good example to copy from.

rhaschke, 2021-02-28:
Hi @felixvd, actually pybind11 bindings are not too far from boost::python. A good starting point might be the python binding I created for MTC: https://github.com/ros-planning/moveit_task_constructor/tree/wip-python-api. As I started from boost::python there as well, you should have a look at the [migration commit](https://github.com/ros-planning/moveit_task_constructor/commit/c97b46202433cd0f7d18936cf239798510c848b5). Next week, after IROS deadline and a project review, I will turn back to MTC and push this a little more forward... Also, there is excellent documentation available for pybind11.

felixvd, 2021-02-28:
Ah, I didn't notice the branch, thanks. I don't know boost::python either, so I would mostly start from scratch on this. That commit looks very instructive though.

What is your estimate of the workload? If it sounds doable within 175 hours, I'd like to add it to the list as a GSoC project.

PeterMitrano, 2021-02-28:
I'm also happy to help write bindings with pybind11, assuming we're doing it all by hand. I don't have experience with bindings for a large project like moveit, so perhaps we need an automatic generation method like what OMPL uses, but TBH that method has been a nightmare for me to work with. So it's not a terrible idea to start with writing them manually.

I also have some examples of pybind11 bindings for moveit code, but it's private on github so I have to see if I can make it public before sharing.

PeterMitrano, 2021-02-28:
As for package structure, not sure it much matters but this is what I use (again I'll link an exmple once I get permission)

* package_name
  * src
    * cpp source code
    * bindings.cpp <--- this is where bindings for the module go
    * package_name
      * `__init__.py`
      * python code, can be useful to add stuff on top of the bindings
  * include
    * cpp header files

PeterMitrano, 2021-02-28:
Here's the repo I've been working on which uses moveit and has pybind11 bindings
https://github.com/UM-ARM-Lab/arm_robots/tree/master/jacobian_follower

rhaschke, 2021-02-28:
> What is your estimate of the workload? If it sounds doable within 175 hours, I'd like to add it to the list as a GSoC project.

I guess much less. Working focussed, migration of existing code should be done within a week. Of course, the work can always be extended to wrap more of the existing c++ code.
Note also, that a student of mine, @janEbert already did a lot of this migration work three years ago: https://github.com/janEbert/moveit/tree/e5a472f16cf206f1acbb32b3650c503cdc612313

@janEbert, I couldn't find a branch of your work. Could you please verify?

felixvd, 2021-03-05:
@PeterMitrano MoveIt Day is coming up. If you write Python bindings for the `PlanningScene` on MoveIt Day, I'll do a full review within 2 weeks or donate 50$ to EFF (or another org of the kind).

PeterMitrano, 2021-03-05:
Sounds like a win-win for everyone. Although I would be nice to know if I'm duplicating with already done by @janEbert  ?

PeterMitrano, 2021-03-07:
@rhaschke on the topic of cpp msg <--> py msg, you say you have it working without any boilerplate via boost python stuff? Can you share how that was done?

There's also this, https://github.com/UM-ARM-Lab/pyrosmsg which is very heavy boilerplate, but the result is extremely clean to use. If you want python bindings for `robotStateToRobotStateMsg`, you just declare it and the pyrosmsg type casters will make it so you can call it with a python RobotState msg and it will do the conversion. Similarly if you pybind a function returning a C++ msg it will be converting to a python message, without any boilerplate in the bindings defintion. I've been using this and it's really nice. The major downside is writing the typecasters by hand, but it's definitely manageable, and perhaps even automatable.

PeterMitrano, 2021-03-07:
I've started some work on this here: https://github.com/UM-ARM-Lab/moveit/tree/planning-scene-pybindings
So far I'm pretty optimistic on going forward with this. I'm also collecting some questions/thoughts here: https://docs.google.com/document/d/1ac-7eD0TgTCOc1SB9KStB0KuhLSm_ZjF1wofBEebxNo/edit?usp=sharing

Here's a teaser of what you can do with it
```
import pymoveit
from pymoveit.planning_scene import PlanningScene
from pymoveit.robot_model import load_robot_model

urdf_path = '/opt/ros/noetic/share/moveit_resources_pr2_description/urdf/robot.xml'
srdf_path = '/opt/ros/noetic/share/moveit_resources_pr2_description/srdf/robot.xml'

urdf_model = pymoveit.urdf.parseURDFFile(urdf_path)
print(urdf_model.getName())

model = load_robot_model(urdf_path, srdf_path)
scene = PlanningScene(model)
```

janEbert, 2021-03-08:
Hi Peter, sorry I took so long to respond.
I believe I still have the branch locally, but keep in mind my changes are probably already bitrotten and I'm just going to guess that you have better fundamental knowledge of ROS and the MoveIt! project which I didn't at the time. :)
Since you've already made progress, I assume you don't want me to find the branch but if there's interest, I'll get to it (and faster than I got to this thread).

rhaschke, 2021-03-08:
@PeterMitrano, I looked into https://github.com/UM-ARM-Lab/pyrosmsg: As you stated, it's a lot of manual code writing for the type casters. What I have done for MTC, is using the serialization-deserialization approach mentioned (as slow) in your repo. However, the code is generic and very concise: [ros_types.h](https://github.com/ros-planning/moveit_task_constructor/blob/wip-python-api/core/include/moveit/python/python_tools/ros_types.h)
Additionally, you could still have more specialized casters, e.g. from your repo, to directly convert some types. From the user's point of view, its exactly the same as in your referenced repo: types are silently converted behind the scenes.

rhaschke, 2021-03-08:
@janEbert: It would be great if you could make your branch available. Thanks a lot.

janEbert, 2021-03-08:
I re-uploaded the [python-cleanup branch](https://github.com/janEbert/moveit/tree/python-cleanup).
My head was at `python-cleanup~2`, so I'm assuming something was up with tests not working. Sadly, I cannot remember. :/

PeterMitrano, 2021-03-08:
Thanks jan. Based on what I see there, your branch is a simply applying boost serialization method to the existing bindings and moving some files around. In that case it's sort of orthogonal to making bindings for the rest of moveit.

That said, you are using an serialization based automatic casting method like @rhaschke is suggesting. If I understand correctly, we can use the serialization based type casters as the default and perhaps still have the manually written ones (which might be faster)? something something sfinae template magic?

I am happy to adopt the serialization based method. If performance is a problem we can copy in our own hand-written type casters.

rhaschke, 2021-03-08:
@PeterMitrano, you are right, Jan's task was to clean up the existing python bindings (employing automatic type converters). If you aim for new binding, this is indeed orthogonal.
Combining the serialization-based casting method as a default and manually-written specializations is perfectly feasible. I don't think this requires SFINAE magic. It's just template specialization.

PeterMitrano, 2021-03-08:
Yes for now I am working on new bindings, since Felix proposed we starting with moveit core, but I would hope that if we like where things are going can overhaul the existing bindings for MoveGroup and PlanningSceneInterface. Currently, those are written with boost python and require a lot of plumbing. My recommendation would be to replace those with pybind11, and reorganize the bindings (as the original issue suggests) so that bindings for a package are within that package.

PeterMitrano, 2021-03-10:
I'd like to get PR in today -- What do y'all think is a good scope for the first PR? What kind of testing should there be? I've nearly the entire "PlanningScene" tutorial ported to Python, but perhaps there are unit tests we should port as well? or maybe that's pointless because it's just bindings? For the first PR I do plan to only make additions, no changes or deletions to existing code, FWIW.

PeterMitrano, 2021-03-14:
@felixvd The PR is now mostly ready, so I'd appreciate your feedback. no rush!

PeterMitrano, 2021-04-25:
(let me know if we should start a new issue for this)

Turns out the recently merged #2547 has some limitations, right now I'm not able to extend that work to other packages because of how the devel-space python stuff works.

**The Problem**

We want to be able to write

```
from moveit import core       # corresponds to moveit_core
from moveit import planning # corresponds to moveit_ros_planning
```
To do this, we would need a `moveit` folder in the devel space (`lib/python3/dist-packages/moveit`) and the `__init__.py` inside it to add to sys.path the src directories of  *all* the moveit python packages. From what I can tell catkin does not support this.

**Brainstorming some possible solutions**

1. give up on using a namespace package, just name the python packages based on their actual ros package names, i.e
```
import  moveit_core
import moveit_ros_planning
```
2. do the above, but then also try to add some magic in the `moveit` metapackage that makes all of these above packages accessible via the current naming convention
3. move all the bindings to a separate package. Then it's not a namespace package, it's just a normal python package with a bit of extra nested structure. I think this might be the cleanest solution?

I haven't prototyped any of these so I'd appreciate feedback. There are probably better soultions I'm missing
@felixvd @mikeferguson @rhaschke