moveit/moveit #901: MoveIt needs adaptions to OMPL 1.4 API

rhaschke, 2018-05-17:
Melodic uses OMPL 1.4, which has some API changes that make `moveit_planners_ompl` fail:
@mamoll @davetcoleman Your help is appreciated on this.

```
Errors     << moveit_planners_ompl:make /root/ws_moveit/logs/moveit_planners_ompl/build.make.002.log                                                                                                                        
In file included from /root/ws_moveit/src/moveit/moveit_planners/ompl/ompl_interface/src/model_based_planning_context.cpp:44:0:
/root/ws_moveit/src/moveit/moveit_planners/ompl/ompl_interface/include/moveit/ompl_interface/detail/projection_evaluators.h:56:68: error: ‘ompl::base::EuclideanProjection’ has not been declared
   virtual void project(const ompl::base::State* state, ompl::base::EuclideanProjection& projection) const;
                                                                    ^~~~~~~~~~~~~~~~~~~
/root/ws_moveit/src/moveit/moveit_planners/ompl/ompl_interface/include/moveit/ompl_interface/detail/projection_evaluators.h:73:68: error: ‘ompl::base::EuclideanProjection’ has not been declared
   virtual void project(const ompl::base::State* state, ompl::base::EuclideanProjection& projection) const;
                                                                    ^~~~~~~~~~~~~~~~~~~
/root/ws_moveit/src/moveit/moveit_planners/ompl/ompl_interface/src/model_based_planning_context.cpp: In member function ‘virtual ompl::base::ProjectionEvaluatorPtr ompl_interface::ModelBasedPlanningContext::getProjectionEvaluator(const string&) const’:
/root/ws_moveit/src/moveit/moveit_planners/ompl/ompl_interface/src/model_based_planning_context.cpp:106:88: error: invalid new-expression of abstract class type ‘ompl_interface::ProjectionEvaluatorLinkPose’
       return ob::ProjectionEvaluatorPtr(new ProjectionEvaluatorLinkPose(this, link_name));
                                                                                        ^
In file included from /root/ws_moveit/src/moveit/moveit_planners/ompl/ompl_interface/src/model_based_planning_context.cpp:44:0:
/root/ws_moveit/src/moveit/moveit_planners/ompl/ompl_interface/include/moveit/ompl_interface/detail/projection_evaluators.h:49:7: note:   because the following virtual functions are pure within ‘ompl_interface::ProjectionEvaluatorLinkPose’:
 class ProjectionEvaluatorLinkPose : public ompl::base::ProjectionEvaluator
       ^~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /opt/ros/melodic/include/ompl/base/StateSpace.h:43:0,
                 from /root/ws_moveit/src/moveit/moveit_planners/ompl/ompl_interface/include/moveit/ompl_interface/parameterization/model_based_state_space.h:40,
                 from /root/ws_moveit/src/moveit/moveit_planners/ompl/ompl_interface/include/moveit/ompl_interface/model_based_planning_context.h:40,
                 from /root/ws_moveit/src/moveit/moveit_planners/ompl/ompl_interface/src/model_based_planning_context.cpp:39:
/opt/ros/melodic/include/ompl/base/ProjectionEvaluator.h:146:26: note: 	virtual void ompl::base::ProjectionEvaluator::project(const ompl::base::State*, Eigen::Ref<Eigen::Matrix<double, -1, 1> >) const
             virtual void project(const State *state, Eigen::Ref<Eigen::VectorXd> projection) const = 0;
                          ^~~~~~~
/root/ws_moveit/src/moveit/moveit_planners/ompl/ompl_interface/src/model_based_planning_context.cpp:146:82: error: invalid new-expression of abstract class type ‘ompl_interface::ProjectionEvaluatorJointValue’
       return ob::ProjectionEvaluatorPtr(new ProjectionEvaluatorJointValue(this, j));
                                                                                  ^
In file included from /root/ws_moveit/src/moveit/moveit_planners/ompl/ompl_interface/src/model_based_planning_context.cpp:44:0:
/root/ws_moveit/src/moveit/moveit_planners/ompl/ompl_interface/include/moveit/ompl_interface/detail/projection_evaluators.h:66:7: note:   because the following virtual functions are pure within ‘ompl_interface::ProjectionEvaluatorJointValue’:
 class ProjectionEvaluatorJointValue : public ompl::base::ProjectionEvaluator
       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /opt/ros/melodic/include/ompl/base/StateSpace.h:43:0,
                 from /root/ws_moveit/src/moveit/moveit_planners/ompl/ompl_interface/include/moveit/ompl_interface/parameterization/model_based_state_space.h:40,
                 from /root/ws_moveit/src/moveit/moveit_planners/ompl/ompl_interface/include/moveit/ompl_interface/model_based_planning_context.h:40,
                 from /root/ws_moveit/src/moveit/moveit_planners/ompl/ompl_interface/src/model_based_planning_context.cpp:39:
/opt/ros/melodic/include/ompl/base/ProjectionEvaluator.h:146:26: note: 	virtual void ompl::base::ProjectionEvaluator::project(const ompl::base::State*, Eigen::Ref<Eigen::Matrix<double, -1, 1> >) const
             virtual void project(const State *state, Eigen::Ref<Eigen::VectorXd> projection) const = 0;
                          ^~~~~~~

```

davetcoleman, 2018-05-17:
I'd like Melodic to target the new ompl_interface @BryceStevenWilley @zkingston so maybe we can save our effort trying to port the old ompl_interface since im sure the Kavraki lab already has their interface working with the latest OMPL @mamoll

zkingston, 2018-05-18:
@davetcoleman we mostly use the `ompl_interface` that I opened a pull request for in #897. @BryceStevenWilley fixed the breaking change issue of `ompl::base::ProjectionEvaluator` -> `Eigen::Ref<Eigen::VectorXd>` in #903, which was done as we removed all `boost::ublas` in https://github.com/ompl/ompl/commit/1396bd09794cd5422ba02a8ef662f42239d06552.

Other things that could be added with the new OMPL features:

1. New planners like BIT*, I-RRT*, RRT#, RRTX, SORRT* (BIT* and I-RRT* are added in #897, as `ModelBasedStateSpace` is changed to inherit from `ompl::base::RealVectorStateSpace`. The other planners are trivial to add after that).
2. `interpolate()` now returns a boolean indicating whether the path is still valid, which can be checked in `ModelBasedPlanningContext::interpolateSolution()`
3. Geometrically constrained planning (documented here: http://ompl-beta.kavrakilab.org/constrainedPlanning.html), which would enable path constrained planning in joint space rather than Cartesian end-effector space.

rhaschke, 2018-05-24:
Fixed with #903.

closed 2018-05-24 by rhaschke

YichenZhou113, 2018-06-20:
Hi, I am having the same problem on my ubuntu 16.04. Is there any way that I could use ompl? I tried with the older version but it said cannot read from or find config.h. So how can I use ompl by now?

rhaschke, 2018-06-20:
@YichenZhou113 If you have updated OMPL and get the same error on Kinetic now, you could apply #903 to your kinetic branch of MoveIt as well.
If you get a different error, you should precisely describe your issue.

YichenZhou113, 2018-06-20:
@rhaschke you mean change the following?
https://github.com/ros-planning/moveit/commit/83474f90b7e96797a4fe32adf133f2e48765199a