moveit/moveit #904: MoveIt! OSS And Its Licenses

davetcoleman, 2018-05-18:
As we work more with clients under contract I'm paying more attention to licenses and so today I ran the [scancode-toolkit](https://github.com/nexB/scancode-toolkit) on all the repos in our ``moveit.rosinstall``. We currently have the following licenses:

License | Category
-- | --
Apache 2.0 | Permissive
BSD-Modified | Permissive
GPL 1.0 or later | Copyleft
LGPL 2.0 or later | Copyleft Limited
LGPL 2.1 or later | Copyleft Limited
MIT License | Permissive

Of concern are the GPL and LGPL licenses of course. The only usages of those licenses are:

- KDL Inverse Kinematics Solver
- PR2 Meshes in ``moveit_resources/pr2_description/urdf/meshes/``

We are already phasing out the PR2 with the very recent switch the the Franka Emika robot, but finding a replacement for the default out-of-the-box IK solver will be a much bigger task, as the KDL solver is quite popular. This issue is to open up discussions for this need to eventually remove those copyleft licenses entirely. 

@mikeferguson might have thoughts since I believe we discussed this a while back at a conference and he's help found (several) robot companies with MoveIt!

mikeferguson, 2018-05-18:
GPL sucks, and should be avoided like the plague. Especially for something dumb like meshes (which really should be covered by some sort of Creative Commons license, rather than a software license).

LGPL is actually probably fine here for the KDL solver -- since we are in fact linking against KDL and/or the solver. The only people who would have problems with LGPL would be someone trying to adapt/improve the KDL solver but keep their new thing private -- but as you said, it's mainly used because it really is "out of the box default".

One additional note I would make towards anyone with a commercial vibe: we've never really had contributor license agreements (CLAs), and unfortunately it's kinda late now, since Willow is gone and many of the original contributors have moved on from where they contributed from. The lack of CLAs basically means that the license headers on each file are really just "on the honor system" and nobody has ever certified they had the right to contribute under such license, etc. However, Apache2 actually does a pretty good job of wrapping up the equivalent of the CLA within the license. In addition, it tries to avoid the nasty problem of someone contributing patented things and then later suing (although, it should be noted these things have never really been tested in court). Therefore, I would recommend that future new contributions (like, whole new modules, not changes to existing files that are otherwise licensed) are Apache2 whenever possible. Most of the ROS2 code is being done under Apache2 AFAIK.

gavanderhoorn, 2018-05-18:
@mikeferguson wrote:
> LGPL is actually probably fine here for the KDL solver 

I was going to write something about this as well (hadn't seen your comment yet @mikeferguson): LGPL is specifically meant to 'work-around' the issues of the GPL wrt 'infection' of other code-bases.

I don't believe there is a direct need to remove KDL from MoveIt based on the fact that it is LGPL.

>Most of the ROS2 code is being done under Apache2 AFAIK.

yes, that is my impression as well.

And an additional datapoint: just about all (new) code in ROS-I is also Apache 2.

IanTheEngineer, 2018-05-18:
I did a similar license investigation prior to starting Sawyer SDK development, and came to pretty much the same conclusions: LGPL should be fine if its only linked against, and Apache 2.0 is preferable for any new development, due to its patent protections & CLA equivalence-ness.

davetcoleman, 2018-05-18:
Great discussion here, thanks!

> The only people who would have problems with LGPL would be someone trying to adapt/improve the KDL solver but keep their new thing private

KDL is modified in MoveIt! to support extra features like consistency limits and mimic joints, so I'm not sure this point is valid. Many organizations/companies use MoveIt! for their private IP / projects. So I'm not convinced this is a safe library to use. I want to be able to safely claim MoveIt! is business-friendly under BSD but I think if you use KDL you are at risk here. I'm working with my lawyer on this debate. 

> However, Apache2 actually does a pretty good job of wrapping up the equivalent of the CLA within the license. 

Great point @mikeferguson, that makes sense to me moving new stuff to Apache2

mikeferguson, 2018-05-18:
> KDL is modified in MoveIt! to support extra features like consistency limits and mimic joints, so I'm not sure this point is valid. Many organizations/companies use MoveIt! for their private IP / [...]

Yes, but the issue isn't so much that there is modified code there, as much as those modifications are already public (thus, in compliance with license).

The issue for commercial users arises when they want to modify AND keep it private. So, if I want to improve the plugin itself, I have to make the modified source available under terms of the LGPL. Any other application which simply links against an LGPL library is not required to be distributed under any particular license terms.

Obviously, my background is more in startups than larger organizations, but the conversation usually goes like this:
 * me: "we have some LGPL dependencies"
 * Lawyer: "do you need to modify those LGPL libraries to build your product?"
 * me: "no"
 * Lawyer: "then we don't really about those"
 * If the answer were "yes" or "maybe", the key question would be: "are those modifications essential IP?". Most organizations don't care about keeping some bug fix private (the overhead to maintain a complete fork usually isn't worth the competitive advantage of having a less buggy version of some common software), but will worry quite a bit if the core of their new product has to be open sourced.

Now, I suppose someone could claim that "all of moveit is one big binary" and thus should all be LGPL. In that case, simply moving the KDL plugin back into it's own repository should suffice to show it is a separate project, which MoveIt simply links against.

Honestly, if someone really wants to find a problem with the OSS nature -- I would imagine that the lack of CLAs is a far bigger risk than any of the LGPL stuff.

simonschmeisser, 2018-06-11:
My personal assesment was also that lgpl dependencies don't matter (there are even ways to allow static linking if you must) ... but also that switching to trac_ik as a default instead of kdl wouldn't hurt. It's BSD and fails less often.

davetcoleman, 2018-06-14:
trac_ik simply uses KDL and another IK solver under the hood, in separate threads. It does not avoid the licensing problem

130s, 2018-06-29:
`scancode-toolkit` looks useful! That looks like traversing the source code to find licenses used IIUC.

I may or may not be necesarilly adding a new perspective, but I've been making a bit more ROSy tool for software license https://github.com/ros-infrastructure/rospkg/pull/143. This feature takes in a single package name to begin with, then search up the dependency chain then returns all software licenses found defined in `package.xml` for ROS packages, and also returns all licenses found from the `dpkg` output for non-ROS/system packages.

I ran for `moveit_setup_assistant` and `moveit_commander` as those 2 look to be the most "leaf" in MoveIt!'s repo (correct me if wrong), AFAI see the [rqt_dep](http://wiki.ros.org/rqt_dep) result. 
[image]

This does reveal some licenses used in the dependency chain for MoveIt!. I do see some `GPL`s, but that by itself doesn't mean we need to take action for it right now.

Note because some (/many?) system packages don't follow the recommendation for license declaration yet, `dpkg` doesn't return the result for all system packages, thus licenses are missing for some packages in this tool's result.

```
$ docker images|grep -i moveit
moveit/moveit                                                      melodic-release                                 8d4528af312b        4 days ago          2.18GB
$ docker run -it moveit/moveit:melodic-release
# cd ~/ws_moveit/src/
# git clone https://github.com/plusone-robotics/rospkg.git && cd rospkg
# git checkout add_license_introspection_syspkg_incl_pr141-144
root@d16851eb4640:~/ws_moveit/src/rospkg# export PYTHONPATH=~/ws_moveit/src/rospkg/src:$PYTHONPATH
root@d16851eb4640:~/ws_moveit/src/rospkg# export PATH=~/ws_moveit/src/rospkg/scripts:$PATH 
root@d16851eb4640:~/ws_moveit/src/rospkg# sw_licenses moveit_commander
:
Path of the output file: /tmp/licenses_moveit_setup_assistant-0.10.1.yml
root@d16851eb4640:~/ws_moveit/src/rospkg# cat /tmp/licenses_moveit_setup_assistant-0.10.1.yml

(System package. License not automatically detected):
- apr
- assimp
- assimp-dev
- boost
- cppunit
- eigen
- glut
- google-mock
- gtest
- libconsole-bridge-dev
- libglew-dev
- libqhull
- libssl-dev
- liburdfdom-dev
- log4cxx
- lz4
- opengl
- python-argparse
- python-coverage
- python-imaging
- python-lxml
- python-mock
- python-qt5-bindings
- qtbase5-dev
- tinyxml
- tinyxml2
Apache 2.0:
- gennodejs
Apache License 2.0:
- ros_environment
Apache-2.0 BSD-2-clause BSD-3-clause BSD-4-clause GPL-2+with_exception GPL-3+with_exception ISC MIT-like zlib:
- cmake
BSD:
- actionlib
- actionlib_msgs
- angles
- catkin
- class_loader
- cmake_modules
- cpp_common
- cv_bridge
- dynamic_reconfigure
- eigen_stl_containers
- gencpp
[... 299 lines cut ...]
GPL-2 GPL-2+ GPL-2+ with Bison exception GPL-3 SIP SIP or GPL-2 or GPL-3:
- python-sip
GPL-2+ GPL-3+ LGPL-2+ LGPL-2.1+ LGPL-3+ LGPL-3+ or GPL-2+:
- libgpgme-dev
GPL-compatible GPL-compatible licenses:
- python
LGPL:
- orocos_kdl
- python_orocos_kdl
LGPL-2.1:
- python-paramiko
- xmlrpcpp
MIT-style:
- python-netifaces
Python:
- python-defusedxml
public domain:
- python-numpy
unknown:
- python-catkin-pkg
- python-rosdep
- python-rospkg
- python-wxtools
- python-yaml
```

130s, 2018-06-29:
By mistake I pasted results from moveit_setup_assistant twice. https://github.com/ros-planning/moveit/issues/904#issuecomment-401447368 is updated to include the output from moveit_commander.

130s, 2019-03-08:
`moveit_ros_perception` depends on [libglew-dev](https://github.com/ros-planning/moveit/blob/ae7832396e3153c6cab053535ccb43c58330d439/moveit_ros/perception/package.xml#L30), part of which is GPL according to a source (yeah, [Wikipedia...](https://en.wikipedia.org/wiki/OpenGL_Extension_Wrangler_Library)).

> GLEW is distributed under modified BSD licenses, and the header file generator is under the GNU GPL.

Not sure exactly though, as far as I look at the current code the "header file generator" is not in use so we're "fine" IMO.

```
$ ack-grep -A 1 -B 1 -i glew .
package.xml
29-  <depend>glut</depend>
30:  <depend>libglew-dev</depend>
31-  <depend>opengl</depend>

CMakeLists.txt
15-  find_package(OpenGL REQUIRED)
16:  find_package(GLEW REQUIRED)
17-  find_package(GLUT REQUIRED)
--
19-  set(perception_GL_INCLUDE_DIRS "mesh_filter/include" "depth_image_octomap_updater/include")
20:  set(SYSTEM_GL_INCLUDE_DIRS ${GLEW_INCLUDE_DIR} ${GLUT_INCLUDE_DIR})
21-endif(WITH_OPENGL)

mesh_filter/include/moveit/mesh_filter/gl_renderer.h
40-#include <moveit/macros/class_forward.h>
41:#include <GL/glew.h>
42-#ifdef __APPLE__

mesh_filter/include/moveit/mesh_filter/gl_mesh.h
40-#include <Eigen/Eigen>
41:#include <GL/glew.h>
42-#ifdef __APPLE__

mesh_filter/src/gl_renderer.cpp
36-
37:#include <GL/glew.h>
38-#ifdef __APPLE__
--
393-
394:    GLenum err = glewInit();
395:    if (GLEW_OK != err)
396-    {
397-      stringstream errorStream;
398:      errorStream << "Unable to initialize GLEW: " << glewGetErrorString(err);
399-

mesh_filter/CMakeLists.txt
11-
12:target_link_libraries(${MOVEIT_LIB_NAME} ${catkin_LIBRARIES} ${gl_LIBS} glut GLEW)
13-
n130s@130s-p50:~/link/ROS/kinetic/cws_planning/src/ros-planning/moveit/moveit_ros/perception$ git log -1
commit b2cbfc45b8d08149c4de0383d6a92c0af5b25165
Author: Robert Haschke <rhaschke@users.noreply.github.com>
Date:   Wed Mar 6 08:35:58 2019 +0100

    fix segfault in chomp adapter (#1377)
```