moveit/moveit #913: Melodic binary release issues

rhaschke, 2018-05-24:
@clalancette Having released MoveIt for Melodic, we have some strange cmake errors in the binary release builds on the ROS build farm:

http://build.ros.org/job/Mbin_uB64__moveit_setup_assistant__ubuntu_bionic_amd64__binary/3/display/redirect

Essentially cmake complains, that `-lpthreads` cannot be found. 
However, I don't understand, why it doesn't look for `-lpthread` as well... 

Do you have seen this error already in a different context?

clalancette, 2018-05-24:
@rhaschke Yeah, we've been seeing that elsewhere.  That particular error message is a red herring; the real error is always further up.  If you look at http://build.ros.org/job/Mbin_uB64__moveit_setup_assistant__ubuntu_bionic_amd64__binary/3/console around time 23:15:34, you'll see that the real error is:

```
23:15:34 CMake Error at CMakeLists.txt:36 (find_package):
23:15:34   By not providing "FindQt5.cmake" in CMAKE_MODULE_PATH this project has
23:15:34   asked CMake to find a package configuration file provided by "Qt5", but
23:15:34   CMake did not find one.
23:15:34 
23:15:34   Could not find a package configuration file provided by "Qt5" (requested
23:15:34   version 5.9.5) with any of the following names:
23:15:34 
23:15:34     Qt5Config.cmake
23:15:34     qt5-config.cmake
23:15:34 
23:15:34   Add the installation prefix of "Qt5" to CMAKE_PREFIX_PATH or set "Qt5_DIR"
23:15:34   to a directory containing one of the above files.  If "Qt5" provides a
23:15:34   separate development package or SDK, be sure it has been installed.
```

rhaschke, 2018-05-24:
Thanks a lot for this hint.

closed 2018-05-24 by rhaschke