moveit/moveit #914: Itegrating Voxblox to moveIt! - build a TSDF directly out of sensor data

Magnusgaertner, 2018-05-24:
Hi everyone, 

I want to integrate Voxblox into moveIt! to directly build a TSDF world in order to improve the performance of optimization based planners like chomp and stomp. This is the main part of my bachelor thesis in which i will also measure how much overall planning time can be saved once one does not have to convert from an octomap to a TSDF representation as the current chomp implementation does. If you have any suggestions on how this could/should be done I'd be happy to hear them. 

Best Regards
Magnus

felixvd, 2018-09-06:
I suppose it's just about submission time. Any updates?

gavanderhoorn, 2018-09-06:
@Magnusgaertner's branches: [Magnusgaertner/moveit/branches/all](https://github.com/Magnusgaertner/moveit/branches/all).

Diff with `kinetic-devel` (where the main development seems to happen): [kinetic-devel...Magnusgaertner:kinetic-devel](https://github.com/ros-planning/moveit/compare/kinetic-devel...Magnusgaertner:kinetic-devel).

Magnusgaertner, 2018-09-12:
@felixvd there are about two months till submission ;) 
A first setup is working allowing to use ie Stomp with an collisionchecker that uses CollisionSpheres computed out of the collision geometry of the robot which are checked against an esdf world representation generated by voxblox's esdf server. However, the integration of a different map type, not just a collision checking plugin that sideloads the representation, is still under development.

davetcoleman, 2018-10-30:
We'd love to a see a PR out of this!