moveit/moveit #925: Ability to build melodic-devel branch on ROS Kinetic

IanTheEngineer, 2018-05-29:
### Description
At the maintainers meeting, there was some discussion around whether or not to build the MoveIt `melodic-devel` branch on ROS Kinetic. This issue will consolidate the information around this request.

Summing up the concerns voiced: Due to the tf2 migration https://github.com/ros-planning/moveit/pull/830 for ROS Melodic, there is some worry that the community will not use or contribute to MoveIt's `melodic-devel` in the near future if MoveIt's source branches cannot be built interchangeably between Kinetic and Melodic. @rhaschke, @v4hn, and @davetcoleman please add your thoughts to this issue if I am not capturing the sentiments properly.

Backstory: The following upstream API changes were made for Melodic to enable MoveIt to use tf2, and would need to be back-ported into ROS Kinetic for MoveIt to have the same branch build on both Kinetic and Melodic:
1) https://github.com/ros/geometry/pull/163 (Exposing `tf2_ros::Buffer` object from TransformListener)
2) https://github.com/ros/geometry2/pull/292 (tf2 Type conversions)
3) https://github.com/ros/geometry2/pull/294 (Additional tf2 Type conversions)
4) https://github.com/ros-visualization/rviz/pull/1236 (Exposing tf2 without ROS deprecation warnings)

- Upstream PRs 2 and 3 only add type conversions to `geometry2`, and can likely be backported. 
- PR 1 changes the API of the `tf::Transformer` in order to expose the underlying `tf2_ros::Buffer`. It's not clear whether this can be backported into ROS Kinetic. @tfoote do you have thoughts on the backportability of this PR?
- PR 4 is the tricky one, it changes the RViz API to expose the `tf2_ros::Buffer` API that PR 1 exposes. @wjwwood is there any chance this new API can be backported into ROS Kinetic? Alternatively, could we just skip backporting RViz's changes if we can utilize the PR 1 Transformer changes directly in ROS Kinetic (assuming PR 1 is backported)?

I can help out with backporting these PR's for each repo if we decide to do so.

### Environment
* ROS Distro: Kinetic & Melodic 
* OS Version: e.g. Ubuntu 16.04 & 18.04

### Requested behavior
`melodic-devel` to build on both ROS Melodic and Kinetic

### Actual behavior
`melodic-devel` only builds on ROS Melodic.

rhaschke, 2018-05-29:
I think in https://github.com/ros/geometry2/pull/292#issuecomment-391826297 @tfoote already agreed to backport these changes to Kinetic in general. However, this issue nicely summarizes all the details. Thanks for providing this, @IanTheEngineer!

v4hn, 2018-05-30:
Thank you a lot for the summary Ian,

we need reviews by @wjwwood and/or @tfoote before we can discuss more here.

The MoveIt Maintainer meeting agreed before that the latest development branch of MoveIt - since a few days `melodic-devel` - should compile on all supported ROS distributions, even if this invokes overhead.
With the changes here, I don't really see how to compile this on indigo anymore. That is a huge break already and I don't like it.

But it will take a long time before most (or even any...) robots arrive on ROS melodic.
Even worse, *maintainers* who want to test the latest development branch on their robots are often bound to kinetic for the near future.
In short: we *need* the branch to compile on ROS kinetic.

Ian, can you elaborate on (4) please?
First you say this is the most critical change, then you say we might not need it?
I'm confused.

For (1), if we can't backport C++11, it is simple to add an adaptive `tf2_ros::BufferPtr` typedef.

wjwwood, 2018-06-01:
> @wjwwood is there any chance this new API can be backported into ROS Kinetic?

If the underlying API is backported to geometry (`tf`) then I think we could also backport the new API in rviz. I'd have to double check that it doesn't break ABI and that there's no issue with using `std::shared_ptr` in older releases (there might be actually).

> The MoveIt Maintainer meeting agreed before that the latest development branch of MoveIt - since a few days `melodic-devel` - should compile on all supported ROS distributions, even if this invokes overhead.

That's a fine goal, but when you're spanning three LTS ROS releases at a time you're likely to span entire tick-tock cycles, which might be impossible to support. For instance, in Kinetic you only have `getTFClientPtr()`, in Melodic you have both `getTFClientPtr()` (deprecated) and `getTF2BufferPtr()`, but in the next LTS you'll only have `getTF2BufferPtr()` (due to tick-tock), so your `<next LTS>-devel` branch won't be able to build on all supported ROS releases.

If you're worried about developers being able to test new additions, then I'd recommend making `kinetic-devel` your default branch, target pr's there, and forward port anything you need to `melodic-devel`. Then you can target changes that can't be done on Indigo/Kinetic to `melodic-devel` without backporting them elsewhere. It's more work, but also less likely to get into an impossible to support situation like you can get when supporting three LTS at a time.

v4hn, 2018-06-03:
> That's a fine goal, but when you're spanning three LTS ROS releases at a time you're likely to span entire tick-tock cycles

... which is why I don't discuss indigo here.
Still, this is no argument to drop support for kinetic in the latest branch - the "previous" LTS release which is still the current for most actual applications.

> I'd have to double check that it doesn't break ABI

Of course this breaks ABI.
See https://github.com/ros/geometry/pull/163/files#diff-2bb36134ef9a320421ff09ade7542da3R391
TF didn't see the need for implementation classes.

@IanTheEngineer would an unsafe raw pointer accessor for kinetic work around the problem for us without changing ABI or do you rely on the smart pointer?

wjwwood, 2018-06-04:
> ... which is why I don't discuss indigo here.

Sorry, I thought Indigo was included in "all supported ROS distributions", but I guess you meant supported by moveit? Indigo is support until April 2019 (https://wiki.ros.org/Distributions#List_of_Distributions).

> Of course this breaks ABI.

I mean in rviz, where I'm not sure it does. Geometry may choose to break ABI, but I don't think they want to.

> TF didn't see the need for implementation classes.

Yeah... Those jerks. 😁

v4hn, 2018-06-04:
We still support the indigo branch with releases, but we stopped backporting changes unless requested by users.
Personally, I would like to extend compatibility back to indigo too, but I agree with you on the associated problems.

IanTheEngineer, 2018-06-04:
> @IanTheEngineer would an unsafe raw pointer accessor for kinetic work around the problem for us without changing ABI or do you rely on the smart pointer?

A raw pointer itself won't work unless we make further modifications to MoveIt. MoveIt's  `move_group` and `planning_scene_monitor` need to maintain shared control of the `Transformer` object to perform TF lookups. This is exactly the issue I was encountered when attempting to cast my way out of modifying geometry and RViz: https://github.com/ros-visualization/rviz/issues/1215#issuecomment-379493477

The way MoveIt has used the RViz transform data has relied on shared ownership of the Transformer object: 

[planning_scene_display.cpp](https://github.com/ros-planning/moveit/blob/melodic-devel/moveit_ros/visualization/planning_scene_rviz_plugin/src/planning_scene_display.cpp#L489-L491) via getFrameManager() from RViz:
```
planning_scene_monitor::PlanningSceneMonitorPtr PlanningSceneDisplay::createPlanningSceneMonitor()
{
  return planning_scene_monitor::PlanningSceneMonitorPtr(new planning_scene_monitor::PlanningSceneMonitor(
      robot_description_property_->getStdString(), context_->getFrameManager()->getTF2BufferPtr(),
      getNameStd() + "_planning_scene_monitor"));
}
```
and [motion_planning_frame.cpp:](https://github.com/ros-planning/moveit/blob/melodic-devel/moveit_ros/visualization/motion_planning_rviz_plugin/src/motion_planning_frame.cpp#L317-L318)
```
move_group_.reset(new moveit::planning_interface::MoveGroupInterface(
opt, context_->getFrameManager()->getTF2BufferPtr(), ros::WallDuration(30, 0)));
```

IanTheEngineer, 2018-06-05:
> Shared pointers do not prevent locking issues.

We are in agreement that shared pointers do not prevent locking, but they do prevent the `TransformListener` from being deleted when MoveIt is still using its pointer.

> I'm unsure what you mean by move_group class. Do you mean MoveGroupInterface?

Yes, please see the link above where I pointed out the usage in `MoveGroupInterface`.

> Alternatively, do we have to share the buffers or could we just create a second tf buffer for kinetic?
Of course it would be overhead, but otherwise it should work, right?

I think that could work. I did spend some time on this approach, but ended up going the route of exposing the `Buffers` upstream to reduce runtime overhead. It's not clear to me how often these `tf2_ros::Buffers` would be created and destroyed, and how many seconds are appropriate to wait for the `Buffers` to be filled before they are considered "ready" for the `MoveGroupInterface`. I'd be happy to take input as to how to handle this best. The two links above show the only two locations where we would need to maintain these `Buffer`'s and their associated `tf2_ros::TransformListener`.

IanTheEngineer, 2018-07-25:
### Update:
The tf2 changes have been backported to Kinetic/Indigo: https://github.com/ros/geometry2/pull/309 Thanks to @tfoote for making that happen.

The only remaining hiccup is to figure out how to deal with `Transformer`'s internal `Buffer` objects. I have pinged @tfoote and @wjwwood to get their assessment on whether we can backport the initial `Transformer` pull request, https://github.com/ros/geometry/pull/163#issuecomment-407804093, as that would be the simplest and cleanest method to handle these two branches.

v4hn, 2018-07-31:
> The tf2 changes have been backported to Kinetic/Indigo

Great!

> The only remaining hiccup is to figure out how to deal with `Transformer`'s internal `Buffer` objects.

Tully and William both made it clear that this change not be backported to remain ABI compatible and not invoke C++11.
I agree with their reasoning.

So we have to find a workaround.

The simplest thing might be to instantiate a separate `tf::Transformer` in this case.

Do you see alternatives @IanTheEngineer?
I did not look into the details here.

davetcoleman, 2018-10-13:
@IanTheEngineer as discussed at ROSCon, can we move this forward? We really need to have this before World MoveIt! Day on Oct 25th...

Can we just use ugly compiler defs with a ``// TODO: remove after dropping Ubuntu 16.04 support`` ?

IanTheEngineer, 2018-10-15:
@davetcoleman yes, ugly compiler defines are an option. Despite their lack of a pleasing ascetic appearance, they actually might be desirable: for true backwards compatibility, I would need to create and maintain redundant tf2 buffers & listeners in both `kinetic` and `melodic`, but with compiler defines, I can use RViz's shared pointers to them (tf1 & tf2 respectively) as intended. I'll look into this option in my free time this week.

davetcoleman, 2018-10-18:
I wanted to test this setup so documented melodic on 16.04:
https://github.com/ros-planning/moveit.ros.org/pull/205

v4hn, 2018-10-24:
This seems to be resolved, although I did not test it myself yet.

closed 2018-10-24 by v4hn